cqparts/cqparts #1: Thread solids invalid

fragmuffin, 2017-10-05:
TBD

jmwright, 2017-10-05:
Is there a way to detect when a thread is invalid so that fixes could be applied conditionally and then rechecked? I know how to check geometry from the GUI, but I've never done it from a script.

fragmuffin, 2017-10-06:
@jmwright 
There is a section in the code to do exactly that, but it's only working some of the time (less than half)
https://github.com/fragmuffin/cqparts/blob/develop/src/cqparts/types/threads/base.py#L251-L260
when building a dodgy thread at the moment, you'll see:

```
thread shape not valid
new thread STILL not valid
```

on the console... context is obvious after reading the above linked code

fragmuffin, 2017-10-25:
These tools might be a clue:
https://www.freecadweb.org/wiki/Mesh_FillHoles
https://www.freecadweb.org/wiki/Mesh_FillInteractiveHole

fragmuffin, 2017-12-22:
This bug is making thread-building completely unmanageable.
It's making building a very painful and error-prone experience... the presence of thread faces makes `cqparts` very unstable.

**Mitigated by Simplifying to Cylinder**

For the time being, I've converted the base `Thread` type to a `Part` so I can exploit the `_simple=True` flag, which causes the `make_simple()` method to build the part's `.local_obj` as opposed to `.make()`. (see f1e98fe)

In other words, by default, threads are _not_ generated.
Instead they are replaced with a cylinder with the thread's outer radius.

**Workaround**

If you'd like to play with _real_ threads, you can turn off default simplification with any of...

```python
import cqparts
from cqparts.display import display
from cqparts.params import Boolean
from cqparts.fasteners.base import FastenerMalePart
from cqparts.solidtypes.threads import base

# --- an individual part
screw = FastenerMalePart()
screw.thread._simple = False
display(screw)

# --- all threads by default
base.Thread._simple.default = False
# you'll have to re-define the thread type, because the default thread
# was instantiated before the above change was made
screw = FastenerMalePart(
    thread=('iso262', {
        'diameter': 3.0,
        'pitch': 0.7,
    }),
)
display(screw)

# --- all threads by default (by environment variable)
# because the above way is messy... you can set an environment variable:
#       CQPARTS_COMPLEX_THREADS=yes
display(FastenerMalePart())  # default of _simple is False at import time
```

none of the above methods are permanent, please keep this in mind if you're planning a future-proofed script.

**Fix Delay**

I also won't have time to fix this for the initial release... but it's very high on the priorities list *after* v0.2

fragmuffin, 2017-12-22:
example of the simplified thread...

[image]

fragmuffin, 2018-02-13:
Test created to raise failure: [test](https://github.com/fragmuffin/cqparts/blob/master/tests/test_fasteners_thread.py) and [catalogue](https://github.com/fragmuffin/cqparts/blob/master/tests/test-files/thread_catalogue_build.py)
A variety of thread dimensions are created, most of which currently fail to create a watertight object.

Error example

```
======================================================================
ERROR: test_triangular_4x1_040 (cqparts.utils.test.CatalogueTest_thread_catalogue)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/nymphii/prj/cqparts/tests/../src/cqparts/utils/test.py", line 147, in test_meth
    obj.build()
  File "/home/nymphii/prj/cqparts/tests/../src/cqparts/part.py", line 91, in build
    self.local_obj  # force object's construction, but don't do anything with it
  File "/home/nymphii/prj/cqparts/tests/../src/cqparts/part.py", line 113, in local_obj
    value = self.make()
  File "/home/nymphii/prj/cqparts/tests/../src/cqparts_fasteners/solidtypes/threads/base.py", line 352, in make
    "created thread solid cannot be made watertight"
SolidValidityError: created thread solid cannot be made watertight
```

fragmuffin, 2018-02-13:
Tests skipped until this issue is closed:

```text
test_ball_screw_10x1_012 (cqparts.utils.test.CatalogueTest_thread_catalogue) ... skipped 'skipped until #1 is fixed'
test_ball_screw_10x3_013 (cqparts.utils.test.CatalogueTest_thread_catalogue) ... skipped 'skipped until #1 is fixed'
...
test_triangular_4x4_042 (cqparts.utils.test.CatalogueTest_thread_catalogue) ... skipped 'skipped until #1 is fixed'
test_triangular_4x5_043 (cqparts.utils.test.CatalogueTest_thread_catalogue) ... skipped 'skipped until #1 is fixed'
```

**Task**

* [ ] remove `FIXME` from [test_fasteners_thread.py](https://github.com/fragmuffin/cqparts/blob/master/tests/test_fasteners_thread.py#L11-L12)

fragmuffin, 2018-03-19:
**Task**

* [ ] remove warning about this fault from `cqparts_fasteners` `README.rst`