cqparts/cqparts #10: Confer with cadquery contributors

fragmuffin, 2017-10-07:
Discuss major design decisions with [cadquery](https://github.com/dcowden/cadquery) contributors before releasing.

**Part / Assembly**
* conceptual problems?
* too simplistic?
* bad use-cases?

**What to Port to cadquery**
```
TODO: list what may be ported to cadquery
```
(could instead be done with future versions, but would be cleaner if it could be done pre-release)

**Reliance on FreeCAD specific API**
Many "tricky" things are done by calling the FreeCAD API's directly.
This will be a problem if `cadquery` supports multiple CAD kernels.

So this loops back to "_porting to `cadquery`_"

jmwright, 2017-10-08:
We haven't defined how FreeCAD vs PythonOCC is going to work in the core library. I would like to see them coexist so that users can switch between them in the FreeCAD module, but they're fairly different implementations. The way Adam Urbanczyk has started with the PythonOCC implementation branch suggests that maybe they can coexist. We need to make a decision on this sometime in the not so distant future.

jmwright, 2017-10-08:
Better tag @dcowden on this one too.

fragmuffin, 2017-10-08:
Regardless of when the decision is made,  or how it's done, I can make the transition much easier for myself (within `cqparts`) by writing the parts test-cases with a select API.

For example

```python
import unittest
import cadquery
import cqparts

class PulleyTests(unittest.TestCase):
    cadquery_base = None
    run_tests = False  # use to rule out classes during unittest discovery

    def setUp(self):
        # probably not how it will work, but you get the point
        cadquery.select_base(self.cadquery_base)

    def test1(self):
        ...
    def test2(self):
        ...

class FreeCADPulleyTests(PulleyTests):
    cadquery_base = 'freecad'
    run_tests = True

class OpenCCPulleyTests(PulleyTests):
    cadquery_base = 'occ'
    run_tests = True

class ParasolidPulleyTests(PulleyTests):
    cadquery_base = 'parasolid'
    run_tests = True
```

fragmuffin, 2017-10-08:
@jmwright @dcowden ok that was a bit abstract.
I guess what I'm trying to say is: porting functionality to `cadquery` will be necessary at some point, but it's power is much more evident when a split is made between CAD kernels within `cadquery`.

Porting functionality to `cadquery` will then also benefit anyone using `cadquery` without `cqparts`.

The above test method will make the discovery process of _what_ to port much more clear.

dcowden, 2017-10-08:
@fragmufgin @jmwright Thanks guys. The most important decision to make is whether and how  cadquery will support multiple backends.

Cq 1.x contemplated multiple backends, but used leaky abstractions. Many users ended up using  functionality from the underlying cad kernel.

Initially I hoped to avoid this need, but now I can see that it is inevitable. There is no way cadquery can practically create abstractions that cover all the functionality available in ANY backend.

That means we should assume that many authors will write code that's bound to a particular implementation eventually. One goal of cqparts would be be to that work for it's users.

Cq 1.x needs to support freecad and only freecad for a while to come. We don't want to break it

We have a few choices for cq2.
We can declare that it supports only OCc.this is the current direction, motivated by simplicity,and the relative unlikiness that we will ever see a parasolid backend.

An earlier version of cq 2 had  abstractions to support multiple backends. 

I think my vote now would be to continue with the idea that cq 2.0 is based on pythonOCC only. It will be much simpler to  coordinate work, and since freecad is based on OCC we lose little in this   decision.  Supporting both freecad and occ from a single cq library is not worth the complexity.  

If cq does become popular enough that there is interest in a parasolid backend (meaning, someone fronts the $100k development license), we can face that then. But it would be fine to consider that a separate project because at that point it would not be free to users.

fragmuffin, 2017-11-25:
Lead `cadquery` contributors have been very active in giving me excellent feedback.
This issue as a sort of *forum* is not required; closing issue.

closed 2017-11-25 by fragmuffin