cqparts/cqparts #11: Change name?

fragmuffin, 2017-10-07:
`cqparts`?... really?, perhaps it should be tied more closely to  the full name `cadquery`

\<PyPI name> : \<lib name>
* [x] `cqparts` : `cqparts`
* [ ] `cadquery-parts` : `cadqueryparts`

jmwright, 2017-10-07:
@dcowden and I refer to CadQuery as CQ in our direct communications, so "cqparts" seems fine to me. We did have somebody post a long time ago that "CQ" was already the name of some other software package, but I had never heard of the software before, and haven't heard of it since.

fragmuffin, 2017-10-08:
@jmwright this is the only thing that pops up
https://docs.adobe.com/docs/en/cq/5-4/exploring/concepts.html
It has nothing to do with CAD, so I can't see a problem.

I also like that `cqparts` has the same package name as it's python import.
There's only 26² possible 2 letter combinations; can't always be unique ;).
Ok, I'll leave it as `cqparts`

fragmuffin, 2017-10-28:
Leaving as simply:
`cqparts` : `cqparts`

closed 2017-10-28 by fragmuffin