cqparts/cqparts #5: Fastener Application

fragmuffin, 2017-10-07:
Create the mechanisms to usefully apply fasteners to a design.

Avoid contrived examples; focus on examples you can see in everyday things.

Good source of inspiration would be ikea assembly instructions:
* http://www.ikea.com/au/en/assembly_instructions/ingolf-stool__AA-943676-2_pub.pdf
* http://www.ikea.com/au/en/assembly_instructions/pahl-underframe-for-table-top__AA-1660280-4_pub.pdf
* http://www.ikea.com/au/en/assembly_instructions/trysil-bedside-table__AA-1904650-1_pub.pdf

fragmuffin, 2017-10-07:
Idea for methodology

> There are 2 seemingly opposite use-cases that come to mind:
> * A pre-selected fastener is applied to a work-piece.
> * A work-piece needs fastening, with whatever fits the work-piece's parameters.
> 
> a proverbial chicken / egg scenario

one solution to this is to split fastener application into 3 classes:

**Evaluation**
The assessment of which solids require interfacing, in what order, and through what depth

_Input:_
* list of solids (arbitrarily ordered)
* vector origin
* fastener's axis (2 vertices, origin, direction)
  (note: the fastener's axis must be flush with, or above the first solid, but may be any distance above it)

_Output_
* List of effected solids (in order)
* A linear edge per effected solid (overlapping the solid along the fastener's axis)

There will be 2 base methods of evaluation:
- vector : faster, but may miss something (eg: vector may glance the edge of a solid, missing it)
- cylindrical : slower, but more accurate

**Selection**
A `Selector` class may be given an `Evaluation` instance to assess what fastener to use.
For example
```python
evaluation = VectorEvaluation([anchor, counterlever], ((0,5,1), (0,0,-1)))
wood_screw = WoodScrewSelector('10g', 'countersunk', 'phillips', anchor_ratio=1.5).select(evaluation)
# selector will select:
#    - a wood-screw with a 10g thread
#    - head will be countersunk
#    - screw drive will be phillips-head
#    - the depth of the selected fastener into the anchor will
#            be at least 1.5 * the height of the counterlever
```

**Application**
A given fastener can be applied based on a given evaluation.
So this one's simple

```python
Applicator(evaluation, wood_screw).apply()
```

**Why Classes over Functions?**
I like the idea of these being instantiated, and possibly serialized before they're evaluated.
This way all parts, and how they're connected can be defined before they're built.

For example, if a selector is used, then the same `Evaluator` instance is used twice.
It's functionally "evaluated" the first time it's referenced.

Another example, you may have 2 fastener applications that interfere with each other's evaluations.
This may be simplified by performing all evaluations first, then selecting each fastener and applying it (in an arbitrary order)

fragmuffin, 2017-11-15:
@dcowden @jmwright : I'm back!
I'll be working on this issue this week, I'll post some results when it's awesome to demonstrate exactly how it works.

dcowden, 2017-11-15:
sweet!!!

fragmuffin, 2017-11-30:
now that #26 is in place (pun intended), this can progress.

fragmuffin, 2017-12-01:
This is particularly tricky... in #26 I've built a build-chain of:

* **objects built** : `cadquery.Workplane` instances populated with solids
* **objects placed** : `cadquery.Workplane` moved from their local coordinates to world, using a constraints solver

However, only **after** placement has occurred can fasteners be evaluated, selected, and applied.
So there needs to be 2 considerable additions to this build chain (with the following limits)

* **additions** : components and constraints must be optionally added only *after* object placement has succeeded
    * can only add components & constraints (they can't be removed)
    * once a component's world coordinates are determined (by a constraints solver), they will not be changed.
    * additional constraints can only effect the position of a newly added component.
* **alterations** : parts may be altered (holes, chamfers, unions, and so on) after they've been placed.
    * alterations may only be made to the object at it's local coordinates (`local_obj` not `world_obj`)

There may also be multiple rounds of additions & alterations (because, why not).

I think the best way to achieve this is by allowing users to create `Assembly` instances that return generators instead of just dicts & lists... for example:

a basic assembly now:

```python
class Thing(cqparts.Assembly):
    def make_components(self):
        return {
            'box': Box(height=10),
        }
    def make_constraints(self):
        return [
            LockConstraint(self.components['box'], Mate((0,0,0))),
        ]
```

Could return generators instead

```python
class Thing(cqparts.Assembly):
    def make_components(self):
        yield {
            'box_a': Box(height=10),
        }
        yield {
            'box_b': Box(width=20),
        }
    def make_constraints(self):
        yield [
            LockConstraint(self.components['box_a'], Mate((0,0,0))),
        ]
        yield [
            LockConstraint(
                self.components['box_b'],
                self.components['box_a'].world_coords + Mate((0,0,10)),
        ]
```

note that the latter references `box_a`'s world coordinates which can only be set when a solver has completed.

So the "build chain" of *build objects*, then *place objects* will be iterated for each `yield`

This is effectively an equivalent of using `RelativeLockConstraint`.

As for alterations, this will be a 3rd step to the build chain, and will follow the same rules.

dcowden, 2017-12-01:
One constraints become more sophisticated ( for example, co planar,
concentric), won't it be necessary to be able to get all of the constraints
at once to evaluate them?

I understand the need to add components and constraints after assembly has
been solved.  An example would be that you have a box bottom and a lid, but
then you want to add a counter-sunk fastener and hole that goes through
both after they are assembled.

It is definitely tricky, because you additionally (probably) want to take
that original-but-now-has-a-fastener-hole-in-it box lid and export it for
manufacturing.

I need to think more about this. The system you're proposing works, but is
difficult to reason about.  My hunch is that an abstraction is missing to
help sort it out.

fragmuffin, 2017-12-01:
> One constraints become more sophisticated ( for example, co planar,
concentric), won't it be necessary to be able to get all of the constraints
at once to evaluate them?

Multiple constraints can be added per iteration (per `yield`), so as long as all constraints relating to a component are added at once, it should be fine.
If it's implemented incorrectly, then the result will be either:

* an unplaced component, or
* a constraint solver error

Since amateurs will be building these assemblies, issues like that will be unavoidable... I'll just try to make the error messages as pleasant as I can.

> An example would be that you have a box bottom and a lid [...]

Yes, that's one of the examples I want to use. It seems like a useful one.

> It is definitely tricky, because you additionally (probably) want to take
that original-but-now-has-a-fastener-hole-in-it box lid and export it for
manufacturing.

I agree, this solution will also offer that (actually, the code is already in place).
A `Part` is built in its local coordinates as `.local_obj`, then when `.world_coords` is set, the local object is moved to `.world_obj`.
If you try to set the value of `.world_obj` (ie: modify it), an exception is raised.
But, if you set (change) either `.local_obj` or `.world_coords`, the `.world_obj` is re-created from `.local_obj`.

fragmuffin, 2017-12-01:
I also don't like how ugly the adding of constraints is.
At the moment it takes 4 or 5 lines just to add the most basic constraints.

I think there should be "constraint helpers" as methods in an `Assembly`... but I also want people to be able to create their own constraints...
I'm sure I'll figure out a neat solution.

maybe something like:

```python
class Thing(cqparts.Assembly):
    def make_constraints(self):
        return [
            self.mk_con('lock', 'box_a', ((0,0,10))),
            self.mk_con('lock_relative', 'box_b', 'box_a__mate_top'),
        ]
```

dcowden, 2017-12-01:
a fluent builder might work well:

`return ConstraintBuilder(thing).lock('box_a',(0,0,10).connect('box_b', 'box_a_mate_top')`

fragmuffin, 2017-12-22:
First working fastener in [`tests/manual/fastener.py`](https://github.com/fragmuffin/cqparts/blob/master/tests/manual/fastener.py).

Which results in:

[image]

There's still many things to improve, but I'm very happy with this result.

dcowden, 2017-12-22:
Awesome! It's really coming along!

fragmuffin, 2017-12-23:
@dcowden @jmwright
What are your thoughts on the design?

Key classes are:
* [`Fastener(Assembly)` class](https://github.com/fragmuffin/cqparts/blob/master/src/cqparts/fasteners/base.py#L198)
* [`ScrewFastener(Fastener)` class](https://github.com/fragmuffin/cqparts/blob/master/src/cqparts/fasteners/screws.py#L49)

The documentation is in shambles in this regard, I want to work on that next.

There are many features I'd like to add, such as

* overlap with anchoring piece:
    * % penetration
    * max & min penetration
* screw / bolt selection via more human-readable metric definition (eg: `M3`)
* final piece tapped (instead of a pilot hole)

Each of these will be a parameter passed to the `Fastener` class

The next `Fastener` class I want to make is a nut & bolt situation.

I'll just make these 2 for the release, because I'm rapidly running out of time!
(not really, but I did say **this** year)

jmwright, 2017-12-23:
@fragmuffin I've been following along and the design looks good to me. I'm sure I'll have more feedback once I get a chance to use fasteners in a project. The year-end has  been pretty crazy for me, so I'm guessing it will be after this initial release that I'll get a chance to dive deeper.

fragmuffin, 2017-12-23:
Thanks @jmwright, as long as you don't see any glaring issues with it.
I'll ping this issue again when I've applied this tool-chain to the [easy install fastener example](https://fragmuffin.github.io/cqparts/doc/tutorials/fastener-easy-install.html) and published it.

fragmuffin, 2018-01-01:
For the [easy install fastener example](https://fragmuffin.github.io/cqparts/doc/tutorials/fastener-easy-install.html), I want the wood-screw & anchor variants to be added to a *catalogue* (#39).
This will illustrate that the `Selector` class can be used to either:

* isolate which wood-screw & anchor to use (from a finite list), or to
* hand-carve a wood-screw & anchor to fit the fastening situation perfectly.

fragmuffin, 2018-01-11:
Fastener concept finished, see: [easy install fastener example](https://fragmuffin.github.io/cqparts/doc/cqparts_fasteners/easy-install/index.html)

closed 2018-01-11 by fragmuffin

dcowden, 2018-01-11:
@fragmuffin, I read through the example, it's incredibly impressive work!

jmwright, 2018-01-11:
Agreed, awesome work!

fragmuffin, 2018-01-12:
Thank you @dcowden @jmwright !
Yeah, I'm quite pleased with how all that has worked out.
I just published [the last page](https://fragmuffin.github.io/cqparts/doc/cqparts_fasteners/easy-install/catalogue-fastener.html) of that example, that utilises the `cqparts.catalogue` to select fastener parts from a finite list (I think that's a bit more realistic).JSONCatalogueJSONCatalogue

dcowden, 2018-01-12:
Really cool.. and very powerful.
How do you envision people sharing and collaborating on component
collections? Would there be a central component repository?

fragmuffin, 2018-01-12:
I think a combination of:

* Contributions to the `cqparts_*` libs I've started, and
* People creating and releasing their own libraries.

Either way I think the standard way of keeping catalogues should be with the library.
Because the act of serialising, then deserialising doesn't react well to changes made to the blueprint (the `Component`).
So they should stay together.

Ideally the script used to create the catalogue should also be with the library, in case a change to the library requires the catalogue to be rebuilt.

fragmuffin, 2018-02-20:
broken link: "_easy install fastener example_" moved to [here](https://fragmuffin.github.io/cqparts/doc/cqparts_fasteners/easy-install)