cqparts/cqparts #6: wiki

fragmuffin, 2017-10-07:
Add sections to the [wiki](https://github.com/fragmuffin/cqparts/wiki)

* Parts
    * [ ] Fasteners, just the basics, and some examples (without application)
    * [ ] Gears
    * [ ] Bearings
    * (more? or is that enough for release?)
* Assemblies (examples)
    * servo motor
* Fastener Details : example usage, and some pretty pictures
    * [ ] Making your own custom fastener
        * [ ] head
        * [ ] screw drive
        * [ ] thread
    * [ ] `Evaluator`
    * [ ] `Selector`
    * [ ] `Applicator`
* (more?)

jmwright, 2017-10-07:
> (more? or is that enough for release?)

It looks fine to me.

Is it your intent to accept community contributions as well? If so, you might want to add a "contributing" section as well.

fragmuffin, 2017-10-07:
@jmwright 
you read my mind, I'm making that _issue_ now

jmwright, 2017-10-07:
We should talk to Dave about putting this repo in the CadQuery docs after the first release. As assemblies get more complex it will be important for users to be able to learn how/where to obtain parts, learn best practices for creating parts, and to learn the best ways to contribute parts to the community.

fragmuffin, 2017-10-07:
@jmwright 
Yes!, I wanted to talk to you and @dcowden about the `Part` and `Assembly` split I've made in this library.
My plan is to use it for things like
* naming,
* colouring, and
* grouping

Would that be considered in scope for `cadquery`?
(ie: do you think I should migrate the `Part` and `Assembly` classes into `cadquery`?)

jmwright, 2017-10-07:
My feeling is that the CadQuery library should stay lower level than the Part and Assembly classes. CadQuery should be at the model creation level, and we can point people to cqparts when they get to the level of parts and above. Tagging @dcowden to see what he thinks.

dcowden, 2017-10-07:
I've come to the same conclusion, but after a lot of pondering.

I think it's vitally important for there to be a well defined way to create assemblies.  This would typically be done withates, which are little more than exported coordinate systems.

I concluded that assemblies is not a cq problem for a very practical reason: performance.  assemblies may include many parts,  and even if they all defined in cq, it would be too slow.  But consider fasteners... In reality these are too expensive to render, much less to build from scratch.

I think it's best if can builds parts, and it's job ends with the production of a step file. 

It would be useful to allow a cq model to export mating coordinate systems. Im not sure if step files allow this, but if they did that would be good.

That would make it easier for another library (perhaps also based on OCC) that knows how to build and render assemblies by reading step files and  allowing the user to assemble them in terms of their mates.

fragmuffin, 2017-10-27:
@dcowden @jmwright  I agree.
This whole `cqparts` library is revolving around:
* `Part` : position & rotation, and colour / transparency
* `Assembly` : position, rotation, list of `Part` instances

so 99% of the code in this library will be core examples; things to use and build from.

I think the default orientation of an atomic `Part` should be the orientation one would expect to fabricate it.
Which reminds me, I should turn all my fasteners upside-down; I think most people would fabricate a fastener with a 3d-printer.
Also, the vector of a fastener's application would be along +Z, which seems more intuitive than -Z.

dcowden, 2017-10-27:
having created lots of models over the years, it would be great if we could arrange a way to re-orient models without forcing the designer to orient them correctly in the first place.  
We have to expect that assemblies will be made from mixes of cq-created parts and external parts-- and its very cumbersome to force everyone to re-orient their models for positioning in the assembly.

What might be slick is to provide a way for model authors to create a mating coordinate system in terms of the corodinate systems already there in the model. This could be done using CQ, via STEP import even.

for example, perhaps I could do something like mates: {
      primary: defaultCsys.rotateX(180),
      secondary: faces(type=Cylinder).coordinateSystem()
}
Anyhow, the idea is three step process:

(1) create models, export to STEP or whatever if you want
(2) create mates for each part, defined in terms of existing coordinate systems and/or geometrcy
(3) link them together, either using mates or absolute positioning

fragmuffin, 2017-11-15:
# Wiki --> Read the Docs

[readthedocs.org](readthedocs.org) looks like a much better option.
I'll work through the [getting started](https://docs.readthedocs.io/en/latest/getting_started.html) guide.

jmwright, 2017-11-15:
@fragmuffin I and @dcowden use docs embedded with the repo that are pushed to GitHub pages. The CadQuery library uses Sphinx, but I've very recently let GitHub push the docs out using its standard method with Jekyll.

Sphinx method with CadQuery:
https://github.com/dcowden/cadquery/tree/master/doc
http://dcowden.github.io/cadquery/

Here's the path I went down with embedded docs and Jekyll.
https://pages.github.com/

One nice thing about having your docs embedded with your repo is that you can manage them the same way as your code. Every time you push to master your docs are updated and pushed out to GitHub pages.

fragmuffin, 2017-11-16:
Thanks @jmwright !, I do like the simplicity of keeping the docs local.  I've recently used github pages on my recent [howto-micropython](https://fragmuffin.github.io/howto-micropython/) distraction.

One advantage that _readthedocs_ has is to document multiple versions from their labels... but that's something I can take advantage of later if I choose.

fragmuffin, 2017-11-16:
@jmwright yep, this is great... I've got documentation going, how I've just got to learn how to populate it... but I think I know enough to get started.

I mainly wanted to do this now so I didn't have a huge "document everything" task just before release, because in my experience, that leads to the worst documentation.
I may still have to rearrange the content so it's not a big mash of ideas and designs, but that's to be expected.

I'm very happy with this, thanks for the advice!

oh!, I'll also see if I can get it automatically linking to `cadquery` objects when they're referenced.

jmwright, 2017-11-16:
> I mainly wanted to do this now so I didn't have a huge "document everything" task just before release, because in my experience, that leads to the worst documentation.

I absolutely agree. If you document during the process, you magically end up with a full set of documentation that has detail and clarity in it that you usually don't get if you wait until the end. I think that some rework and restructuring of the docs at the end is worth that.

fragmuffin, 2017-11-17:
I imagine people browsing these docs will be regularly switching between `cqparts` and `cadquery` documentation.
So I've decided to use the same theme as `cadquery` (it's a good theme), but I'll change the colours so it's still clear which page they're looking at.

so side-by-side they look like:
[image]

I'm not sure if I'm going for red, but you get the point

fragmuffin, 2017-11-19:
Regarding sphinx autodoc of `ParametricObject` parameters...

With the following code in the sphinx `config.py` file:

```python
def process_docstring(app, what, name, obj, options, lines):
    try:
        if issubclass(obj, cqparts.params.ParametricObject):
            new_lines = [
                '.. warning::',
                '',
                '    This class is a ``ParametricObject``',
                '',
                '    TODO: list parameters for constructor input.'
                '',
            ]
            for (i, new_line) in enumerate(new_lines):
                lines.insert(i, new_line)
    except TypeError: # raised by issubclass if obj is not a class
        pass

def setup(app):
    # Parameter Mapping
    app.connect("autodoc-process-docstring", process_docstring)
```

_sphinx_ rendered html has a big warning at the top of any class inheriting from `ParametricObject`...
excellent :smiling_imp: 

now to populate with a list of it's parameters, and to make it informative.

dcowden, 2017-11-19:
That's a pretty beat trick...

dcowden, 2017-11-19:
^neat. Dang autocorrect!

fragmuffin, 2017-11-20:
@dcowden 
Yeah! gotta love automated documentation tools dev'd by people that know their audience :+1:

I've committed a 1st draft for this solution: [`cqparts.utils.sphinx.add_parametric_object_params`](https://github.com/fragmuffin/cqparts/blob/master/src/cqparts/utils/sphinx.py#L39)

I've built it into the `cqparts` library because it's something people using `cqparts` to make their own parts may want to use.
So that method is also documented by sphinx... very meta

it's working well enough... it's difficult to figure out exactly where I want people to put their documentation.
the `:type {name}: {doc}` content is part of the class
but `:pararm {name}: {doc}` is specific to the parameter's instance, so it's added to the `Parameter` constructor.

and particularly difficult when it comes to [`HeadType`](https://github.com/fragmuffin/cqparts/blob/master/src/cqparts/fasteners/base.py#L66) used for `Fasteners`

**Fasteners Refactor Imminent**
oh!, and all of this has made me realise that I've got the whole "fastener" terminology at the wrong level!.
A `Fastener` is an `Assembly`, because in the case of a _nut_ and _bolt_, the combination of the two **is** the `Fastener`.

So I think what I've called a `Fastener` up to this point is actually a component of a fastener, perhaps terminology is separated by:

* _male_ : defined by an external thread
* _female_: defined by an internal thread

But a fastener can have many of each, or none of either (eg: a rivet is still a valid fastener).

Anyway, that's a discussion for #4 or #5. but I think I've got a major refactor in my future.

dcowden, 2017-11-20:
@fragmuffin cool stuff!  To add even a little more confusion,  if fasteners are a feature, it may also include the hole that was removed from the mating parts as well. (After all, a lot of the time the holes are the things we end up needing most for fabrication....

fragmuffin, 2017-11-20:
@dcowden yes!
That's exactly what I'm aiming for, to "apply" a fastener to one or more parts. Which will remove, or add material accordingly.
I'm almost there, very exciting.

fragmuffin, 2017-11-25:
sphinx docs are stable for now, and ultimately this is a never-ending task, so this issue is closed.

closed 2017-11-25 by fragmuffin