cqparts/cqparts #7: Servo Motor(s)

fragmuffin, 2017-10-07:
Create servo motor `Assembly` (at least one, made parametric)

Parameters should include:
* Mounting points:
    * locations
    * hole diameters
    * flange fillet radius
* Motion range (30 <= range < 360)
* shaft diameter
* shaft key
* (more?)

blooop, 2017-10-26:
I'm not sure the best way to represent this, but some indication of where the data/power socket is.

Eg:  http://www.trossenrobotics.com/dynamixel-mx-28-robot-actuator.aspx

The cables are often quite difficult to plug in when the servo is mounted and it would be nice to have a cad representation of how accessible the plugs are.

fragmuffin, 2017-10-27:
That's a good point @blooop.
This would definitely be a good future improvement.
The target servo just has wires hanging out, but the servo's `Assembly` should have a sense of electrical connectivity. Which may eventually lead to calculated cable lengths.

As for the servo you linked, the connector would probably come from an import of [KiCad](https://github.com/KiCad/kicad-packages3D).

I'll keep all this in mind while designing it 👍

jmwright, 2017-10-27:
Not sure if this helps, but CadQuery community member @easyw has created a [workbench](https://github.com/easyw/kicadStepUpMod) for integration of electronics and mechanical systems in 3D. I'm not sure if it defines electrical connector interfaces though.

I think that we'll probably want interfaces and constraints defined in CadQuery code somehow, but I always like to take a look at existing open source work when talking about things like this.

fragmuffin, 2017-10-27:
@jmwright 

> I always like to take a look at existing open source work when talking about things like this

That sounds like a good habit to get in to :smile:

I wouldn't want this library to dive too deeply into the electrical side of things, especially because it already seems to be done so well. (@easyw, that quad-copter looks amazing!)
(I've added #19 to track that aspect)

For something like a servo, the `Assembly` instance could contain a list of connections.
But I think this should be user code; an adaptation of `cqparts`... for example:

```python
# library code
class Assembly:
    pass

# User Code
class ElectronicAssembly(Assembly):
    wires = []

class MyServoMotor(ElectronicAssembly):
    wires = ['GND', '+5V', 'SERVO_SIGNAL']
```

That's quite crude, but it's just illustrating my point.
I'd like to build that into an example, but initially make it just that; an example usage.

fragmuffin, 2017-10-27:
Oh... major point I've neglected to mention in all this...

This `ServoMotor` assembly is a functional example of the `cqparts` library.
It's a case-study I'll be implementing, mainly to show that everything works, and to demonstrate how to use it.

This will **not** be the root of all future servo-motors to use as an implementation base.

fragmuffin, 2018-03-02:
Removed from initial release project...
I'd still like to do this, but I need to release this before I go mad.

zignig, 2018-05-17:
once dc motors are settled I will hit this.