cqparts/cqparts #19: Wrap 3rd party assemblies

fragmuffin, 2017-10-27:
Wrap other's realizations of an "assembly" into a `cqparts.Assembly` instance.
(and the same for a `cqparts.Part` for atomic models)

**Use Cases**
Try wrapping models from:
* https://github.com/easyw/kicadStepUpMod
* https://github.com/KiCad/kicad-packages3D
* imports of standard model formats.
* (more?)

jmwright, 2017-10-27:
Have you defined what the cqparts.Assembly object might look like yet?

jmwright, 2017-10-27:
Please ignore my last question. Answered in #7

fragmuffin, 2017-10-27:
@jmwright no problem... it's currently defined in [part.py](https://github.com/fragmuffin/cqparts/blob/develop/src/cqparts/part.py), but it's just conceptual code at this point... I've mostly been working on the "types" folder that's filled with helper scripts to build things like screw heads, screw drives, and threads.

jmwright, 2017-10-27:
It seems like there would be two separate types of assemblies.

1. The wrapped assembly as you've mentioned above.
2. A CadQuery native assembly that defines the constraints in a script somehow.

Is that how you see it?

fragmuffin, 2017-10-27:
@jmwright yes.
Although, you could argue these are the same thing:
The wrapped assembies inherit from `Assembly`, which inherits from `ParametricObject` (they're quite neat).
So an assembly could inherit from a wrapped assembly and have parameters added to it, so it could be a customised further.

There are very few limits at this point, which is awesome.

jmwright, 2017-10-27:
@dcowden Have a look at the [ParametricObject code](https://github.com/fragmuffin/cqparts/blob/develop/src/cqparts/params.py). I hadn't seen that until @fragmuffin brought it up.

dcowden, 2017-10-27:
ok, will do

fragmuffin, 2017-10-28:
@jmwright @dcowden each `Parameter` can be heavily customised, see [fasteners/base.py](https://github.com/fragmuffin/cqparts/blob/develop/src/cqparts/fasteners/base.py) for an example

dcowden, 2017-10-28:
Cool, that's a very complete parameter framework you have.

How could I integrate it or something similar into cq_cmd, I wonder.

Cqgi needs a better way for script authors to prededine the types of the paremeters. Right now it infers the types from the default values, but there is now way to indicate, for example, a range .

I don't want script authors to have to extend a base class or define a parameter metadata file (which I did in early implementations).  What I iterated to I think is supporting comments on the line right above the variable declaration. 

I could replace my basic parameter stuff inside of cqgi with an improved version more like yours

fragmuffin, 2017-10-28:
Thank you @dcowden.
You're welcome to copy it over if you have a use for it.