cqparts/cqparts #20: Springs

jmwright, 2017-10-28:
A question from a KiCAD community member sparked this issue. As the name implies, should springs be part of this collection?

fragmuffin, 2017-10-28:
@jmwright yeah!, I can do springs.

There should also be an option for simplified rendering, perhaps 3 stages:
* full quality helix
* hollow cylinder
* solid cylinder

But this is something I need to cover generally... still not sure how to define an overall model quality.
I was thinking of setting some sort of "polygons per mm^3" parameter.

The main challenge with springs is creating a tapered helix... that is, a helix with a changing pitch.
Do you know how to generate this?... the only way I can think of is to manually create it as a 3d spline.
More generally, there should be a way to generate a function

jmwright, 2017-10-28:
> I was thinking of setting some sort of "polygons per mm^3" parameter.

Are you talking about setting the quality of a tesselation?

> The main challenge with springs is creating a tapered helix... that is, a helix with a changing pitch.

The 3D spline should be a good option. Another method for a decent approximation might be to stack helixes of multiple pitches on top of each other. It would take some decent math built into the function, but I think it should be doable. This method might be too slow at higher quality settings though.

> More generally, there should be a way to generate a function

Do you mean generate a function to create geometry, or something else? How would that work?

fragmuffin, 2017-10-28:
@jmwright 

> Are you talking about setting the quality of a tesselation?

I wanted to have a generalised concept of what to "make" when a `Part` or `Assembly` object is displayed or exported...
* if the desired quality is _below_ a threshold: a _cylinder_ is returned (a simplified model)
* if the desired quality is _above_ (or equal) to a threshold: a circle is swept through a helix to create the full complexity model.

But this model quality factor would be dependent on the scale of the part.

**For example:** one giant spring could be holding up something with hundreds of tiny springs as part of it's assembly... when rendering the whole thing (simply for display purposes), one would want:
* the tiny springs to be rendered as cylinders, while
* the larger spring should be the highest quality.

> Do you mean generate a function to create geometry, or something else? How would that work?

For example, a [conical spiral](https://www.wolframalpha.com/input/?i=conical+spiral) is another curve that would be used to create fully collapsible springs.
One way to represent this is:

```
a = 1; r =1
x(t) = r * t * cos(a * t)
y(t) = r * t * sin(a * t)
z(t) = t
```

I guess we've already discussed how to do this;  to have `x(t)`, `y(t)` and `z(t)` as functions, and give a range of `t`, and the number of points to be represented as a spline (linearly spaced).

```python
def wire_points_generator(fx, fy, fz, t1, t2, steps):
    step_size = (t2 - t1) / steps
    for i in range(steps):
        t = i * step_size
        yield (fx(t), fy(t), fz(t))

from math import sin, cos
(a, r) = (1, 1)
conical_spiral_points = list(wire_points_generator(
    fx=lambda t: r * t * cos(a * t),
    fy=lambda t: r * t * sin(a * t),
    fz=lambda t: t,
    t1=0, t2=1, steps=100,
))
```

jmwright, 2017-10-28:
I've never tried to make a 3D spline with FreeCAD, but it sounds like it's possible.

https://forum.freecadweb.org/viewtopic.php?t=12032

I wonder if it would work with your wire_points_generator setup as-is.

fragmuffin, 2017-10-29:
Yeah, I'm sure it is possible, thanks for the link!

I think it would be great to have something that intelligently used more points for increased accuracy.
For example, for a parabola:

```
x(t) = t
y(t) = t ** 2
z(t) = 0
{-5 <= t <= 5}
```

you'd want more points around to `t=0` because the [parabola's curvature](https://www.wolframalpha.com/input/?i=curvature+y+%3D+x%5E2) is greatest there.
Or perhaps it should be based of the [rate of change of curvature](https://www.wolframalpha.com/input/?i=derivative+2+%2F+((4x%5E2+%2B+1)%5E(3%2F2))).

I'll ponder this when I'm making springs :wink:

fragmuffin, 2018-02-20:
Removed from the release project, but still relevant.