cqparts/cqparts #21: Automated Collection Consumption

jmwright, 2017-11-03:
I've been experimenting with some GUI concepts including an automated loader for third-party collections like this, so that's where this question is coming from. This repo is the first of its kind, so it can help shape the standard(s) for future collections.

If something like a GUI (i.e. CadQuery FreeCAD module) or a web service wanted to consume this collection, how would it go about doing that? Will there eventually be an `__init__.py` that lays out what parts are available and the path(s) to them within the repo? 

Another question that's not as critical:
How would a GUI or service go about rendering/displaying thumbnails of all the scripts and/or part variations for users to skim through quickly? Should there be a `thumbnails` directory that holds png images of the results of the scripts in the collection which is built by a CI system each time changes are pushed? Is it even possible to show all the variations of a scripted part? Probably not, and in that case should it just show a representation of the type of "thing" the script produces? Alternatively, should the users just rely on the part/script names to know what result a script/part will give them?

fragmuffin, 2017-11-03:
Sounds interesting @jmwright !

**Registering Parts / Assemblies**
This is all taken care of with:
`from cqparts.search import register_class, index, class_list`
https://github.com/fragmuffin/cqparts/blob/develop/src/cqparts/search.py

so an external library can "register" their `Assembly` with keywords
```python
import cqparts
@cqparts.register_class(
    type='motor',
    current_class='dc',
    part_number='ABC123X',
)
class SomeMotor(cqparts.Assembly):
    def make(self):
        return {}  # build assembly content

motor_class = cqparts.find(part_number='ABC123X')
motor = motor_class(shaft_diameter=6.0)
```

your library could find all registered classes (both inside and outside the `cqparts` library) by looping through: `from cqparts.search import class_list`

Although, I've just noticed that there's no way to see what criteria an `Assembly` was initially registered with...
ok, updated that... when registered, the search criteria is for that class is added to the `class_criteria` dict in the same `search.py` module.

**Thumbnails**
Each `Assembly` / `Part` should have a default set of parameters to build it.
So every registered class could be built with no parameters to create a "default" model.
This is the way everything defined within `cqparts` will work, but that's almost impossible to enforce in others' implementations.

So creating thumbnails should be quite straight forward.

You could let me know what parameters you use to frame the thumbnail, so I may put some default parameters in the `Assembly` and `Part` classes that may be overridden if people want to control what their thumbnail will look like.

Cached thumbnails would definitely be needed for performance client-side.
At the moment there are very few restrictions on people's use of `cqparts`, they can create parts wherever they want in their own codebase, so I'd like to keep that freedom.
Perhaps each class can have a thumbnail image filename as part of the class, perhaps the default can be a _thumbnails_ directory in the same directory as the assembly.
What are your thoughts?

jmwright, 2017-11-03:
> when registered, the search criteria is for that class is added to the class_criteria dict in the same search.py module.

Does that mean that my GUI will be able to tell a single part from an assembly, or does that mean something else?

> You could let me know what parameters you use to frame the thumbnail

I'm not that far yet. I was mainly interested in the feasibility of doing thumbnails so that it could influence my experiments.

> Perhaps each class can have a thumbnail image filename as part of the class, perhaps the default can be a thumbnails directory in the same directory as the assembly.
What are your thoughts?

Having the thumbnails included would lower the computational cost for the client, but makes the collection's codebase more messy. Maybe just lazy loading/executing/rendering with cache on the client side is the way to go. Having default parameters for each part our assembly would be very helpful though.

fragmuffin, 2017-11-03:
> Does that mean that my GUI will be able to tell a single part from an assembly, or does that mean something else?

`cqparts.Part` and `cqparts.Assembly` are just classes, when they're inherited by a specific implementation, they're still just classes (ie: you can't even define a bounding box from the class alone). When the class is instantiated, it gets it's parameters (such as `diameter`, `wall_width`, and so on).
When the instance is "made" (with `make()`, or by getting `some_instance.object`) it's created, and only then can you determine what `Part` instances have been created inside an `Assembly`, or nested `Assembly`.
All of this will be much clearer when close #7 

> I'm not that far yet. I was mainly interested in the feasibility of doing thumbnails

Oh!, that's fine... I just meant when you're there.
I understand it's no small job, but I'm very interested.

> Maybe just lazy loading/executing/rendering with cache on the client side is the way to go

It would be a good start, I see no reason why having thumbnails generated at library deployment couldn't be supported _after_ the initial concept is released.
That would make a 1st version easier.
Client-side thumbnail rendering should always be supported anyway, because we can't enforce people's usage of `cqparts`, or their deployment of their own libraries.

> Having default parameters for each part or assembly would be very helpful though

No problem... I'll need a naming convention at some point.
* `Part` or `Assembly` specific parameters should always be simple, and intuitive (such as the aforementioned: `diameter`, and `wall_width`)
* default, and overridable parameters that support library behaviour, or 3rd party applications (such as your _thumbnail_ generator) will need parameters defined at the same scope, but should have a prefix/suffix... perhaps a classic `_` prefix would be enough (created #23 to deal with this)

jmwright, 2017-11-04:
I'm pushing us into scope creep on this issue, but how will sub-assemblies work? For instance, if a user is placing an alternator assembly in a car assembly? Is that beyond the scope of what cqparts is going to define?

fragmuffin, 2017-11-04:
No, that's not scope creep at all.
The way I see it is: `Part` is to `Assembly` as _file_ is to _folder_; so what you're asking is: "can a folder contain another folder".
Yes!, absolutely, there's no reason it can't.

In my random brainstorming on the train, I put this together as a concept for the servo-motor

```
servo
 ├─ enclosure
 │   ├○ shell
 │   ├○ end_cap
 │   └─ fasteners
 │       ├─ front_left
 │       │   ├○ bolt
 │       │   ├○ washer (?)
 │       │   └○ nut
 │       ├─ front_right
 │       │      ...
 │       ├─ back_left
 │       │      ...
 │       └─ back_right
 │              ...
 ├─ gearbox
 │   ├─ bearing
 │   │   ├○ outer_ring
 │   │   ├─ rolling_elements
 │   │   │   ├○ ball_1
 │   │   │   │  ...
 │   │   │   └○ ball_n
 │   │   ├○ retainer
 │   │   └○ inner_ring
 │   ├○ main_shaft
 │   ├○ middle_shaft
 │   └○ drive_shaft
 ├─ motor
 │   ├─ base_bearing ...
 │   ├─ drive_bearing ...
 │   ├─ rotor
 │   │   └○ shaft
 │   └○ stator
 └─ driver
     └○ pcb
```

where:
* `├○` denotes a `Part`, and
* `├─` denotes an `Assembly`

```python
servo = ServoMotor(shaft_diameter=3, coupling='double_hex')  # and so on.
print(servo.tree_str())  # prints the above list

# Furthermore, each sub-component can be found using it's name in the tree
gearbox_bearing = servo.find('gearbox.bearing')
gearbox_shafts = servo.find('gearbox.*_shaft')  # will return a list
```

All of this will be documented once it's functional.

fragmuffin, 2017-11-04:
I'll close this topic when I've linked to an appropriate wiki page where the following is documented:
* Part / Assembly searching
* Listing all parts / assemblies
* Searching within an assembly

jmwright, 2017-11-05:
```
part = file
assembly = folder
```
I like the flexibility that provides. It allows for arbitrary layers of sub-assemblies.

From the GUI standpoint, when it opens the servo example above, what's the entry point? Where does it start in rendering the entire assembly?

fragmuffin, 2017-11-05:
> Where does it start in rendering the entire assembly?

It's built from the top down, but actually, the leaves are the first to get structure.
So I guess it's more accurate to say:
* Parameters, and their compound effects filter downward
* `cadquery.Workplane` instances are built from the bottom up
   (because they can only be built when all their parameters are known)

It sounds more complicated than it is, building them should be quite intuitive.
I can't wait to get back to work on all this.

jmwright, 2017-11-05:
Thanks for the explanations. I'm also wondering where the positioning and/or constraints for placing the parts and sub-assemblies will live. @RustyVermeer has used a simple JSON method in Jupyter Notebooks which just sets rotation and translation. At some point we'll want to have constraints as well. Where will those reside for parts and sub-assemblies?

fragmuffin, 2017-12-19:
Hi @jmwright ,
I just realised that I can close this now:

* **Part / Assembly searching** and **Listing all parts / assemblies**: 
  https://fragmuffin.github.io/cqparts/doc/api/cqparts.html#module-cqparts.search
* **Searching within an assembly** 
  https://fragmuffin.github.io/cqparts/doc/api/cqparts.html#cqparts.part.Assembly.find and
  https://fragmuffin.github.io/cqparts/doc/api/cqparts.html#cqparts.part.Assembly.tree_str

closed 2017-12-19 by fragmuffin