cqparts/cqparts #24: Logo

fragmuffin, 2017-11-17:
Make a logo

### Criteria:

* can be built with `cqparts`
* can be animated (I'm a fan of the simple yet flashy [webpack](https://webpack.github.io/) logo)
* represents what the library does
* as simple as possible, yet distinguishable

I like the idea of the logo being:

* a symbolic representation of what the library does
* a functioning example
* a demonstration of it's potential, (ie: it's animation)
    * export to web-compatible format (something importable by [three.js](https://threejs.org/) and [a-frame](https://aframe.io/))
    * embed animation data (eg: times, transforms, and transition types) in an `Assembly` implementation.

fragmuffin, 2017-11-17:
1st draft:
[image]

using the "A" compass from the `cadquery` logo, to illustrate it's building onto `cadquery`

@dcowden: would you mind if I used the drafter's compass from the `cadquery` logo?

**Problems:**
* The shapes (without surrounding letters) spells "AR", which means nothing
* I want the "R" block to be fastened to "A" with a screw, but that interferes with the "R", and would be very visually cluttered.

fragmuffin, 2017-11-17:
or something like.
[image]

where
* "cq" is a blue cube.
* "parts" is a red rectangular block.
* they're fastened together (perhaps a horizontal bolt through everything)
* animated by removing the bolt, then spinning each block around 360deg, then re-fastening the bolt.

anyway, I'm getting ahead of myself... I'm going to do something more useful now

jmwright, 2017-11-17:
@fragmuffin The CadQuery logo is in flux right now. The original was made with a logo generator, and there's no real source material available. I've created a [temporary logo](https://github.com/jmwright/cqcad/blob/master/cqcad/content/images/cadquery_logo.svg) for my experimental GUI. I haven't found a font that recreates the hand drawn feel while still being readable yet, so I've just used a generic plain font for now. The main thing I wanted was something that I could use as an application icon or favorite icon on a website, and that's what the compass has ended up being.

dcowden, 2017-11-17:
I'm ok with anyone using the cq logo, but I really don't like it a lot. I
think maybe I need to figure out a better one.

fragmuffin, 2017-11-17:
@jmwright @dcowden ok then, I won't base anything off the compass design unless that changes.
I'm liking the flexibility of the blue and red logo above.

When I get around to animating it, I can have the `cadquery` logo on the back of the blue cube, then spin it around briefly to show it, whatever it may be.

**Colour?** (Australian spelling :wink:)
The colours could be anything, they're currently based on the [documentation colours](https://github.com/fragmuffin/cqparts/issues/6#issuecomment-345167140).

Are you inclined to have a base colour for your documentation / logo?...
I ask because it would be good for me to have that colour in my logo... because `cqparts` would be nothing without `cadquery`, so having that reflected in the logo seems apt.

This feels like such a trivial thing to spend time on.
I have to keep reminding myself that it is actually quite important.

jmwright, 2017-11-17:
On the FreeCAD workbench I just have "CQ" as the icon because I didn't know what else to do at the time. What if the blue box (or whatever color it ends up being and whatever style it's done in) becomes the CadQuery logo? Then projects like cqparts and cqcad can just bolt whatever their red box is onto the CadQuery logo That gives a continuity to the CadQuery ecosystem, but would probably require some logo usage guidelines to be created at some point. Thoughts?

dcowden, 2017-11-17:
i agree, lets go with something like that cq is a relatively appealing
character combination, as is @fragmuffins' split box logo.

for now lets say the 'official' cq logo is lowercase cq in Roboto bold
font, with a box the same color as the cq docs (#2980B9)

[image: Inline image 1]

or rendered in white with a 2px border
[image: Inline image 2]

jmwright, 2017-11-17:
@dcowden The images didn't show up in your message.

dcowden, 2017-11-17:
sorry about that, i guess email pictures dont work...
https://docs.google.com/presentation/d/1pAg2WCOPe6TOGeTenuc5eKMohiGgZfMmJks3BoNYtmE/edit?usp=sharing

jmwright, 2017-11-17:
That's got a kind-of periodic table vibe to it.

dcowden, 2017-11-17:
yeah it does-- i thought the same thing. which means we're both the same geek wavelength. but i ended up thinking that was a good thing.  We could consider going all-in on the vibe by making it uppercase C

jmwright, 2017-11-17:
If we push too far with the periodic table feel, will people get confused and think that CadQuery is a chemistry thing? It could absolutely be used by chemists, but it's a CAD scripting language.

dcowden, 2017-11-17:
that's a good point. lets stick with this then.

jmwright, 2017-11-17:
@dcowden Can I go ahead and create vector versions of the light and dark variations of your mockups? I can push them to the CadQuery repo. You can do PNG exports after that if you want, but almost all browsers support direct SVG rendering these days.

@fragmuffin I'm guessing that you're in bed due to the time zone difference, but if you're ok with all of this I'll probably plan on following suit with you on building off the base CQ logo. Let Dave and I know what you think.

dcowden, 2017-11-17:
@jmwright absolutely feel free!  choose another font if you like it better-- just go with what looks good.

jmwright, 2017-11-17:
I put the logos in the [doc/_static/logo](https://github.com/dcowden/cadquery/tree/master/doc/_static/logo) directory.

I also included the needed Roboto bold font, and the Apache license that goes with it.

jmwright, 2017-11-17:
Sorry, that broke the doc build process. The logo materials are [here](https://github.com/dcowden/cadquery/tree/master/logo) now.

jmwright, 2017-11-17:
Sigh. I'm not sure what's going on. It seems like something's actually changed with travis-sphinx. The nowarn option isn't available anymore or something. I'll figure it out here after bit.

fragmuffin, 2017-11-18:
Hmm. I'm very confused, the [preview on github](https://github.com/dcowden/cadquery/blob/master/logo/cadquery_logo_dark.svg) looks wrong:
[image] (screenshot)
the off colour is confusing me the most, I have no idea why that would be.

But if I install the font, and view the SVG, it looks correct:
[image] (also screenshot)

Could you avoid using a font, and convert the font to curves before exporting to svg?
(but still credit the font)

Also, here's the `cqparts` logo now.
[image]

but... why am I suddenly so hungry...

[image]

s**t, I may pick another colour ;)

here's some other options.
[image]
All of these have the same _saturation_ and _value_ as the `cadquery` logo colour `#2980b9`, just the _hue_ is changed.

I like the look of the all blue one, but the colour distinction is not obvious... so I think I'll go for the green (the shade with the `*` on it)

What do you think @jmwright @dcowden ?

jmwright, 2017-11-18:
Ok, last update. travis-sphinx 2.0.0 was pushed out 6 days ago and broke some things. I've got it fixed now. Here's the final location of the logos.

[doc/_static/logo](https://github.com/dcowden/cadquery/tree/master/doc/_static/logo)

jmwright, 2017-11-18:
@fragmuffin Here's what the GitHub preview looks like for me.

[image]

I think the logo you've selected with the green looks fine. If you settle on that I'll choose one for cqcad that follows the same pattern but has a different color.

fragmuffin, 2017-11-18:
@jmwright that fixed the background colour, but I'm still seeing a generic font.
The letters can be easily converted in _inkscape_ by selecting the text then `Path > Object to Path`. This will remove any font requirements of the client.

Now I'm leaning toward:
[image]
It looks very nice in the documentation with a dark grey background.

But I still think the red looks good, and I don't care so much that it looks like the dominos logo.
Do you think it's better to have **colour** on the suffixed word `parts` / `cad`... or it doesn't matter

ok, I've officially thought about this too much, and I need to come back to it again later.

jmwright, 2017-11-18:
Weird, the font renders correctly for me. Do you use Firefox by any chance? Maybe Chrome has the Roboto font built in.

> The letters can be easily converted in inkscape by selecting the text then Path > Object to Path.

Yeah, I was trying to avoid creating something that is harder to modify though. I guess it really doesn't matter. The font and colors are there. If I also document the font size used in Inkscape it's pretty trivial to recreate the logo. I'll try to do that tomorrow.

> Do you think it's better to have colour on the suffixed word parts / cad... or it doesn't matter

I think that colors are fine. We can use colors to differentiate between projects in the CadQuery ecosystem.

fragmuffin, 2017-11-18:
@jmwright 

> Do you use Firefox by any chance?

I don't habitually, but I do have it (I typically use chrome)
[tries the link in Firefox]
oh yeah!, there it is!... chrome is doing a weird thing

> I was trying to avoid creating something that is harder to modify though

I respect that sentiment, but I think it's going to be quite difficult to get a font to render correctly in an SVG on all clients.
My first google found [this article](https://github.com/JChemPaint/jchempaint/wiki/The-svg-font-problem-and-its-solution), which is a means to embed a font within an SVG... but it apparently doesn't work on Firefox... I find that a little bit funny.

In the long run, I'd love it if I could get `cqparts` to export an svg render... I love it when logos are build with the tools they represent (eg: Matlab).

> I think that colors are fine. We can use colors to differentiate between projects in the CadQuery ecosystem

ok, I think I'll probably go with the red, even though it makes me want pizza.
But I'm still going to sit on it for a few days before I commit to it.

I think it's between these two:
[image]
[image]

jmwright, 2017-11-18:
@fragmuffin Please try the [logos](https://github.com/dcowden/cadquery/tree/master/doc/_static/logo) now. I converted the text to paths and the saved separate Inkscape files in case we want to tweak the logos later.

fragmuffin, 2017-11-18:
@jmwright Yep, that's fixed it :+1:

fragmuffin, 2017-11-18:
@jmwright 
I also just noticed, the background of the light logo is transparent.
So on a light grey background...
[image]

But in inkscape this isn't obvious.
Alternatively, with a white background it would look like:
[image]

I'm not saying the transparency looks bad, I'm just confirming that the transparency is what you intended;
I'm actually quite bad at doing visual fine-tuning. (I'm much more of a "function over form" kind of guy)

jmwright, 2017-11-18:
@fragmuffin Good catch. I think we should enforce the white background on the light logo. @dcowden thoughts?

dcowden, 2017-11-19:
yah, i think the white BG is intended. either white bg or blue bg.

jmwright, 2017-11-19:
Ok, I've fixed the background so that it will always be white.

jmwright, 2017-11-19:
The dark icon looks pretty nice in the Ubuntu Unity dock.

[image]

dcowden, 2017-11-19:
Yeah that does look nice!
I'm glad we went away from the compass thing

fragmuffin, 2017-11-19:
Yeah, I'm very pleased with the way it "stacks" with other modules; very clean.

@jmwright the white background looks good, although the [cadquery_logo_light.svg](https://github.com/dcowden/cadquery/blob/master/doc/_static/logo/cadquery_logo_light.svg) image doesn't have the text converted to paths (I'm seeing the old changed font problem)

fragmuffin, 2017-11-19:
It's also quite easy to CSS style: https://codepen.io/FraggaMuffin/pen/yPpKRp

jmwright, 2017-11-19:
Sorry, I forgot to convert the text to a path when I saved it as plain SVG. I'll fix it when I'm at a computer later.

jmwright, 2017-11-20:
I converted the text to paths in the light logo.

fragmuffin, 2017-11-21:
ok!, I've slept on it now, I'm going with _grey_.

When I build the logo using `cadquery` I'll give the _parts_ block a thread that screws into the _cq_ block. The _grey_ look will go well with the concept of a fastener (which will be one of the more powerful parts of `cqparts`)

**Dark**
[image]

**Light**
[image]

Thanks for all of your input @jmwright & @dcowden !

closed 2017-11-21 by fragmuffin