cqparts/cqparts #25: Publish Docs

fragmuffin, 2017-11-19:
* [ ] commit the `_build` directory.
* [ ] publish docs with _github pages_

jmwright, 2017-11-23:
How long do you estimate until the release? I was just going to look through them periodically as they progressed.

fragmuffin, 2017-11-23:
There's a lot to get through, but I really want to release before the new year.
However, that may mean I don't make as many sample parts as planned.

fragmuffin, 2017-11-24:
@jmwright 
I think you're right; I'll publish the docs in the next 10hrs or so (got to sleep now)
It's not a difficult task to do, and it will make it easier for you and Dave to critique my designs & instructions.

jmwright, 2017-11-24:
Sounds good. It will also help me as I'm building out the experimental GUI. I'm trying to think through how constructs like parts and assemblies fit in from a UI/UX perspective, both in pure scripting and using mouse interactions (and the hybrid combination of the two). cqparts is factoring heavily in the design philosophy that I'm trying to sort out.

fragmuffin, 2017-11-24:
@jmwright 
That sounds very interesting!, I can't wait to see this all fit together.

* [ ] I'll put a link to the constraints tutorial here before I close this issue

that should help you see what I'm planning.

fragmuffin, 2017-11-25:
@jmwright
I've published it using the `docs` folder in the `main` branch... 
However, I'd like to use the `gh-pages` method in future, I just don't have time to fiddle with that today.

Aaaaand, that hasn't worked, for some reason none of the `_static` files have been published...

fragmuffin, 2017-11-25:
got it!, the `.nojekyll` was at the wrong level... working on the constraint tutorial

fragmuffin, 2017-11-25:
@jmwright 
a start to the `Constraint` documentation is here:
https://fragmuffin.github.io/cqparts/doc/parts/constraints/index.html

I'm re-thinking the "return a dict" initial idea for `Assembly.make()` implementations, but it's a start.

closed 2017-11-25 by fragmuffin

jmwright, 2017-11-25:
Awesome, I've got the docs open now.

fragmuffin, 2017-11-25:
Almost every page is half finished, and TODO's riddled throughout, but it's a start :wink:

jmwright, 2017-11-26:
I think all the TODO's are fine. I like to evolve docs as the project evolves.

Some minor fixes in the docs so far. P = Paragraph, S = Sentence
* [Pysical Analogies->Exceptions to the Rule](https://fragmuffin.github.io/cqparts/doc/parts/index.html#physical-analogies) P3;S2 - "you may want it’s " should be "you may want its" I think.
* [Constraints](https://fragmuffin.github.io/cqparts/doc/parts/constraints/index.html#constraints) P1;S1 "relative to it’s parent" should be "relative to its parent" I think.
* [Constraints->Types of Constraints->Lock](https://fragmuffin.github.io/cqparts/doc/parts/constraints/index.html#lock) P1;S1 - " it’s parent’s origin:" should be " its parent’s origin:" I think.
* [Make Your Own Assembly->Design Considerations](https://fragmuffin.github.io/cqparts/doc/tutorials/assembly.html#design-considerations) Bullet Point 2 - The wording is a little confusing there. It's almost like there should be a dash or colon between "change mounting points" and the rest.
* Bullet point 3 has the same issue as Bullet Point 2 from the last item.

Questions:
* [Fasteners->Definition](https://fragmuffin.github.io/cqparts/doc/parts/fasteners/index.html#definition) - In the my_bolt_fastener example, what happens if a user just wants the bolt and the washer by the head to go in a threaded hole in a plate? Is that just a contrived example and I'm overthinking it?
* [Constraints](https://fragmuffin.github.io/cqparts/doc/parts/constraints/index.html#constraints) You reference base.Mate here. Are you going to change that to base.Constraint?
* [Fastener:Easy Install->Using the Fastener](https://fragmuffin.github.io/cqparts/doc/tutorials/fastener-easy-install.html#using-the-fastener) - The code there uses `from cqparts import Part`. I just wanted you to be aware that FreeCAD has a Part module that could conflict with that if cqparts users are employing FreeCAD internals too. It's not likely, but I've pulled in the Part module before because of something CQ wouldn't do. Not sure it's worth changing your module name, just something to be aware of.
* [Servo Motor->Display](https://fragmuffin.github.io/cqparts/doc/examples/servo.html#display) - What does `display` do? How does that interact with the graphical environment that cqparts is running inside?

fragmuffin, 2017-11-26:
@jmwright 

regarding: "it's" vs "its"
Oh boy, you're right, I've been justifying those as possessive apostrophes for years!
I've made thousands of grammar errors! thank you for helping me avoid making thousands more :+1: 

> what happens if a user just wants the bolt and the washer by the head to go in a threaded hole in a plate? Is that just a contrived example and I'm overthinking it?

You're definitely not overthinking it. I think this is still a valid `Fastener` variant.

If the parts being fastened already have their alterations (a hole in one part, and a tapped thread in another), then the `Fastener` would merely serve as an `Assembly`. I would encourage people to still use a `Fastener` instead of an `Assembly` to distinguish the components that serve to mechanically hold everything together. However, they may need to apply the constraints themselves.

But it would be much better design to leave the 2 parts unaltered so the hole, and thread could be applied **by** the fastener.

> Constraints You reference base.Mate here. Are you going to change that to base.Constraint?

No, the constraint does not hold references to other constraints; `base.Mate` is correct.

> The code there uses from cqparts import Part. I just wanted you to be aware that FreeCAD has a Part module that could conflict [...]

Hmm, yeah, I didn't think of that, but I also won't be changing the name. `Part` is going to be a very common class name, I think that's going to be unavoidable, so people will have to import one as a different name, or reference it as `cqparts.Part`.
I might encourage people to use the `cqparts.Part` and `cqparts.Assembly` so they only need to `import cqparts`, and then the distinction is clear.

> What does display do?

That's not planned... in the short term I think it will just recursively iterate through components of assemblies and `show` them.
But I was going to ask you closer to release what you thought it should do.
It's quite malleable since it's just a function that takes a `Component` as input.

jmwright, 2017-11-27:
> What does display do?

> That's not planned... in the short term I think it will just recursively iterate through components of assemblies and show them.
> But I was going to ask you closer to release what you thought it should do.

The [CadQuery Gateway Interface](http://dcowden.github.io/cadquery/cqgi.html) (CQGI) addresses the challenge of displaying the results of scripts across build environments. That's where the show_object call comes from in newer scripts. You also get things like parameter editing on the fly, and debug objects. It seems like it would be best to make use of CQGI somehow.

fragmuffin, 2017-11-27:
@jmwright 
Yes, that's definitely the plan for release, I'm just more concerned right now with getting them to build ;)