cqparts/cqparts #26: Part placement

fragmuffin, 2017-11-22:
Place parts with 

* translation : vector (easy)
* rotation
    * quaternion
    * angles

need to look at how this is done conventionally.

jmwright, 2017-11-22:
I don't know if it's a standard way to do it, but have a look at the "Assembling the Design" section [here](https://rustyvermeer.github.io/design/ipad_stand/index.html) to see how @AdamVermeer has handled assemblies in Jupyter Notebooks.

fragmuffin, 2017-11-22:
At a high level, I'm thinking an implementation of:

```python
import cadquery
import cqparts

class Box(cqparts.Part):
    def make(self):
        return cadquery.Workplane('XY').box(1, 2, 4)

box = Box()

# Position
box.position = cadquery.Vector(1, 2, 3)
box.position = (-4, -6, 10)
box.location = # alternative name
box.translation = # alternative name

# Rotation / Orientation
box.quaternion = cqparts.Quaternion(w=0.7, x=0.4, y=0.4, z=0.4)
box.angles = # euclidean angles?
box.orientation = # alternative name for one of the above?
box.transform = # a 4x4 standard transform matrix
```

@jmwright, @dcowden
do you have any preference for naming, or the method of _location_, _rotation_, and _transform_ in general.

note: these translations and rotations will cascade down through all the `Assemblies`.
I'll have tests that make sure they don't get performed in the wrong order.
So practically I just need to focus on locating and rotating a simple primitive (like the above cube)

fragmuffin, 2017-11-22:
Thanks @jmwright ! I'll read that when I get home

fragmuffin, 2017-11-22:
Using the same method as Adam's _ipad_stand_ would be something like:

```python
box.object = box.object \
    .rotate((0,0,0), (0,1,0), 90) \
    .rotate((0,0,0), (1,0,0), 270) \
    .translate((38.1, 247.65, 76.2))
```

But there are a few ways this won't work with `cqparts`

`rotate` and `translate` are _changes_ to the position & orientation; they're not an absolute set.
I think I can support both with a _translate_ operation like `box.position += Vector(1, 2, 3)`

But there's another complication:

```python
>>> isinstance(box, cqparts.Part)
True
>>> box.position = Vector(1, 2, 3)  # however...
>>> box.object  # is only created when requested
<cadquery.cq.Workplane at 0x7fde663afdd0>
>>> # so setting Vector(1, 2, 3) would only be applied when box.object is first created

>>> # once box.object is made, there's 2 ways to set / change a position
>>> box.position = Vector(-1, 10, -30)  # set position directly
>>> # or
>>> box.object.translate((38.1, 247.65, 76.2))
>>> # but all of that should be "under the hood", so it doesn't matter how people set the position
```

I didn't want to do this, but I think I need to wrap the `box.object` so location & orientation changes to `box.object` are identical to `box`.
So `box.object` won't be a `cadquery.Workplane`, but a `cqparts.WrappedWorkplane`, enforced by `cqparts.Part`.

dcowden, 2017-11-22:
I would be interested to see an example where we are positioning one part
relative to another vs absolute. Once there are several parts, it seems to
me that people will think like ' line up this parts face with this other
parts face' vs angles and positions.

I think it might make things easier if there were any abstraction for a
'mate', which is just a predefined coordinate system. A part would expose a
mate, which consists of s vector and two othogonal directions.

Then you can connect two parts by referencing a mate.  By default, when two
mates are connected, the coordinate systems of the mates align. Other
options allow for inverse and offset mates.

That leaves the question of which part is rotated in space to fulfill the
mate.  To handle that, you could allow parts to be marked as fixed. These
parts are moved and translated as you described above, with vectors and
rotation mayrices. When other parts mate to a fixed part, the fixed part
stays fixed, and the others move to align as needed.

This system would save a little of math creating an assembly. I put a few
components where I want them, and the rest are calculated for me be
matching up their exported mates.

fragmuffin, 2017-11-23:
@dcowden 
ok, I'm just wrapping my head around this... :thinking: 
You've spoken about _mates_ before, but I now think I'm finally getting it.

I like the idea of a `Mate` class, then it can have all the bells & whistles without polluting `Part` and `Assembly` classes.
So each `Part` and `Assembly` has a `.mate` that can be set in their constructor.

Then when the `.object` (the `cadquery.Workplane` instance) is made, it's moved so it's origin coordinate system aligns with the `.mate`.

The  mate could also be changed while building (so after `.object` is created), which will move the internal `.object`.  Although I think that should be avoided because doing that excessively would take a lot of unnecessary processing, and result in rounding errors.
Alternatively, the `Mate` could be immutable... no, I can't do that because then it's impossible to place something based on it's size... you'd need to create the part instance, query it to see how big it is, then set it's mate accordingly.
(all of this is fine-tuning, and can be done later)

**Euclidean Geometry**
All of this is going to require a 3rd party lib to do the geometry, I'm inclined to either use [`euclid3`](https://pypi.python.org/pypi/euclid3) (I've used that before) or `numpy`. Numpy is a bit more heavyweight, and I've had trouble installing it in the past, so I'd like to avoid it if there's a simpler alternative.
Do you have any suggestions?

fragmuffin, 2017-11-23:
do you think I'm crazy for having minimal requirements, because an alternative that's quite intuitive is `scipy`

dcowden, 2017-11-23:
It might be better to store a mate as a transformation from the original
coordinate system. That way, it's possible for a part to offer more than
one mate.  When you choose a mate to use, you simply apply the
transformation required to align the csys of the two parts. In this way,
the assembly doesn't affect the part definitions at all, and could even be
added to an imported step that wasn't built with cq.

Regarding dependencies, I guess my opinion is not to fear them, for two
reasons: (1) we already have so many it's a problem we are solving ( in
fact the latest docker image I published has scipy via anaconda ) and (2)
it's worth using a dependency to prove out your concept more quickly. You
can always replace it later

dcowden, 2017-11-23:
@fragmuffin do you have access to either onshape or solidworks? I really
like how mates are done in those, and what I'm describing basically works
like they do. If you don't has access, I can take some screenshots of
Onshape assemblies so you can see.  Once you see them in action it's much
more clear.  The power really comes from the ability to create different
kinds of alignment constraints when you connect the mates. We have been
talking about the simplest kind of mate.. a fully constraining coincident
mate.  But what's really powerful is the ability to make coplanar or axial
mates, which position the parts based on solving more than one mate.

It may be a reach to do this, but if you do, it would be enterprise class.
It would require evaluating all of the connected mates and parts together,
not one at s time.

Let me know if some screenshots would be helpful

fragmuffin, 2017-11-23:
@dcowden 
Regarding dependencies: that's a good philosophy, I think the fully constrained mate will simply be a 4x4 `numpy.ndarray`, after some reading. Most libs that provide more intuitive functions end up creating a numpy array anyway.

I don't have access to solidworks or onshape... Or rather, I think I may have a freebee account with onshape, I don't know what that offers though.
Screenshots, or even a YouTube video of somebody introducing the behaviour you're describing would be great.

I think I understand what you're describing. I think that's definitely a future improvement, but i can at least design it so it won't be a major change.

fragmuffin, 2017-11-23:
@dcowden 
There are a lot of videos on solidworks' mates, this one for example:
https://www.youtube.com/watch?v=rKqn3CmaFv0
So essentially these result in a 3d constraint solver... very neat, and yes!, I'd love `cqparts` to get to that stage at some point.
Don't know if I'll have time, but this should definitely shape the design of the initial release.
(also, I just noticed I'm using the same terminology as solidworks, _parts_ and _assembly_, although I guess this isn't surprising; it is quite intuitive)

**Assembly Structure**
This has shown me that constraints, or _mates_ are **not** paired with parts.
Rather, each mate references an arbitrary number of parts.
So each `Assembly` has a list of `Parts` or `Assembly` instances, and a separate list of `Mate` instances.

**Solving / Building an Assembly**
The process of building an `Assembly` would be to successively _solve_ the `Mate` instances, then apply the solution's _transform_ to a `Part`. (keeping in mind that sometimes a _transform_ may only be found with multiple `Mate` instances referencing the same parts)

In the early days, a _solution_ to a fully constrained `Mate` is explicitly given to the `Mate`, so for now that's easy, but I think this design allows for some very interesting expansions.
(I'm going to have to start to learn about constraint solving)

**Locking**
You mentioned a *locked* part being the base.
I don't think this is necessary because it's the equivalent of applying a fully-constrained (identity matrix) `Mate` to it, locking it to the assembly's coordinate system.

dcowden, 2017-11-24:
@fragmuffin, yes definitely it's a real challenge to build this directly,
after all, it's competing with an enterprise CAD system right?!

I'm mainly mentioning it as a guide to ensure those doors remain open. To
be honest I really have no idea how to go about implemting a 3d solver, so
anything is better than nothing! What you're doing is awesome stuff....

fragmuffin, 2017-11-24:
@dcowden 
I try to keep doors open, so your insight has been very helpful!
I'm not at all experienced with CAD modelling, but I'm comfortable programming.
It's the whole reason I started this project.

I'm inclined to call these `Constraint` rather than `Mate`. It's a more accurate name for their role.
I think the term "mate" can be reserved for some intuitive functions.
For example: (psudocode)

```python
def mate_faces(part1, face1, part2, face2, mirrored=False):
    constraint = PlanarConstraint(...)
    return constraint
```

now I'm just working out how to put all this together.

dcowden, 2017-11-24:
Yep I think you have it....

jmwright, 2017-11-24:
From the perspective of a GUI that integrates cqparts into its workflow, it seems like there are multiple levels of things the user can interact with. From high level to low, they are the following.
- Project (collection of assemblies, parts, models and related files for things like analyses and iterative design)
- Assembly (collection of parts and constraints)
- Part (collection or union of models)
- Model (combination of 3D primitives that serve a single purpose or make a single object)
Does that sounds about right?

fragmuffin, 2017-11-24:
@dcowden 
Regarding constraints that don't result in a single solution (eg: a linear actuator, like a pneumatic piston), how do you see the _motion_ fitting in with the `cadquery` stack?

fragmuffin, 2017-11-24:
@jmwright 
That does sound right, but I consider the onus of **project** and **model** to be that of the author.

To clarify, the **project** could be in the form of a github repository, and a **model** (as a unit) being the output of a script in that repository.

Would you agree?

fragmuffin, 2017-11-24:
@jmwright @dcowden 
I'm trying to figure out the best data format to store a coordinate system offset (translation, and rotation) as the primary means of communication across _constraints_ and _solvers_.

**3D transform matrix**
I've just brushed up on my matrix algebra, namely [rotation & translation matrices, and combining them](http://planning.cs.uiuc.edu/node100.html).
So my ideal data format is a `numpy.ndarray` (with caveats described below).

**Alternatives?**
Do you think another format would be better?

**Flexibility / 3rd Party Tooling**
I'll need functions to do the conversions to and from this data format.
For example, to create a 4x4 affine transform _matrix_ with the input:

* Euler angles
* x-axis vector, y-axis vector (or any other, perpendicularity asserted)
* or more arbitrarily: vector A, vector B (perpendicular to A), vector A', vector B' (transformed)
* start vertex, rotation axis vector, angle
* (and preferably more)

And also to reverse each of these functions (with a _matrix_ as input)

I'm having some difficulty finding a common library to do it.
I'm sure I could write all of these, but it seems wildly out of scope for `cqparts`.
Here are a few examples I've found, which are along the right track, but they're not exactly mainstream publications:

* https://www.lfd.uci.edu/~gohlke/code/transformations.py.html
* https://afni.nimh.nih.gov/pub/dist/src/pkundu/meica.libs/nibabel/eulerangles.py

**Am I Crazy?**
Considering this is the maths used in **every** 3d rendering application (at least, I thought it was), I'm very surprised that there isn't a glaringly obvious solution...
I feel like I'm missing something... I'm probably simply looking for the wrong search terms.
What do you think I should use?

`FreeCAD.Matrix` provides some basic functionality, but no where near enough to be intuitive.

dcowden, 2017-11-24:
@fragmuffin 

> Regarding constraints that don't result in a single solution (eg: a linear actuator, like a pneumatic piston), how do you see the motion fitting in with the cadquery stack?

Very good question!  When an assembly has degrees of freedom, GUIs will generally allow you to move the parts around to get a feel for how it works, which is very handy to make sure the assembly works correctly. I actually use that sometimes for mechanical projects.

The best translation of this behavior into a script would be what i'd call a 'constrained move'-- after the assembly is solved, you should be able to move a part to one position or another, and then the entire assembly should re-solve based on the constraints.   

It would be really nice if when you move a part, the stack automatically removes the directions of your move that are impossible, and applies the others.  For example, if you are moving a piston, and the piston axis is in direction (0,1,0),  move(1,1,1) should become a move of (0,1,0) since it is constrained the other ways.

dcowden, 2017-11-24:
@fragmuffin 
>I'm having some difficulty finding a common library to do it.
I>'m sure I could write all of these, but it seems wildly out of scope for cqparts.

Why not use the implementations already built into CQ?  It already provides some easier to use wrappers around FreeCAD Matrix and Vector.  The matrix wrapper is 
[here](https://github.com/dcowden/cadquery/blob/master/cadquery/freecad_impl/geom.py#L197)

For an example of how to use a matrix to transform between coordinate systems, have a look at the CQ [Plane class](https://github.com/dcowden/cadquery/blob/master/cadquery/freecad_impl/geom.py#L473).  It transforms between a local coordinate system to the plane to world coordinates and back, which is the same operation you would use to rotate an assembly into another coordinate system

>@jmwright @dcowden
>I'm trying to figure out the best data format to store a coordinate system offset (translation, and rotation) as the primary means of communication across constraints and solvers.

One option would be to use a CQ Plane object.  It was designed to represent a coordinate system, having a normal vector, an origin, and an X direction.  Many 3d libraries have different primitives for a coordinate system, but I chose to combine them. The bonus is that if you use a Plane, you already have the functionality to rotate it to another coordinate system built in, via [toWorldCoordinates](https://github.com/dcowden/cadquery/blob/master/cadquery/freecad_impl/geom.py#L473), and to [rotate other shapes](https://github.com/dcowden/cadquery/blob/master/cadquery/freecad_impl/geom.py#L518) into it from worldCoordinates 

The definition of a Plane is three, 1x3 vectors. One point, one normal, and one addtional direction for the X axis:

`'XY': Plane((0,0,1, (1, 0, 0), (0, 0, 1))`

These are all defined relative to the global coordinate system, so the above represents a coordinate system  with its origin at x=0,y=0,z=1, with its normal pointing in the x direction, and its x direction in the global z direction.

CQ automatically computes a coordinate system at the center of a face with its normal pointing outwards when you select one-- that's how it sketches on a face.  So I imagine that you could define a mate by simply using a CQ selector to select the face, and then get the underlying plane.  So this:

`Workplane('xy').circle(2,0).extrude(11.0).faces(">Z").workplane().plane`

Should actually return what i'd call a mate-- a plane, defined in terms of global model coordinates, which serves as a reference for connecting other parts. ( in this case, the end of a shaft)

So i think it could work something like this:
  * start with two cqparts, each having a mate defined by a Plane, like above
  * each constraint has these properties:
         * mate1 -- the first mate ( a plane, expressed in global coords of model 1)
         * mate2 -- the second mate ( a plane , expressed in global coords of model 1)
         * a type -- coincident, parallel, concentric, perpendicular
         * a boolean -- alligned or anti-aligned
  * the assembly has its own global coordinate system.
  * an assembly has a list of parts and a list of constraints
  * part can be added to an assembly in these ways:
       * 'fixed'-- the part is glued into the assembly and cannot move.  In this case, the user supplies rotations and offsets relative to x,y,and z of of the part for its location.  The system automatically creates a co-incident constraint relative to the parts' global coorindate system, and adds it to the constraint list ( this way, a 'fixed' object is just another constraint).
      * 'mated' -- the part is added by choosing two mates ( probably by their names), and a constraint type.
  * once the user is done, code runs to 'solve' the assembly. this is the hard part
       * i really have no idea how to do it
       * for now, perhaps we could implement a 'dumb' solution, and just implement transformations in the order they were constrainted?

dcowden, 2017-11-24:
@fragmuffin hey, yah google, for example like this one:  https://labix.org/python-constraint

from there, we'd need to figure out how to represent mates as contraints-- that seems do-able.

jmwright, 2017-11-24:
Tagging @adam-urbanczyk here as I'm wondering if he'll have thoughts on all of this as well. Adam, if you haven't seen anything about cqparts yet, let us know if you need some context for this discussion.

@fragmuffin 
>  ...I consider the onus of project and model to be that of the author.
> To clarify, the project could be in the form of a github repository, and a model (as a unit) being the output of a script in that repository.

On the Project->Repo comparison that's how I'm seeing it. On Model->Script, I'm not so sure. A single script could technically output a part as well if I'm understanding cqparts correctly so far. The Model->Script comparison is accurate for what we have now (or is at least the most common use case), but I think we're moving into a broader application of scripting with what we're discussing here. Agree or disagree?

> When an assembly has degrees of freedom, GUIs will generally allow you to move the parts around to get a feel for how it works, which is very handy to make sure the assembly works correctly.

I think there's a fair amount of work to make this happen with CQ-specific constraints/mates, but I agree that it's important. I plan to start with the PythonOCC viewer (that Adam has used before) and rebuild/extend it from there to add the mouse interaction capability we need. Moving assemblies gets into territory that I'm not familiar with though, so we'll so how it goes.

How is cqparts doing in regartds CQ 2.0 compatibility? Is it going to be major surgery to get cqparts working with the fluent CQ 2.0 API, and then again when the direct API is implemented?

fragmuffin, 2017-11-24:
@dcowden 

> One option would be to use a CQ Plane object. It was designed to represent a coordinate system, having a normal vector, an origin, and an X direction.

Aghh!, I can't believe I missed that... so that answers the question: "am I crazy" :wink:
I found the `cadquery.Matrix` wrapper, but stopped there.
Yes, that's a good structure to use, if I want to add any other helpers, or utilities to the class, I can add them to `cadquery`, instead of branching off a new coordinate system for people to learn.
Thank you!

> `Workplane('xy').circle(2,0).extrude(11.0).faces(">Z").workplane().plane`
> 
> Should actually return what i'd call a mate-- a plane, [...]

That's great... ok, I'm really glad that's solved.

> for now, perhaps we could implement a 'dumb' solution, and just implement transformations in the order they were constrainted?

Yes, the initial plan is to only implement a `LockConstraint`, which simply provides a part, and a `Plane`.
That `Plane` instance **is** the solution for the solver.
So the solver will just loop through constraints and return the `(part, plane)` pair.

> i have interest in geometric constraint solving for core cq as well-- its necesary to do 'real' 2d sketching

Well, I think once we have 3d solving going, 2d should be a snap! (pun intended :stuck_out_tongue_winking_eye: )

fragmuffin, 2017-11-24:
@jmwright

> On Model->Script, I'm not so sure. A single script could technically output a part as well if I'm understanding cqparts correctly so far [...]
> but I think we're moving into a broader application of scripting with what we're discussing here

I think we may have something different in mind regarding a "Model", could you elaborate?, perhaps an example of the application of a model.

> I think there's a fair amount of work to make this happen with CQ-specific constraints/mates, but I agree that it's important

As long as we can callback when something relevant has changed, all that callback has to do is re-run the solver, then iterate through the solver's solution and  set each part's new position (if changed).
It could be a bit tricky, but initially we could just have a "Solve" button.

> How is cqparts doing in regartds CQ 2.0 compatibility? Is it going to be major surgery to get cqparts working with the fluent CQ 2.0 API, and then again when the direct API is implemented?

It will be a bit of work, a lot of it will require the same changes... I'm not worried about it, but I'm sure it will need a transition.
As for the fluent API, I'm assuming that's just optional; you can create exactly the same `Workplane` object using either fluent or direct API's... in which case, converting to direct API is optional.
Ultimately `cqparts` will be the same, it's just how the `make()` functions are implemented that change.

jmwright, 2017-11-25:
> I think we may have something different in mind regarding a "Model", could you elaborate?, perhaps an example of the application of a model.

Maybe I'm just hung up on old terminology. When @dcowden created CadQuery, he called the scripts on parametricparts.com "Model Scripts" if I remember right. I think I just took the term "model" from there. The problem is that a model script can define a plain block, or it can define a 3D printer extruder mount, which to me is more of a part. Those are both from the CadQuery examples. The main thing I'm trying to figure out is when a user clicks File->New in the GUI, what are they creating? A model, a part, a project? Every option is a script (or collection of scripts) behind the scenes, but what that script represents from the user's perspective can change. What I have right now is just a "New" menu item and a "New Project" item. It seems like I'll almost have to make the user click New, and then bring up a dialog that asks whether it's a new script (for coders who don't want the abstraction), part, assembly or project.

> As long as we can callback when something relevant has changed, all that callback has to do is re-run the solver, then iterate through the solver's solution and set each part's new position (if changed).

It shouldn't be hard to re-run the solver after a mouse drag event ends on a part in an assembly. That doesn't take into account the ability to do live previews as the user is dragging, but I'll settle for baby steps that don't peg CPU utilization to start with.

jmwright, 2017-11-25:
@fragmuffin Your definitions [here](https://fragmuffin.github.io/cqparts/doc/parts/index.html) clear this up and offer a foundation for future discussions.

dcowden, 2017-11-25:
Just my $0.02, I think more concrete terms tend to be better. Most users
will be able to understand the difference between a 'part', an 'assembly',
more clearly than a 'model'.

Is it fair to say that in both cases and in cqparts, users will be making
files which eventually create all of the abstract object types? Is it also
fair to say that a single cqparts python file could create one or more
parts and also maybe even more than one assembly?

If these statements are true, I'd use terms part and assembly to refer to
the abstractions, and then just call order things 'files'.  I agree that
the user is very likely to have their files in some sort of organized
structure they call either a project or a repo or a folder.

fragmuffin, 2017-11-25:
@jmwright 

> It seems like I'll almost have to make the user click New, and then bring up a dialog that asks whether it's a new script (for coders who don't want the abstraction), part, assembly or project.

Good point... could we have multiple templates (or I think the kids are calling them "snippets" these days :older_man:).
Creating a new *part* **or** a new *assembly* as a file makes it seem like the script (the file) is fundamentally different in some way, or that the *need* to be separated out into files.
So you can start a new script (which just has the standard `cadquery` make a cube and display it as a template), then have snippets of *part*, *assembly*, *model*, etc...

Some snippets could be for `cqparts`, and other snippets could be for other libraries to come in future.
Those templates could link back to the tutorials in the `cqparts` documentation so the "bread crumbs" are placed for willing people to follow.

> It shouldn't be hard to re-run the solver after a mouse drag event ends on a part in an assembly. That doesn't take into account the ability to do live previews as the user is dragging

oh!, yes, I totally agree.
I wonder if FreeCAD has some sort of *pre-move* hook that we can give a callback.

jmwright, 2017-11-26:
> Creating a new part or a new assembly as a file makes it seem like the script (the file) is fundamentally different in some way, or that the need to be separated out into files.

Script file versus part file is more a logical separation within the GUI. There are a significant number of programmers who use CadQuery because traditional CAD just doesn't make sense to them, or forces them outside of their normal thought processes. Those users will likely want to skip any part, assembly and project abstractions and just start coding their widget. So, while there's nothing special about a script file versus a part file, from a UI/UX perspective I think it's worth making the distinction.

> ...could we have multiple templates..?

Yeah, sure. Allowing users to choose between a cqparts Part or Assembly template and one from another framework cleanly will be interesting, but certainly doable.

> So you can start a new script (which just has the standard cadquery make a cube and display it as a template), then have snippets of part, assembly, model, etc...

I'm probably mis-understanding things here, but I think that if the user selects the option to create a new assembly, the GUI should create a new folder, correct? According to the cqparts docs:

> Part is to Assembly as file is to folder.

Although you can stuff the parts and mates into a single script, trying to conflate all of that could end up making the assembly harder to understand and maintain.

> I wonder if FreeCAD has some sort of pre-move hook that we can give a callback.

FreeCAD is hit and miss on how low-level you can get with its components. We can look into that, but this kind of functionality will probably be implemented in a custom version of the PythonOCC viewer. We can still display FreeCAD-CQ parts and assemblies in it, and then we'll have access to lower level mouse events and things like that.

I just realized that all this GUI stuff is serious scope creep for this issue. Sorry.

fragmuffin, 2017-11-26:
@jmwright 

> I'm probably mis-understanding things here, but I think that if the user selects the option to create a new assembly, the GUI should create a new folder, correct? According to the cqparts docs:
> 
>> Part is to Assembly as file is to folder.

Hmm... well, the file:folder analogy is just that; an analogy. In practice, most assemblies will be in a **file** named for the assembly, not a folder.

> I just realized that all this GUI stuff is serious scope creep for this issue. Sorry.

No worries! I think it's definitely a valid topic of discussion, feel free to create issues in this project on anything you want to discuss.

I think a lot of this will be clear when I've got a few initial complex & nested assemblies.
The servo motor is my proof-of-concept assembly that I'm targeting.
The fasteners are assemblies, but they'll rarely be nested, and most of their value comes from the `fasteners.utils` content.

fragmuffin, 2017-11-26:
Back on topic.

I'm concerned about the build steps regarding placement.
This is a long post, but it's discussing the fundamentals of building up an `Assembly`.

For discussion, I'll create a box stack

* 2 boxes, with variable heights
    * the first at the `Assembly` origin (so, it's essentially "fixed")
    * the 2nd on top of the 1st, and rotated around the z axis

Each box is the same `Part` with different parameters.

```python
import cadquery
import cqparts
from cqparts.params import *
class Box(cqparts.Part):
    length = PositiveFloat(10)
    width = PositiveFloat(20)
    height = PositiveFloat(40)
    def make(self):
        return cadquery.Workplane('XY').box(
            self.length, self.width, self.height,
            centered=(True, True, False)
        )
```

Using this in an assembly `Assembly`:

```python
from cqparts.constraints.base import Mate
from cqparts.constraints.principal import LockConstraint

class BoxStack(cqparts.Assembly):
    first_level = PositiveFloat(10, doc="height of 1st level")
    second_level = PositiveFloat(15, doc="height of 2nd level")
    angle = Float(45, doc="angle of stacked block")
    def make(self):
        # --- Create and add components
        box1 = Box(10, 10, self.first_level)
        self.add_component('box_a', box1)
        box2 = Box(20, 30, self.second_level - self.first_level)
        self.add_component('box_b', box2)

        # --- Constrain components
        # lock to origin
        self.add_constraint(
            LockConstraint(box1, Mate(0,0,0, (1,0,0), (0,0,1)))
        )
        # lock to the top of box1 at a 
        self.add_constraint(
            LockConstraint(box2, Mate(
                0,0,box1.height,
                (1,0,0), (0,0,1)
            ).rotated((0,0,self.angle)))
        )
```

Then creating an assembly instance:

```python
box_stack = BoxStack(angle=30)  # just change the angle
```

At this point `box_stack.components` is a dict of the components.
`box_stack.components['box_b'].object` returns a `cadquery.Workplane` instance of the stacked box, but it's placed at the origin.

**Solving creates a copy of everything**
When the solver is in this chain, it must create a copy of the `cadquery.Workplane` that's translated and rotated based on the solver's solution.
So this means that every part will have 2 `cadquery.Workplane` instances, one is in its original coordinate system, and the other is transformed according to the solver's solution.

Does that sound necessary, having a copy of everything?

**Changing parts after solving**
There's another thing missing from this stack, fasteners provide a good illustration.
Consider the above example with a screw holding the top block to the bottom one.

* the block parts are made
* both blocks are put into an assembly, with constraints (locking them to a coordinate system other than their base coordinate system)
* only **after** the solver has a solution can the fasteners be assessed and applied.

I'm wondering if this is a good solution:

```python
class BoxStack(cqparts.Assembly):
    def make_components(self):
        # ... return dict of components
    def make_constraints(self):
        # ... return list of constraints
    def post_placement(self):
        # ... run after the solver to make changes to the altered parts.
```

ok, I have more thoughts, but the above novel should be enough for now

adam-urbanczyk, 2017-11-26:
@jmwright  I do need some context. How is cqparts related to cq?

Some random thoughts,  not sure if they are still relevant:
* there is a class in OCC for representing coordinate system: gp_Ax3. Using workplanes as coordinate system only sounds like an overkill to me
* For solvers you'll probably need some external library. AFAIK e.g. FreeCAD usese the BFGS and/or SQP algorithms for solving the constraints
  * in the end it might be practical to code this in C++/ cython/..
* for representing position an intuitive option is:
  * translation vector, ( rotation angle, axis of rotation unit vector)

fragmuffin, 2017-11-26:
@adam-urbanczyk 

> I do need some context. How is cqparts related to cq?

`cqparts` applies a hierarchy of components, each component is created using `cadquery`.
It's also a means to catalogue these hierarchies, so they can be created multiple times in one project, each using different parameters.

> * there is a class in OCC for representing coordinate system: gp_Ax3. Using workplanes as coordinate system only sounds like an overkill to me
> [...]
> * for representing position an intuitive option is:
>     * translation vector, ( rotation angle, axis of rotation unit vector)

To clarify, a `Mate` isn't a `cadquery.Workplane`, just a `cadquery.Plane`.
`cadquery.Plane` is little more than what you're describing.

> For solvers you'll probably need some external library.

I couldn't agree more, but solvers are out of scope for this topic, and there won't be a solver for the initial release.

fragmuffin, 2017-11-26:
ok, this might be messy, but I'm going to work with this build chain for now:

* create `cadquery.Workplane` with geometry for the `Part` at its origin.
* run (dummy) solver
* move a copy of `cadquery.Workplane` into place
* optional alterations
    * make changes to the moved `cadquery.Workplane` copy
    * move a copy of the copy back to the `Part` origin (0,0,0), and use it to replace the part's generated object.

### Limitations:

**Affected by Constraints**
The effect of alterations may be affected by constraints.
If a part is moved, it, or another part's geometry, may change.

In reality, the alterations will only be run once, as the solver is re-run the part will move but its geometry won't change.

**Alterations can't touch Constraints**
Alterations cannot set or change constraints.
I think this is a good thing; alterations are *fine-tuning*, they should not make major structural changes.

fragmuffin, 2017-11-27:
Nope, I've just realised how messy making changes to the moved part is.

**Alterations**
are made to the `Part.object` in it's original coordinate system, so all world coordinates will have to be converted to local coordinates.
Then when the object is re-assigned, its copy is moved back to world coordinates, replacing the original piece. (so the initial move is purely to assess what alterations need to be made)

fragmuffin, 2017-11-29:
@jmwright @dcowden @adam-urbanczyk
Hey all, I've got questions regarding...

## Constraints Solver:

**Constraints for the initial release**
For the first release I'll just be implementing a `LockConstraint` and `RelativeLockConstraint`, they're both documented in the [Constraints section](https://fragmuffin.github.io/cqparts/doc/parts/constraints/index.html) of the docs.
Simply put, `LockConstraiant` explicitly sets an object's world coordinate set... whereas `RelativeLockConstraint` sets an object's coordinate set relative to another object.

also see [`block_tree.py`](https://github.com/fragmuffin/cqparts/blob/master/tests/manual/block_tree.py) for my "proof" script I'm focusing on... I can fine-tune simple shapes and angles to make sure everything meets up perfectly.

**Initial Solver Design**
I'm inclined to go for the following initial design.

1. collect constraints at `Assembly` level into a dict: `{<Component>: [<constraint1>, <constraint2>, ...]}`
   where all constraints in the list are constraining the key `Component`
1. loop through `.items()`, attempt to find a valid world coordinate system for each `Component`.
1. if > 0 solutions were found in the last loop, then loop through unsolved components again (necessary for relative lock)
1. continue until all components have a world coordinate system, or 0 solutions are found (raise exception)

**Should this be recursive?**
Do you think, instead, it would be necessary to collect a list of all constraints recursively?
This way constraints could span `Assembly` instances...

Can anyone describe an example of needing to constrain *across* groups of components?
Or is everything typically built in a pure hierarchy?

It's crude, but it will work. with a big-O of `O(n/2)`, but realistically much much less.
Also, with the current constraints, it's only possible to have 1 constraint, so I'll just raise an error if there's more than one constraint for a particular `Component`.

**Future Solver Design**
I understand this will just be a "band-aid" until a real solver is in place.
@adam-urbanczyk : you mentioned BFGS and SQP nonlinear optimisation methods, do you have any experience implementing these?, if so does the above design sound like it could be expanded without changing people's existing implementation?

dcowden, 2017-11-29:
> Do you think, instead, it would be necessary to collect a list of all constraints recursively?

I can't think of any use cases when we only consider LockConstraint and RelativeLockConstraint-- it seems that these are necessarily  unary by definition.

All of the CAD systems I've ever used represent constraints as a binary thing-- you cannot ever have a 3-way or n-way constraint.  IE, if you want a plane to be parallel to several things, you have to do them in pairs. I think that works fine.

fragmuffin, 2017-11-30:
Thanks @dcowden, I've made some good progress over the last couple of days.
Even though it's 35 degC here, and my study only has a 150mm diameter fan :sweat:.

I'll post a screenshot here as soon as I've got the solver working in a hierarchy >=3 deep

fragmuffin, 2017-11-30:
Yesss!, ok, ok....
first working solver and parts placement
script: [`block_tree.py`](https://github.com/fragmuffin/cqparts/blob/master/tests/manual/block_tree.py)

**result**
[image]

jmwright, 2017-11-30:
@fragmuffin Looks great!

fragmuffin, 2017-11-30:
@jmwright I disagree; it looks hideous! But it works!!

fragmuffin, 2017-11-30:
Closing this issue, as this is what I set out to achieve.
Further enhancements to follow in #30.

closed 2017-11-30 by fragmuffin

dcowden, 2017-11-30:
That's so awesome @fragmuffin!

I think it worked out very well. Your system is the only scriptable
assembly method that I know of!

Are you able to load a step file and the define mates and include it in an
assembly? I think that would be super useful because a lot of times all of
the parts can't be scripted with cq.

fragmuffin, 2017-11-30:
Thanks @dcowden !
I see no reason why not, that's one of the items I want to address in #19.

adam-urbanczyk, 2017-12-06:
@fragmuffin  in the end you need a list of terms (functions of all DOFs [3 rotations + 3 translations per part] that represent each constraint) that you can feed to the minimization algorithm.