cqparts/cqparts #29: Documentation Feedback

fragmuffin, 2017-11-28:
Please use this issue to give feedback on the documentation published at:
https://fragmuffin.github.io/cqparts/doc/

**Note:** this documentation isn't always updated with changes to `master`; it often lags a little behind.

This issue will be closed upon the release of `v0.2` regardless of the state of documentation.

fragmuffin, 2017-11-28:
@jmwright I've made this topic for general doc feedback, based on your comment in #28

Good catch on the constraints code! I've fixed it up for now (commit: 6059f10), but so much of the base code is in flux right now.
That code is in a `doctest` block, so future changes should cause test failures... and quite frankly, it causes an error now, because I haven't put the `Component.build()` method in ;)

fragmuffin, 2017-12-01:
@jmwright @dcowden

Added tutorials for:

* **Parametric Objects** https://fragmuffin.github.io/cqparts/doc/tutorials/parametric-object.html
* **Parts** https://fragmuffin.github.io/cqparts/doc/tutorials/part.html

If you've got time, your feedback would be welcome.
Tomorrow I'll finish the above _Parts_ and move on to _Assembly_.

This documentation is helping me think :smile:

jmwright, 2017-12-01:
@fragmuffin If I pull the current master into CQFM (the FreeCAD module) will I be able to walk through these examples? Is all the cqparts code there?

dcowden, 2017-12-01:
@fragmuffin  Lovin' how its coming so far. The biscuit example and the easy-install examples are really great examples to use!

fragmuffin, 2017-12-02:
@jmwright : Yep, all the examples work with the current master... almost all the documented example code is wrapped in a sphinx `doctest`, and at this point they all pass.

Thanks @dcowden, I still don't know exactly how they're going to work, but I tend to write documentation / tutorials like this when I get stuck in my design. As soon as I start making excuses in the documentation, I change the code so there are no longer any excuses to make ;).
It's a philosophy that's worked well for me so far.

fragmuffin, 2017-12-09:
Just added...

**Assembly Build Cycle**
https://fragmuffin.github.io/cqparts/doc/parts/assembly-build-cycle.html

dcowden, 2017-12-09:
I love the style and detail of this!

It's easy to read and follow.

I'm a little confused about relative mates work, though. When I join two
parts, I expect to reference a mate on both parts. In the example, the mate
in the plate made perfect sense, but I was surprised that it wasn't
necessary to define a mate in the cylinder. When mating two parts, I expect
to provide a mate type( the class name in this case),and two mates, which i
would expect to come from two Part subclasses. So, like:

return CylindricalConstraint(thing1.mate_hole, thing2.bolt_mate)

It might be cool if each part has a world coordinates mate built in, so
that you can use world coordinates as a mate if you want:

CylindricalConstraint (thing1.mate_hole,thing2.world)

In short, in your example, the mate must be using global coordinates in the
cylinder as the mating csys, but I think i'd prefer it to be more explicit.
Or am I totally missing it?

https://fragmuffin.github.io/cqparts/doc/parts/assembly-build-cycle.html

fragmuffin, 2017-12-11:
@dcowden 

>I love the style and detail of this!
> It's easy to read and follow.

I'm glad to hear that! I was very pleased with how it worked out.
At the moment the loading times are terrible because as a web-page it's very poorly designed, but I'll fix that later, and it will look the same, but load a lot quicker.

> In short, in your example, the mate must be using global coordinates in the
> cylinder as the mating csys, but I think i'd prefer it to be more explicit.
> Or am I totally missing it?

[... writing demonstrative code ...]
I was originally writing you a reply on how to do it by using `CoordSystem` arithmetic, but I got lost simply connecting 2 parts together.

But it should be *much simpler* than that.

So you're totally right; that **is** missing.

Currently the `LockConstraint` and `RelativeLockConstraint` constructors are:

```python
LockConstraint(<component>, <mate>)
RelativeLockConstraint(<component>, <mate>, <relative_to>)
```

But I'll re-factor them to:

```python
LockConstraint(<component>, <local_mate>, <world_mate>)
RelativeLockConstraint(<component>, <local_mate>, <relative_to>, <relative_mate>)
```

Thank you for your feedback!, this helps me a lot.
I have very little CAD experience; it's things like this that illustrate just how helpful your feedback is :+1:

fragmuffin, 2017-12-11:
@dcowden 
on second thought... I've made it so a `Mate` simply inherits from `CoordSystem`.... but I'm now thinking that's all wrong.

`Mate` should be a `CoordSystem` relative to a `Component`...
So rather than `Mate` _being_ a `CoordSystem`, a `Mate` should *use* a `CoordSystem` and a `Component`

So with `Mate` changed to keep reference to a `Component`, the above `Constraints` would change to:

```python
LockConstraint(<mate>, <world_coords>)
RelativeLockConstraint(<mate>, <to_mate>)
```

which is so much more intuitive!

I think I have some major refactoring to do.

fragmuffin, 2017-12-11:
@dcowden 
I've updated the way Mates, and Constraints work... have a look at

* [Constraints](https://fragmuffin.github.io/cqparts/doc/parts/constraints/index.html)
* [`Assembly` build cycle](https://fragmuffin.github.io/cqparts/doc/parts/assembly-build-cycle.html), and
* [Make your own `Assembly`](https://fragmuffin.github.io/cqparts/doc/tutorials/assembly.html)

and let me know what you think ;)

dcowden, 2017-12-11:
@fragmuffin yes i like that much better!  Much more explicit.

For documentation purposes, the kinds of mates I think you end up with are below. I've used the 'CAD' terms if that's helpful. I know both CAD and programming, so the CAD terms make sense to me, but i'm not sure they'd make sense to everyone.

For clarity, I've noted the degress of freedom the contraint applies
Fixed.  orient one mate to a fixed position and orientation in space ( DONE: LockConstraint) DOF: 6

**Coincident**.  orient two mates, matching both orientation and position ( DONE: **RelativeLockConstraint** ) DOF: 6

**Cylindrical**: orient the origin of two mates, and fix an axis of roation on the z axis ( DOF: 4 )
Revolve: orient two mates to share an axis of rotation ( cylindrical, but without matching the origin) (DOF: 1 )

**Parallel/Coplanar** : orient two mates so that their x-y planes are parallel, with some fixed offset. if offset is zero, the planes are coplanar, otherwise, they are offset by some amount ( DOF: 2 )

**Slide** : orient two mates so that their z axes align, and can only slide along that direction (DOF: 1 )

**Spherical**:  align the origin of two mates, but allow them to rotate freely otherwise ( DOF: 3 )

**Slot**: orient two mates so that the origin of one mate must lie on one selected  axis of another, and must not rotate  ( DOF: 5 )

In CAD, these are typically different types of mates. In programming, there can be more permutations, and its possible to represent these and more with fewer kinds of mates, which is pretty cool actually. For example, the cylindrical mate above requires orientation around the Z axis, but it would be easy to allow the user to specify any of the three axes for rotation.

fragmuffin, 2017-12-11:
Thanks @dcowden ! I'll put this into #30

closed 2018-01-10 by fragmuffin