cqparts/cqparts #30: Advanced Constraints

fragmuffin, 2017-11-30:
Create constraints that allow more freedom of placement.

**Implementation**: non-linear Optimisation
`scipy` offers a range of optimisation methods.
https://docs.scipy.org/doc/scipy-0.14.0/reference/generated/scipy.optimize.minimize.html

Optimisation methods could be defaulted, then optionally fine-tuned by advanced users, all through one common interface (`scipy`).

---

Inspiration for constraints... naming observed from official _OnShape_ and _Solidworks_ tutorials:

**Solidworks**

* angle, coincident, concentric, distance, lock, parallel, perpendicular, and tangent mates
* limit, linear/linear coupler, path, symmetry, and width
* cam-follower, gear, hinge, rack and pinion, screw, and universal joint

**OnShape**

* Fastened
* Revolute
* Slider
* Planar
* Cylindrical
* Pin Slot
* Ball

fragmuffin, 2017-12-05:
### Todo List

**cleanup tasks**

* [ ] remove `todo` box from [Constraints](https://fragmuffin.github.io/cqparts/doc/parts/constraints/index.html)

fragmuffin, 2017-12-11:
Clarification from @dcowden (paraphrased slightly)

> * **Fixed:** orient one mate to a fixed position and orientation in space ( DONE: _LockConstraint_) DOF: 6
> * **Coincident:** orient two mates, matching both orientation and position ( DONE: RelativeLockConstraint ) DOF: 6
> * **Cylindrical:** orient the origin of two mates, and fix an axis of rotation on the z axis ( DOF: 4 )
> * **Revolve:** orient two mates to share an axis of rotation ( cylindrical, but without matching the origin) (DOF: 1 )
> * **Parallel:** orient two mates so that their x-y planes are parallel, with some fixed offset. if offset is zero, the planes are coplanar, otherwise, they are offset by some amount ( DOF: 2 )
> * **Coplanar:** a _Parallel_ constraint with a zero offset
> * **Slide:** orient two mates so that their z axes align, and can only slide along that direction (DOF: 1 )
> * **Spherical:** align the origin of two mates, but allow them to rotate freely otherwise ( DOF: 3 )
> * **Slot:** orient two mates so that the origin of one mate must lie on one selected axis of another, and must not rotate ( DOF: 5 )

I think this is as good a list as any, it makes perfect sense to me on first reading, and it comes from an experienced modeller & programmer!

@dcowden: you mentioned potentially allowing a change of axis.
For example, the _Cylindrical_ could revolve around the y-axis instead of the z-axis.

I think I like the simplicity of restricting how the constraints are implemented. It also makes documenting them much easier, and therefore more likely to be understood by laypeople.

But that wouldn't mean the creations themselves are limited, authors can create a new mate by getting an existing one and rotating the mate itself.
With that in mind, I'll write the code now so you can add a `CoordSystem` to a `Mate`... for example

```python
mate = Mate(component, CoordSystem((0, 0, 10)))
new_mate = mate + CoordSystem(xDir=(0,1,0), normal=(1,0,0))
```

**What I'll do before initial release:**
I want to remove "Constraint" from the end of each constraint, it's unnecessary, and a bit oldschool.
If authors want that to be clear they can write `from cqparts import constraint` and `constraint.Fixed`
I'll also rename `LockConstraint` to `Fixed`, and `RelativeLockConstraint` to `Coincident`

dcowden, 2017-12-12:
@fragmuffin sounds good.

This is a picture of the assembly toolbar for Onshape-- its very similar to Solidworks. You can see the icons for the kinds of mates.

[image]

The first icon to the right of "Insert" is the icon they use for a mate.

SeanDS, 2019-07-17:
Are you also thinking about constraints between parts/assemblies? I'd love to have some way to constrain parts to be separated by lengths or angles.

fragmuffin, 2019-07-30:
@SeanDS : Yes! absolutely. constraints will be between any 2 `Components`, which includes both `Part` and `Assembly`

everlanes, 2019-10-06:
Hi, I have a proof of concept for a flexible solver using the scipy [least square root solver](https://docs.scipy.org/doc/scipy/reference/generated/scipy.optimize.least_squares.html#scipy.optimize.least_squares). It is working quite nicely and I can apply multiple constriants to any part.

For now I have to following constraints working:
* Origin: constrains origin and direction of a part to a CoordinateSystem()
* OriginDir: constrains only the orientation of a part to a CoordinateSystem()
* Position: constrains position and direction of two mates
* Direction: constrains only the direction of two mates
* Plane: constrains one mates position to be located in a plane parall to the seconds mate plane

Next steps would be to clean it up and integrate it into the cqparts code. This would mean to replace the current solver.solve method the Assembly.build function and to add new constraint classes.

My questions:
1. is it desired that I work on this?
2. is anyone else currently working on this?
3. can I modify the Assembly.build function or is anyone else currently working on this?
4. which branch should I fork to minimize merge conflicts?

dcowden, 2019-10-06:
This is so awesome!

SeanDS, 2019-10-06:
@everlanes: I don't have time right now to check this out, but it sounds awesome! There seem to be very few Pythonic CAD libraries that support constraints, so having these in cqparts would be great. Thanks for working on this, and I'll let you know if I have any feedback when I try it out :-)

everlanes, 2019-10-11:
I have uploaded a first draft for a numeric solver with flexible constraints in #158 

All tests are passing in my development environment, but sadly not with AppVeyor - OCC dependency is missing and scipy dependency should be added. I'm just not sure, where to add these dependencies.