cqparts/cqparts #31: Tests

fragmuffin, 2017-12-01:
Write tests for:

* [x] Parameters
* [ ] Parametric Object
* [ ] Parts
* [ ] Assemblies
* [ ] Solver with various Constraints
* (more?)

fragmuffin, 2018-01-12:
start point:

```
Name                                                                Stmts   Miss  Cover
---------------------------------------------------------------------------------------
/home/nymphii/prj/cqparts/src/cqparts/__init__.py                      20      0   100%
/home/nymphii/prj/cqparts/src/cqparts/assembly.py                     138    116    16%
/home/nymphii/prj/cqparts/src/cqparts/catalogue/__init__.py             3      0   100%
/home/nymphii/prj/cqparts/src/cqparts/catalogue/catalogue.py            2      0   100%
/home/nymphii/prj/cqparts/src/cqparts/catalogue/json.py                49     28    43%
/home/nymphii/prj/cqparts/src/cqparts/codec/__init__.py                61     14    77%
/home/nymphii/prj/cqparts/src/cqparts/codec/amf.py                     11      4    64%
/home/nymphii/prj/cqparts/src/cqparts/codec/gltf.py                   201     17    92%
/home/nymphii/prj/cqparts/src/cqparts/codec/step.py                    23      1    96%
/home/nymphii/prj/cqparts/src/cqparts/codec/stl.py                     11      0   100%
/home/nymphii/prj/cqparts/src/cqparts/codec/svg.py                     11      0   100%
/home/nymphii/prj/cqparts/src/cqparts/codec/threejs_json.py            40     22    45%
/home/nymphii/prj/cqparts/src/cqparts/component.py                     25      3    88%
/home/nymphii/prj/cqparts/src/cqparts/constraint/__init__.py            4      0   100%
/home/nymphii/prj/cqparts/src/cqparts/constraint/base.py                1      0   100%
/home/nymphii/prj/cqparts/src/cqparts/constraint/constraints.py        23     16    30%
/home/nymphii/prj/cqparts/src/cqparts/constraint/mate.py               29     21    28%
/home/nymphii/prj/cqparts/src/cqparts/constraint/solver.py             31     27    13%
/home/nymphii/prj/cqparts/src/cqparts/display/__init__.py              11      4    64%
/home/nymphii/prj/cqparts/src/cqparts/display/freecad.py               20     16    20%
/home/nymphii/prj/cqparts/src/cqparts/display/material.py              40     15    63%
/home/nymphii/prj/cqparts/src/cqparts/display/web.py                   65     47    28%
/home/nymphii/prj/cqparts/src/cqparts/errors.py                         7      0   100%
/home/nymphii/prj/cqparts/src/cqparts/params/__init__.py               16      0   100%
/home/nymphii/prj/cqparts/src/cqparts/params/parameter.py              25      7    72%
/home/nymphii/prj/cqparts/src/cqparts/params/parametric_object.py      62     25    60%
/home/nymphii/prj/cqparts/src/cqparts/params/types.py                  98     12    88%
/home/nymphii/prj/cqparts/src/cqparts/params/utils.py                   7      4    43%
/home/nymphii/prj/cqparts/src/cqparts/part.py                          59     22    63%
/home/nymphii/prj/cqparts/src/cqparts/search.py                        38     12    68%
/home/nymphii/prj/cqparts/src/cqparts/utils/__init__.py                 8      0   100%
/home/nymphii/prj/cqparts/src/cqparts/utils/env.py                      6      1    83%
/home/nymphii/prj/cqparts/src/cqparts/utils/geometry.py                58     25    57%
/home/nymphii/prj/cqparts/src/cqparts/utils/misc.py                    27     10    63%
/home/nymphii/prj/cqparts/src/cqparts/utils/wrappers.py                 7      6    14%
/home/nymphii/prj/cqparts/src/cqparts_misc/__init__.py                  2      0   100%
/home/nymphii/prj/cqparts/src/cqparts_misc/basic/__init__.py            3      0   100%
/home/nymphii/prj/cqparts/src/cqparts_misc/basic/indicators.py         33      7    79%
/home/nymphii/prj/cqparts/src/cqparts_misc/basic/primatives.py         51     13    75%
---------------------------------------------------------------------------------------
TOTAL                                                                1326    495    63%
```

need to bring that average up significantly... I only have myself to blame.

fragmuffin, 2018-01-15:
Average coverage is over 90% now.
I'm getting some string encoding errors coming from `cadquery`, I need to figure those out soon too.
I'm testing on Python v2.7, I suspect it's a python 3 migration thing, my environment may be misconfigured.

jmwright, 2018-01-15:
We've had issues recently with the Python 3 update and Adam pushed a few changes. You might make sure you have the latest master to see if that helps.

fragmuffin, 2018-01-17:
@jmwright 
I had 2 problems, one was my fault, fixed in 3e2505c
The other I've just pushed in is in [cadquery PR #218](https://github.com/dcowden/cadquery/pull/218)

fragmuffin, 2018-01-17:
Coverage average at 97%

The main difference now is with the `materials.py` file, however I expect this file to be stripped out with replaced content...
#36 will replace the `_render` parameter with a more intuitive `_material` parameter.
(not planned for initial release)

```
Name                                                                Stmts   Miss  Cover
---------------------------------------------------------------------------------------
/home/nymphii/prj/cqparts/src/cqparts/__init__.py                      20      0   100%
/home/nymphii/prj/cqparts/src/cqparts/assembly.py                     142      2    99%
/home/nymphii/prj/cqparts/src/cqparts/catalogue/__init__.py             3      0   100%
/home/nymphii/prj/cqparts/src/cqparts/catalogue/catalogue.py            2      0   100%
/home/nymphii/prj/cqparts/src/cqparts/catalogue/json.py                49      0   100%
/home/nymphii/prj/cqparts/src/cqparts/codec/__init__.py                61      1    98%
/home/nymphii/prj/cqparts/src/cqparts/codec/amf.py                     11      0   100%
/home/nymphii/prj/cqparts/src/cqparts/codec/gltf.py                   201      2    99%
/home/nymphii/prj/cqparts/src/cqparts/codec/step.py                    23      0   100%
/home/nymphii/prj/cqparts/src/cqparts/codec/stl.py                     11      0   100%
/home/nymphii/prj/cqparts/src/cqparts/codec/svg.py                     11      0   100%
/home/nymphii/prj/cqparts/src/cqparts/codec/threejs_json.py            40      0   100%
/home/nymphii/prj/cqparts/src/cqparts/component.py                     25      0   100%
/home/nymphii/prj/cqparts/src/cqparts/constraint/__init__.py            5      0   100%
/home/nymphii/prj/cqparts/src/cqparts/constraint/base.py                1      0   100%
/home/nymphii/prj/cqparts/src/cqparts/constraint/constraints.py        23      0   100%
/home/nymphii/prj/cqparts/src/cqparts/constraint/mate.py               29      0   100%
/home/nymphii/prj/cqparts/src/cqparts/constraint/solver.py             31      0   100%
/home/nymphii/prj/cqparts/src/cqparts/display/__init__.py              11      4    64%
/home/nymphii/prj/cqparts/src/cqparts/display/freecad.py               20      0   100%
/home/nymphii/prj/cqparts/src/cqparts/display/material.py              40     11    73%
/home/nymphii/prj/cqparts/src/cqparts/display/web.py                   65      1    98%
/home/nymphii/prj/cqparts/src/cqparts/errors.py                         7      0   100%
/home/nymphii/prj/cqparts/src/cqparts/params/__init__.py               16      0   100%
/home/nymphii/prj/cqparts/src/cqparts/params/parameter.py              25      0   100%
/home/nymphii/prj/cqparts/src/cqparts/params/parametric_object.py      62      0   100%
/home/nymphii/prj/cqparts/src/cqparts/params/types.py                  98      5    95%
/home/nymphii/prj/cqparts/src/cqparts/params/utils.py                   7      4    43%
/home/nymphii/prj/cqparts/src/cqparts/part.py                          52      0   100%
/home/nymphii/prj/cqparts/src/cqparts/search.py                        38      0   100%
/home/nymphii/prj/cqparts/src/cqparts/utils/__init__.py                 8      0   100%
/home/nymphii/prj/cqparts/src/cqparts/utils/env.py                      6      1    83%
/home/nymphii/prj/cqparts/src/cqparts/utils/geometry.py                54      0   100%
/home/nymphii/prj/cqparts/src/cqparts/utils/misc.py                    27      0   100%
/home/nymphii/prj/cqparts/src/cqparts/utils/wrappers.py                 7      6    14%
---------------------------------------------------------------------------------------
TOTAL                                                                1231     37    97%
```

fragmuffin, 2018-01-17:
**Plan for parts libraries testing:**
Parts library test coverage will be mostly covered by building the geometry of every catalogue in the library.

So I'll create a `unittest.TestCase` generator that pulls all it's data from a catalogue (one test per catalogue entry).

That's a bit messy, but it will yield nice looking test-results with minimal effort.
It will also encourage people to add their parts to a catalogue, or create a new catalogue for their component(s).

And besides, the test-suite is where all the messy/dodgy code should go, isn't it?

fragmuffin, 2018-01-18:
Added catalogue test automation.
So you can test an entire library catalogue by witing

```python
from cqparts.catalogue import JSONCatalogue
from cqparts.utils.test import CatalogueTest

catalogue = JSONCatalogue('my_catalogue.json')
CatalogueTest.create_from(catalogue, add_to=globals())
```

That will be picked up by `python -m unittest discover`, and run 1 test per catalogue entry.
To demonstrate, the test directory's `partslib` is tested, resulting in:

```
test_box_a (cqparts.utils.test.CatalogueTest_catalogue) ... ok
test_box_b (cqparts.utils.test.CatalogueTest_catalogue) ... ok
test_box_c (cqparts.utils.test.CatalogueTest_catalogue) ... ok
test_cs_01 (cqparts.utils.test.CatalogueTest_catalogue) ... ok
test_cs_02 (cqparts.utils.test.CatalogueTest_catalogue) ... ok
test_cs_03 (cqparts.utils.test.CatalogueTest_catalogue) ... ok
test_cyl_a (cqparts.utils.test.CatalogueTest_catalogue) ... ok
test_cyl_b (cqparts.utils.test.CatalogueTest_catalogue) ... ok
test_cyl_c (cqparts.utils.test.CatalogueTest_catalogue) ... ok
test_ford (cqparts.utils.test.CatalogueTest_catalogue) ... ok
test_holden (cqparts.utils.test.CatalogueTest_catalogue) ... ok
test_merc (cqparts.utils.test.CatalogueTest_catalogue) ... ok
```

Each part's geometry is built, and assembly constraints are solved... ultimately it's currently just making sure all deviations listed in the catalogue can be generated without an exception being raised.

I'm quite proud of that one :blush: 
Next step is to create catalogues for existing components, so they may be tested using this method.

fragmuffin, 2018-01-26:
I've been a bit distracted with the idea of making a scraper to build catalogues. #43
Not directly related to this issue, but having a detailed catalogue for any library proves the catalogue testing concept (`cqparts_fasteners` was the logical choice).

fragmuffin, 2018-02-13:
Just discovered that the `CatalogueTest` utility was just re-creating the same part over and over again... fixed in 698a0c1.
The above catalogue tests were re-run with no issue... phew!