cqparts/cqparts #33: Link cadquery exporters

fragmuffin, 2017-12-05:
Support `Part` and `Assembly` exporting (`Assembly` exporting mainly for viewing)

fragmuffin, 2017-12-05:
@dcowden : I've been looking into json exports first, mainly because they're directly importable into a-frame, which I've been experimenting with in the [Make your own `Assembly`](https://fragmuffin.github.io/cqparts/doc/tutorials/assembly.html) page.

**What format is this?**
https://github.com/dcowden/cadquery/blob/master/cadquery/freecad_impl/exporters.py#L316-L358
(and don't say JSON)
I assumed it was a version of [glTF](https://github.com/KhronosGroup/glTF/tree/master/specification) but I find very little correlation between the specifications and the exports.

Also, I search for "TJS" and find nothing helpful.

I feel like I'm taking crazy pills :stuck_out_tongue_winking_eye: :pill: 

**Merging Models**
I ask because I want to export multiple models (or `Part` instances) with multiple materials, and merge them into one file for viewing an `Assembly`.

Any advice would be appreciated.

dcowden, 2017-12-05:
Ha TJS is a three.js file. It can be used to render a model in s browser using the three.js JavaScript library.

fragmuffin, 2017-12-05:
ahh, ok, that was my first problem... I was looking up "**T**essellated **J**ava **S**cript", because tessellation is mentioned a lot in the _TJS_ exporter code.

I think I'm getting warmer now: https://github.com/mrdoob/three.js/wiki/JSON-Object-Scene-format-4

Thanks :+1:

dcowden, 2017-12-05:
@fragmuffin on and sorry I didn't address your second question-- merging multiple models for viewing.
I think a JSON format would still be good for viewing an assembly.  ThreeJS can handle multiple objects in a scene, but three.js JSON is not very standard. The nice thing about three.js is that you can send different colors and materials for each object quite easily.

I'm not aware of any industry standard format for assemblies in professional CAD space.  STEP might be able to handle it, but i've never seen that done.  It might be worth making a FreeCAD file with multiple solids in it, and see what STEP it produces. Even so, that's not an assembly, its a static set of objects, with no information about their colors, relationships, etc.

If you want to stick closer to standards, there would be only two others I can think of: VRML and its successor, X3D.

It also might be cool to look at WebGL, and determine what file format would be renderable that way.  I think WebGL has very broad browser support, and I'm sure it supports multiple materials and objects. I suspec there is some kind of JSON file you could use that could be rendered easily with WebGL, and if so that's probably better than using my TJS format because it looks like WebGL is kind of the 'right way' to do 3d viewing in a browser now.

fragmuffin, 2017-12-05:
@dcowden no problem regarding multiple objects, that's how I'm currently displaying on the tutorial.
Check out [Make your own `Assembly`](https://fragmuffin.github.io/cqparts/doc/tutorials/assembly.html), it's using the same method as the [iPad stand](https://rustyvermeer.github.io/design/ipad_stand/index.html)
Although I'm getting a bit tired of creating a new document each time I want to display something.
At some point I'd like to create a sphinx extension that runs the code in a block and renders the model with the `display` command.

I agree, the json format is only good for display... but it seems it's actually fantastic!
I'm considering using the `display` command to start up a HTTP service on localhost, then open a browser link... If I get very clever I could make it render to the same window, much like the way FreeCAD works now.... but all of that is out of scope right now.

Regarding VRML, X3D, WebGL, and so on... I think exporting to [glTF](https://github.com/KhronosGroup/glTF) may be the best standard to follow. It's highly recommended for aframe, and aframe loads it via a three.js loader anyway (so it's supported there too). Both three.js and aframe are built on WebGL, and also support WebVR!

I'll give _glTF_ a spin, and see how I go... the only problem with it is it will have to be multiple files:
the `*.gltf` file is sort of like an `index.html`, containing some information, but mostly linking out to everything else.... including geometry, which is stored in binary files.
I haven't read about the binary file format yet, but it shouldn't be too difficult.

I'll post my progress here.

also, I'll put this code in `cqparts` for now, but it should be migrated over to `cadquery` in future.

dcowden, 2017-12-05:
@fragmuffin sounds great, thanks for all the work you're doing! I didnt know about gITF, it looks nice! If you like it let me know, a back-port for cq would be good so that cq can natively export that format as well.

fragmuffin, 2017-12-12:
I've split #35 from this issue, because this issue is just about linking the existing `cadquery` exporters to `cqparts` components.

fragmuffin, 2017-12-13:
this took a while because I wanted to put the import/export mapping on steroirds :pill: 

**Export Example**

```python
from cqparts.basic.primatives import Cube
cube = Cube()

# pull an exporter out that has reference to the cube
step_exporter = cube.exporter('step')
step_exporter('cube.step')

# or in a single line:
cube.exporter('step')('cube.step')
```

**Import Example**

```python
from cqparts import Part
cube = Part.importer('step')('cube.step')
# at this point, the file is not imported
cube.local_obj  # as soon as the object is requested, it's "made", or in this case, imported
```

**Bad Exporter**

This also gives an informative error if the export you want doesn't exist...

```python
>>> from cqparts.basic.primatives import Cube
>>> Cube().exporter('abc')
TypeError: exporter type 'abc' is not registered: registered types: ['amf', 'gltf', 'json', 'step', 'stl', 'svg']
```

closed 2017-12-13 by fragmuffin