cqparts/cqparts #37: Display with Web-Interface

fragmuffin, 2017-12-19:
Create a means to display a `Part` or `Assembly` with _WebGL_ by exporting the component being displayed, then open a link to it in.

The initial implementation will simply create a web-service to a static page displaying the model data.
It will be terminated with a keyboard interrupt (Ctrl+C).
(crude, but functional)

**Out of Scope**
The following is explicitly out of scope for this issue, but should be considered for future design.

* Fork process into a daemon, that can handle:
    * Multiple calls to `display`
    * Re-displaying with new model data
    * Don't re-export if model data is identical

* Keeping a line of communication (ie: via web sockets) to the environment that started the display. This will facilitate:
    * Measurements
    * Moving / animating parts, re-solving their positions based on constraints

**Question**
I haven't used _jupyter_ yet, where would this fit in with that environment?

jmwright, 2017-12-20:
Tagging @RustyVermeer for the Jupyter question above. he has the most experience with it.

fragmuffin, 2017-12-20:
Thanks @jmwright 

To illustrate what this does, running the commands...

```python
>>> from cqparts.basic.primatives import Cube
>>> from cqparts.display import web_display
>>> web_display(Cube(size=10))
press [ctrl+c] to stop server
127.0.0.1 - - [20/Dec/2017 17:38:54] "GET / HTTP/1.1" 200 -
Created new window in existing browser session.
127.0.0.1 - - [20/Dec/2017 17:38:55] "GET /model/out.gltf HTTP/1.1" 200 -
127.0.0.1 - - [20/Dec/2017 17:38:55] "GET /model/out.bin HTTP/1.1" 200 -
^C[server shutdown successfully]
>>>
```

will do the following:

* create a temporary folder
* export model data (using glTF implemented in #35)
* start http service in a thread
* prompt the OS to open a web-browser on the locally hosted page
* wait for user to stop the web-service with ctrl+c
* shutdown http service, and delete temporary folder

It has a lot of room for improvement, particularly with rendering, but for now it's just a means to view models.

RustyVermeer, 2017-12-22:
@fragmuffin, there are a few ways you could tackle things in Jupyter.

With the points you've outlined above, you could very likely achieve a similar result.

* create a folder in a location relative to the active notebook (.ipynb file which is connected to an IPython kernel)
* export model data (same way you do it already)
* use a templated single-page html page to display the model, or othewise have this as part of your export function
* use some built in display functions available in Jupyter to display an iframe beneath the active cell. The iframe will be of the html file generated in the previous step

Some limitations:
* you *could* delete the temp. files immediately after displaying the iframe. It will have loaded the page already. BUT, if you then refresh or restart your jupyter notebook, it will 404, as it tries to reload a non-existent resource.
* you have to maintain an html template to go along with this display functionality. 
* probably can't have this going in too many cells... loading a ton of iframes might be bad for performance. 

You could alternatively do something like what I've done and use a jupyter widget called [pythreejs](https://github.com/jovyan/pythreejs). It uses the [proper Jupyter widget 'machinery'](http://jupyter.org/widgets.html).

I've used it successfully in my [cqnb module](https://github.com/RustyVermeer/cqnb) using the following steps:
* show_object() function executes, sending the cell's script to CQGI, which parses the script.
* the parsed script then is used to generate Jupyter's interactive widgets and a button which will execute the model display function upon click.
* user changes any settings they wish and then clicks 'run interaction' to invoke the display function which:
* executes the CQ script and then tessellates the result, which is immediately placed into a JSON format.
* the JSON is manipulated somewhat to match what pythreejs expects (for example, computing face normals)
* pass the data to pythreejs as a plainbuffergeometry. 
* render. This results in the object being displayed to the user, and is interactive.

If you want to actually try out this second method, you can visit [https://trycadquery.com](https://trycadquery.com) and open up an example or two. (NOTE: stick with a simple example, as the server doesn't have a ton of ram to work with... don't want to crash the kernel :) )

And, for reference, here's the iframe method:

[image]

And the same object displayed using the cqnb module:

[image]

fragmuffin, 2017-12-22:
Thanks @RustyVermeer !, that's super helpful.

At the moment I think jupyter integration is out of scope, but as you've pointed out, it seems very plausible.
Especially with all the good work you've done!

I'll close this issue, but I've raised #38 to deal specifically with jupyter integration (at some point in the future)

closed 2017-12-22 by fragmuffin