cqparts/cqparts #40: Split libraries

fragmuffin, 2018-01-03:
In building `cqparts` I've build too much into the one library.
So to compartmentalise, and to help design, I'll split it into 3 libraries:

- `cqparts`
    - the whole `Part` & `Assembly` thing
    - importing / exporting
    - display utilities
    - (essentially anything **before** any specific geometry is defined)

- `cqparts_misc`
    - miscellaneous parts & assemblies
    - an optional place for collaborators to add their own creations, if they don't want to create their own libray

- `cqparts_fasteners`
    - isolation of fasteners concept, and implementation, including...
    - fastener head parts
    - screw drive parts
    - thread parts

jmwright, 2018-01-03:
@fragmuffin Will there be more libraries later on? `fasteners` is so specific and `misc` is so broad.

fragmuffin, 2018-01-03:
@jmwright 
I agree:

- `fasteners` is specific, but it's true to its content
- `misc` is just that, a group of random things, I don't expect any of them to be particularly well designed. 

For example, I have a plan to make some different motor types, but none of them will be designed well enough to fabricate and work from their models...
The focus would be more about creating a shell that looks like a motor.

So with that in mind, If I end up designing a motor well, I would want to add it to a `cqparts_motors` library... so perhaps I could add:

- `cqparts_motors`
- `cqparts_springs`
- `cqparts_bearings`
- `cqparts_gears`

what do you think?

jmwright, 2018-01-03:
So the `misc` library will be things that don't fit in any other category or are low fidelity. Is that correct?

I think those groupings make sense. In the GUI I'm working on I've done some very early thinking about how cqparts would integrate. Two of the options that users have when they select File->New are 'Part' and 'Assembly'. If the user selects either one of those a cqparts template is used. I have also put in a placeholder 'Collections' dialog box that allows the user to add a library like `cqparts_motors` so that they have access to all the parts in that library. Of course, the motors library has a dependency (`cqparts_bearings`), so I now realize that the 'Collections' dialog will need some way of discovering and handling those dependencies.

I've still got a ways to go before I can get serious about integrating cqparts, but it's high on the priority list.
* Figure out the GUI build system. I want to automate PyInstaller builds for all the major OSes, as much as possible.
* Work with Adam U. to finalize the CadQuery PythonOCC implementation and get it integrated.
* Work with you to make sure cqparts and PythonOCC are compatible.

Once all that is done I'm hoping to get you set up and running the GUI (if you're willing) so that we can start figuring out how cqparts and the GUI will work together.

It's not my intent to hijack this thread, but the work you're doing here is directly related to what I'm trying to think through with the GUI.

fragmuffin, 2018-01-04:
> Of course, the motors library has a dependency (cqparts_bearings), so I now realize that the 'Collections' dialog will need some way of discovering and handling those dependencies.

Just to clarify, I see all of the above `cqparts_*` libs being pip-installable, which means any dependencies will be sent to PyPI, and figured out by pip.
If somebody wants to make their own importable module for their own project, then that's just python, nothing special.

> I'm hoping to get you set up and running the GUI (if you're willing) so that we can start figuring out how cqparts and the GUI will work together.

Yeah!, I'm willing!

I should warn you about my opinions on GUI's though...
I find the necessity for a GUI *removes* power from applications, I've seen it happen over and over in my 
career, which is the whole reason I sought out `cadquery` in the first place.
However, CAD is a very visual art, so a GUI is necessary to *view* the things created; so to say `cadquery` should not have a GUI at all is naive.
I would however argue that a `cadquery` GUI should not be an editor; that a `cadquery` (or `cqparts`) GUI should strictly have a 1-way role, as a viewer, not an editor.

That's one of the reasons why I put so much effort into the glTF exporter, because it serves as an excellent medium to convey all of a model's geometry, material, movement, and assembly hierarchy into a format that can be imported to a web interface in a few lines of threejs code.

OK, wow, I'm hijacking my own thread... I've created #41 to separate out from this issue.

fragmuffin, 2018-01-04:
### Tasklist

- [ ] `cqparts_misc`
- [ ] `cqparts_fasteners`
- [ ] `cqparts_gears`
- [ ] `cqparts_bearings`
- [ ] `cqparts_springs`
- [ ] `cqparts_motors`
- [ ] `cqparts_torquelimiters`
- [ ] `cqparts_gearboxes`
- (more?)

fragmuffin, 2018-01-04:
**Question**
Does anyone know if there's a limit to how many PyPI libraries can be registered by a single user?

jmwright, 2018-01-04:
I don't know of any limit to the number of libraries. A quick search turned up a size limit, but not much else in the way of limits.

closed 2018-01-06 by fragmuffin