cqparts/cqparts #41: GUI Discussion

fragmuffin, 2018-01-04:
# Thread

This is a temporary issue to talk `cqparts` (and `cadquery`) GUIs

fragmuffin, 2018-01-04:
from #40

> I find the necessity for a GUI removes power from applications, I've seen it happen over and over in my
career, which is the whole reason I sought out cadquery in the first place.
However, CAD is a very visual art, so a GUI is necessary to view the things created; so to say cadquery should not have a GUI at all is naive.
I would however argue that a cadquery GUI should not be an editor; that a cadquery (or cqparts) GUI should strictly have a 1-way role, as a viewer, not an editor.
> 
> That's one of the reasons why I put so much effort into the glTF exporter (#35), because it serves as an excellent medium to convey all of a model's geometry, material, movement, and assembly hierarchy into a format that can be imported to a web interface in a few lines of threejs code.

I'm also a big fan of WebGL, although I don't have that much experience with it (yet).
So I'm looking forward to expanding the glTF 2.0 exporter to export more than just geometry and materials, but I'll also need a detailed WebGL viewer (or at least recommend a viewer).

I imagine creating a `.json` file to accompany the `.gltf` to guide a viewer to display a model with more specific context.

jmwright, 2018-01-04:
> I find the necessity for a GUI removes power from applications

Agreed. You get some things in return for that loss of power and narrowing of options though. We have some users who won't ever mess with a GUI, and others who wouldn't use CadQuery if it didn't have a GUI.

> I would however argue that a cadquery GUI should not be an editor; that a cadquery (or cqparts) GUI should strictly have a 1-way role, as a viewer, not an editor.

Many users would not use the CadQuery workbench for FreeCAD if it was view-only. It also wasn't until that workbench was introduced that we saw a major upswing in the number of users. So I'd be hesitant to make the GUI I'm working on view-only since it will be our PythonOCC equivalent of the workbench. With that said, our workbench can be put in a kind of view-only mode by opening a script and changing the setting to use an external editor. Then when the script is updated externally, the 3D view will update. I have plans to make sure that the new GUI supports that type of mode where you are not stuck with the editing capabilities that come built-in to the GUI.

However, even with that argument for an editor GUI, I don't disagree with having a view-only gITF/WebGL application. There are several use cases for CadQuery, and I think that makes it impossible to create a one-size-fits-all user interface model. That's one reason why users aren't discouraged by @dcowden and I when they want to try something new. We've tried to encourage user choice in how people use CadQuery, and we're excited that there's a growing ecosystem of ways to interact with it. As a community we're still feeling out the edges of what people can and will do with CadQuery, and I feel that it's perfectly valid (and necessary) to explore all sorts of user interaction models. We need to keep pushing the bounds and seeing what works for different users and use cases.

dcowden, 2018-01-04:
@fragmuffin thanks for starting the dialog.  Two thoughts to add:

(1) The type of viewer I most want to see is one that is read only, but allows changing and updating some parameters and re-generating the part.  CadQuery was designed around the idea that building correct design intent into a model is hard-- but it needn't be hard for non-technical users to modify the key inputs to get a new object.  For example, consider a cup. I want the 'viewer' to allow showing the default configuration, AND to allow exposing the height and diameter to the user, so they can change it. But i dont necessarily think many users will hack the actual code for the model without a full (fast) development environment.  This is a viewer that's somewhere between read only and read-write, i guess.

(2) GUIs still have the value of allowing a user to select things, which is very hard to replicate.  A good example is the use of trimming when sketching. I'd say at least a third of my sketches use construction lines and trimming to accomplish my geometry, to avoid me having to do math.  Its easy to do this in a GUI-- draw a line, and then click the side i want to trim.  Its kind of hard to do anything like that programmatically.

fragmuffin, 2018-01-05:
@jmwright @dcowden 

I just re-read my post, I just realised how black/white it sounds...
I should clarify, all of my previous strong statements are with regard to the **necessity** for a GUI... so if cadquery **must** be bundled with a GUI, then it must not be mandatory to use it... but from what you've said, I think you both agree.
I would hate to see the development of `cadquery` devolve by scope-creep; sinking development effort into renderers, and IDEs.

> We have some users who won't ever mess with a GUI, and others who wouldn't use CadQuery if it didn't have a GUI

I agree, getting that audience brings content and exposure to cadquery.
And the GUI's are currently 1-way, they're used to display what's been coded.

> when the script is updated externally, the 3D view will update

That would be great to see!, oh!, I have ideas for this too (I'll post them in a separate post below)

> (1) The type of viewer I most want to see is one that is read only, but allows changing and updating some parameters and re-generating the part.

I also agree!, this is an excellent compromise. To change a generated model with sliders & buttons without becoming a code generator.
I may be able to help with that, #39 has become an exercise in serializing `Component` instances, I'll separate that out into another post (after this one, it may be long).

> A good example is the use of trimming when sketching. I'd say at least a third of my sketches use construction lines and trimming to accomplish my geometry, to avoid me having to do math

That is a good example, I have been [using trigonometry a lot](https://github.com/fragmuffin/cqparts/search?utf8=%E2%9C%93&q=sin+OR+cos+OR+tan+OR+pi) in the `cqparts` lib, where in many cases, a sketch would definitely be quicker to make.
Although building them with the same freedoms we've come to expect of CadQuery may be difficult.

It sounds like the problem of building sketches is similar to the problem of building assemblies...
Could this be another aspect of `cqparts`?... so `Part` is to `Assembly` as a `Wire` is to a `Sketch`... with mates & solvers and everything.
Anyway, that's a bit off topic (also, I'm not looking for more work :worried: )

jmwright, 2018-01-05:
> (1) The type of viewer I most want to see is one that is read only, but allows changing and updating some parameters and re-generating the part.

> I also agree!, this is an excellent compromise. To change a generated model with sliders & buttons without becoming a code generator.
> I may be able to help with that, #39 has become an exercise in [...]

The latest version of the FreeCAD workbench has a variable pane built in like this if the script uses CQGI (show_object). It doesn't have sliders or anything fancy, but you can change the variables and re-execute the script without editing anything by hand. Rusty Vermeer's Jupyter notebook plugin also exposes variables in a form that lets you change them dynamically. We haven't gotten much feedback yet, but I don't think most people are aware the functionality is there. I really need to update the workbench documentation and do some new tutorial videos.

fragmuffin, 2018-01-05:
@jmwright 

### Viewer Auto-Update

I can see this being very useful.

**django**

To use `django` as an example. when running [`./manage.py runserver`](https://docs.djangoproject.com/en/2.0/ref/django-admin/#runserver), changes will be automatically applied when refreshing a page served on *localhost*.
This works beautifully, even for very large code-bases.

**Large Models**

The `django` use-case is an on-demand service; it will only re-compile and execute the code required by the page being refreshed.

What happens with a very large & complex model?
If the model is re-generated every time a file is changed, it could essentially lock up a PC's CPU every time a file is saved.

I'm thinking of something more like an `Assembly`, say an assembly of an entire car, with all it's sub-components.
That sort of project would also need a *very large code-base*.
If the viewer is online; watching for file-changes, but only the gearbox is being displayed, then only the gearbox geometry should be generated.

This is possibly where glTF 2.0, and a web-service comes to the rescue!
In that example, a `.gltf` json file can be generated, but each `.bin` geometry buffer can be generated on demand.

**disclaimer**

OK, a lot of that was just brainstorming...
I don't know what you've already done, or what your plans are, does any of this help?

jmwright, 2018-01-05:
> Viewer Auto-Update

There's an option in the workbench to execute on save which does something similar. It's disabled by default though because (as you've mentioned) it's fairly easy to create a model that takes awhile to generate, and I didn't want users having to sit through that. That setting will be carried over into the new GUI. It sounds like what you're suggesting is to only re-generate and re-render what's changed (or being displayed). Seems like that, or some way of reducing the fidelity of the assembly would be needed to keep users from going mad. Those options seem like they'll be very challenging to implement though.

fragmuffin, 2018-01-05:
### Model Parameters

As mentioned, I've found the necessity for the *cataloguing* of `Component` instances, and their parameters... so as a result I've worked on *serializing*

What you describe could be easily achieved by:

* object -> serialize
* generating a GUI from serial data (to change data)
* deserialize changed data -> object
* display object

**Serializing**

```python
>>> from cqparts_misc.basic.primatives import Box
>>> box1 = Box()  # creates a unit cube
>>> box1
<Box: height=1.0, length=1.0, width=1.0>
>>> data = box1.serialize()
>>> data
{'class': {'module': 'cqparts_misc.basic.primatives', 'name': 'Box'},
 'lib': {'name': 'cqparts', 'version': '0.1.0'},
 'params': {'_render': {'alpha': 1.0, 'color': (192, 192, 192)},
  '_simple': False,
  'height': 1.0,
  'length': 1.0,
  'width': 1.0}}
```

That can easily be saved to file

**Deserializing**
Then it can be pulled back out (in a new python instance, or another machine) with:

```python
>>> # change data, to re-build object with different parameters
>>> data['params']['width'] = 2.0
>>> from cqparts.params import ParametricObject
>>> box2 = ParametricObject.deserialize(data)
>>> box2
<Box: height=1.0, length=1.0, width=2.0>
```

This was designed to create finite lists of object variants... such as a collection of metric bolt sizes easily available for purchase....

**More complex part**
For a deeper example: a male fastener part (screw, or bolt) serializes as:

```python
>>> from cqparts_fasteners.male import MaleFastenerPart
>>> MaleFastenerPart().serialize()
{'class': {'module': 'cqparts_fasteners.male', 'name': 'MaleFastenerPart'},
 'lib': {'name': 'cqparts', 'version': '0.1.0'},
 'params': {'_render': {'alpha': 1.0, 'color': (192, 192, 192)},
  '_simple': False,
  'drive': {'class': {'module': 'cqparts_fasteners.solidtypes.screw_drives.cruciform',
    'name': 'PhillipsScrewDrive'},
   'lib': {'name': 'cqparts', 'version': '0.1.0'},
   'params': {'_render': {'alpha': 1.0, 'color': (192, 192, 192)},
    '_simple': False,
    'chamfer': 0.3,
    'depth': 2.0,
    'diameter': 3.0,
    'width': 0.6}},
  'head': {'class': {'module': 'cqparts_fasteners.solidtypes.fastener_heads.cylindrical',
    'name': 'PanFastenerHead'},
   'lib': {'name': 'cqparts', 'version': '0.1.0'},
   'params': {'_render': {'alpha': 1.0, 'color': (192, 192, 192)},
    '_simple': False,
    'access_diameter': 5.2,
    'access_height': 1000.0,
    'diameter': 5.2,
    'dome_ratio': 0.25,
    'domed': False,
    'fillet': 1.0,
    'height': 2.0}},
  'length': 4.5,
  'neck_diam': None,
  'neck_length': 0.0,
  'neck_taper': 30.0,
  'thread': {'class': {'module': 'cqparts_fasteners.solidtypes.threads.iso262',
    'name': 'ISO262Thread'},
   'lib': {'name': 'cqparts', 'version': '0.1.0'},
   'params': {'_render': {'alpha': 1.0, 'color': (192, 192, 192)},
    '_simple': True,
    'diameter': 3.0,
    'inner': False,
    'lefthand': False,
    'length': 4.51,
    'min_vertices': 20,
    'pilothole_radius': None,
    'pilothole_ratio': 0.5,
    'pitch': 0.35,
    'rounding_ratio': 0.5,
    'start_count': 1}},
  'tip_diameter': None,
  'tip_length': 0.0}}
```

fragmuffin, 2018-01-05:
@jmwright 

> The latest version of the FreeCAD workbench has a variable pane built in like this if the script uses CQGI (show_object). It doesn't have sliders or anything fancy, but you can change the variables and re-execute the script without editing anything by hand

I must admit, I've been avoiding using `show_object` because it's not python.

So for example, I've created a `cqparts.display` module that handles the display of `cqparts` objects.
Ideally, it would simply be a pass-through to `cqparts` to handle the display, however because `show_object` is not python, this is not possible.

Is there a way to change the CQGI to be native python? (I haven't looked at its implementation in detail)

jmwright, 2018-01-05:
@fragmuffin I don't understand. Do you mean that it's not idiomatic Python?  `show_object` is implemented/handled by the build environment and calls the CQGI `build` function. CQGI is implemented in Python.

https://github.com/jmwright/cadquery-freecad-module/blob/master/CQGui/Command.py#L188

jmwright, 2018-01-05:
I should also mention that the workbench doesn't implement `show_object`, it just looks for it as a marker to tell if a script is CQGI-compliant or not.

https://github.com/jmwright/cadquery-freecad-module/blob/master/CQGui/Command.py#L156

You could certainly implement `show_object` and have it call CQGI's build function. [Here's](https://github.com/RustyVermeer/cqnb/blob/f0763b353586fe809ce5a3c9049254161ce4c80f/cqjupyter.py#L166) the way Rusty has handled it with the CadQuery Jupyter notebook. I think that Rusty's implementation is better and cleaner than mine, and I'll probably switch the workbench over to his implementation eventually.

dcowden, 2018-01-05:
@fragmuffin

> As mentioned, I've found the necessity for the cataloguing of Component instances, and their parameters... so as a result I've worked on serializing

 the idea of serializing makes good sense at the component level. I think its natural to move to a declarative storage model when you've abstracted the code away into a component.   When you're at the cadquery level, the serialized version _is_ the code. 

> I must admit, I've been avoiding using show_object because it's not python.

I'd definitely like to work through this more. CQGI was designed to make it possible for a scripts to work in most any _display_ context.  Your use case is a bit different-- you need a display, but you have an additional layer of code ( components and such) that make the abstraction less useful.  

I never imagined that _all_ cadquery code would use CQGI-- only scripts that are run in some kind of execution envirionment that provides a gui, and in which the user is copy-and-pasting scripts from one source or another. All bets are off when you're using cadquery as a library in some other project.

Your case is a bit of a weird mix. You are controlling the cadquery code, using it more like a library. But then you have the display case as well.  Since your components are not cadquery, and can be serialized declaratively, i think you should have your own display method.  Presumably, your components will always only be displayed using your library-- and whatever GUI you choose to bind your libary to is fine.

I definitely think it's a mistake to wrap cadquery with a component abstraction, but then try to keep all of the code suitable for copy-and-paste into any arbitrary GUI-- which is what CQGI is designed to do

fragmuffin, 2018-01-05:
@jmwright 

> I never imagined that all cadquery code would use CQGI [...] Presumably, your components will always only be displayed using your library-- and whatever GUI you choose to bind your libary to is fine

OK, I see... it's good for people to have a choice as well.
I may also be able to make a generator that iterates through the wrapped `cadquery.Workplane` instances within an `Assembly`...
then they could use the CQGI for their display purposes if they preferred it to whatever I come up with :thought_balloon: .
Does `show_object` work in a loop? [... tries it out ...] yep!, that'll be fine.

I'll stick with the glTF 2.0 web-based concept then...
`cqparts` has always been a side-project for something I want to use it for... and that something will use WebGL quite heavily, so I can apply what I learn there back into this project :+1:

fragmuffin, 2018-01-06:
taging @RustyVermeer

RustyVermeer, 2018-01-06:
A lot to think about here, eh!? ;)

So, I still consider myself basically just a CQ user. I'd like to offer my perspective, hopefully it's helpful.

* if we want CQ to become a larger community (I think we do), I think we really need to make the start-up experience much nicer. In my mind it means:
    * installation as simple as OpenSCAD: single .exe and a simple `sudo apt-get cadquery`  would be awesome.
    * written tutorials + video (video tutorials are quite nice for a 'before you buy', but less handy for someone trying to figure out more specific details, in my opinion)

* different contexts for CQ should continue to be encouraged, but should be made much more explicit and clear. Anyone who is totally new to CQ should be able to read a step-by-step guide for how to set things up, according to the context they want. So we should be very clear about how to install / set up for:
    * command line only
    * stand alone program with GUI (OpenSCAD as inspiration)
    * web context: Jupyter and/or some server you can self-host

    Such contexts could theoretically be handled with the same code. For example, the GUI program could also be callable from the command line in 'headless' mode. That capability would allow for someone to quickly make servers that run CQ. The docker idea isn't too far off from this.

* The GUI I think would be most useful is basically an OpenSCAD equivalent. Editor on left, console bottom right, 3d view top right (positions modifiable in user config). I think what would be the right amount of interactivity is:
    * camera manipulation. Rotate, Pan, Zoom, re-orient according to Front, Top, Right, etc
    * locate cursor in the editor, then, left-click on a face, edge, or vertex in the viewer. This auto-places the clicked object's ID as text in the editor at cursor location. Useful for filling out selectors. Not sure if this is even remotely possible with current CQ set up, but it would be fairly intuitive for the user.

For some GUI inspiration, check out [Antimony](http://www.mattkeeter.com/projects/antimony/3/). Pretty amazing project. Obviously, we would have a text editor, not a node-based visual editor... and it is quite extensive in features, I don't think we need to get anywhere close to that for users to be happy.

-----

I imagine the steps to take to achieve this would be something as follows:

1. Update documentation to give the reader a clear understanding of the current possibilities. Incl. the Docker installation method, the FreeCAD Impl, and how to run CQ on your own server (jupyter or other context)
    * Simultaneously, organize core developers. Perhaps something like this:

        * CQ current core: @dcowden (lead), @jmwright
            * core feature devs. as needed:
                 * DXF myself
                 * Assemblies @fragmuffin 
                 * etc. as community continues to grow / make improvements and suggestions
        * CQ PythonOCC core: @adam-urbanczyk (eventually merges with core above)
        * FreeCAD context lead: @jmwright 
        * Jupyter context lead: myself
        * command line context lead: tbd... or @dcowden ?
        * community management: all of us currently. @jmwright seeming to take the most initiative currently (great work!) Might be more effective to have a lead?
                * moderating the forum
                * sharing / writing tutorials as they come up
                * maybe posting on reddit or similar to get some more eyes on the library?
        * plugins: follow some sort of template, but can be developed by anyone, perhaps added to a community plugin repo.
    * basically, **core features** and **contexts** are _different._ The contexts should all ref. the same core and not implement new features themselves. Features should exist in the core and then become available in all contexts. 

2. Reach feature parity between the FreeCAD CQ and the PythonOCC CQ. 

3. For all appropriate contexts, switch to the PythonOCC CQ version. (The Docker command line and Jupyter contexts would probably be first to do this)

4. Sunset the FreeCAD-dependent code. Re-design the freecad module as needed to have it rely only on the PythonOCC CQ code. The OpenSCAD module, for example, requires the OpenSCAD.exe to exist. It essentially is just running the SCAD code and then importing it into FreeCAD. Similar idea might work for CQ?

5. Set up additional repos such as a community plugin repo, interesting projects repo, etc. 

And that's as far into the future as I'm comfortable guessing
--- 
Hmm... that was a lot of talking. I hope it's not all noise :/

dcowden, 2018-01-07:
I"m in agreement with this direction.  from your suggested steps, I might
suggest moving the priority of #4 up-- i dont want to document a bunch of
FreeCAD stuff, and freecad based installation, only to re-do it when we
dont need FreeCAD.

I agree we need a single-line installer for all platforms. If we can agree
we want that to NOT require FreeCAD, then I think its also important that
we establish what our GUI platform will be. obviously our installer will
need python+dependencies+OCC+ a gui framework.

We need to decide what we want to do on python2 vs python 3.

jmwright, 2018-01-07:
@RustyVermeer Thanks for all the thought you've put into this.

> if we want CQ to become a larger community (I think we do)

I want sustainable growth while still hopefully keeping our community as great as it is now, but yes.

> installation as simple as OpenSCAD: single .exe and a simple sudo apt-get cadquery would be awesome.

The new GUI I'm working on has the plumbing in place for this, but only x64 Linux builds have been done so far due to time constraints. I can upload that build somewhere if anyone wants it. The GUI isn't ready for public consumption (can't be used to make or display anything yet), but when it is I'll do release zips for all major platforms. Related to the question from @dcowden related to Python 2 vs 3, I've thrown all-in on Python 3 on the GUI, and we're not that far from Python 2/3 compatibility in the core library if I understand correctly. I need to get the GUI usable for everyone so that we can decide if/how it fits into the CadQuery ecosystem, and I'm not too far from having that ready.

> written tutorials + video (video tutorials are quite nice for a 'before you buy', but less handy for someone trying to figure out more specific details, in my opinion)

I get requests periodically from people wanting more video tutorials. I agree that we need more tutorials for all the different paths to using CadQuery (text and video).

> Anyone who is totally new to CQ should be able to read a step-by-step guide for how to set things up, according to the context they want.

Agreed. We need a cohesive way for people to find their entry point to the CadQuery ecosystem (command line vs Jupyter vs GUI vs whatever). It sounds like the entry point to the docs should be something like a "how do you want to use CadQuery" page with a button for each path (command line, Jupyter/server, etc).

> locate cursor in the editor, then, left-click on a face, edge, or vertex in the viewer. This auto-places the clicked object's ID as text in the editor at cursor location. Useful for filling out [...]

The things you've mentioned for the GUI operation all sound good, and are things that I've been thinking about. Mouse interaction between the code editor and 3D view is something that I definitely want to experiment with. I think the PythonOCC implementation will allow us to do a lot more in that department.

The steps for moving forward that @RustyVermeer has suggested sound good. I'm guessing that most (if not all) of the people on this thread have day jobs like me, so finding time to do everything at a fast enough pace could be difficult.

fragmuffin, 2018-01-07:
@RustyVermeer 

> installation as simple as OpenSCAD: single .exe and a simple sudo apt-get cadquery would be awesome

Yes, that would be fantastic.
I think I originally pushed `cadquery` down my list, and tried a few others before it, simply for that reason.

> The GUI I think would be most useful is basically an OpenSCAD equivalent

**Use Electron?**
I'm a fan of using web-based solutions wherever practical.
I think `cadquery` definitely needs an easily web-hosted display method, and it would be fantastic if it could have similar features to an immutable *editor*:

* camera control
* render control (wireframe, solid-wireframe, shaded, textured, that sort of thing)
* selectable meshes, faces, wires, vertices (with information about them)
* model information
* assembly information (for `cqparts`, assembly hierarchy, animation selection, and so on)
* (and more)
* all of the above could be "plugins" to the display, so the display may be as simple as possible, or include all the "bells and whistles".

One way to combine this effort into a single binary (or single cross-OS app) and avoid doing double-work on web-based displays / editors is to package a web-based solution with [electron](https://electronjs.org/).

They've got a list of example apps, many of them are free to download and install, it's difficult to tell that they're web-based... personally my favourite editor at the moment is [atom](https://atom.io/) (which runs with electron)

> written tutorials + video (video tutorials are quite nice for a 'before you buy', but less handy for someone trying to figure out more specific details, in my opinion)

I agree, video tutorials are more marketing than education.
Personally I like functional examples, and `cadquery` has lots of those :heart: 

> Hmm... that was a lot of talking. I hope it's not all noise :/

I don't think so!... although all of this is only relevant to `cadquery`... should we move the discussion to that project?...
either way it's fine; this is just a discussion thread after all ;)

jmwright, 2018-01-07:
> Use Electron?
 
@fragmuffin I experimented with an Electron-based GUI awhile back, but kept running into high CPU utilization issues. It's not hard to get it running if you want to take a look. It's just a basic GUI, it lacks almost all the features you've listed above. https://github.com/jmwright/cadquery-gui

dcowden, 2018-01-07:
what about using atom? I use it ( combined with platformio ) for microcontroller development, and it works well. It has a package management system, so it would be easily to let people install cq once they have atom.

fragmuffin, 2018-01-08:
@jmwright @dcowden 

> what about using atom?

Atom's package manager won't install things into the OS, so I think there would still have to be an `apt install cadquery` to install pre-requisites... but I think it's definitely an option!
It looks like we may run into the same problems @jmwright did... CPU load?

> I experimented with an Electron-based GUI awhile back

That's fantastic! I had no idea you'd already done that...
I think I've seen it before, but I hadn't noticed it was made of electron(s) ;)
So it has high CPU utilisation issues?, do you think that's fixable?, or do you think that is a reason to use another method?

jmwright, 2018-01-08:
@fragmuffin 
> So it has high CPU utilisation issues?, do you think that's fixable?, or do you think that is a reason to use another method?

The CPU utilization issues were mostly noticeable when rotating and panning the object in the 3D view. There was also something akin to a memory leak where after sitting for awhile the app would become sluggish and would tax the CPU when sitting idle. The first problem could be an issue with my viewer code (double rendering or something), but the second seemed like something other users were experiencing at the time with Electron in general. There was also a discussion at the time about the performance limitations of web rendering for CAD that made me doubt the viability of the Electron GUI for anything more than simple geometry. Onshape has done a nice job with their web rendering, but a search of their forums still shows that there are rendering issues related to different computers' ability to handle WebGL. However, I'm sure it's only a matter of time until the performance issues are sorted out. Too many people want heavy 3D rendering capabilities inside their browser.

My next step with the Electron GUI was to try using raycasting to select edges, faces, etc. I dropped the project before doing any of that though due to the performance issues (real and anticipated). It's very possible that any CPU utilization issues have been fixed in Electron in the past 2 years though. WebGL support and performance has advanced as well.

At the time, Dave and I saw Electron as a very attractive option, because the app can be a cross-platform desktop application, but also run as a full web app with little to no changes.

I can make you a collaborator on cadquery-gui if you're interested in digging in to it at all.

fragmuffin, 2018-01-08:
@jmwright 

> I can make you a collaborator on cadquery-gui if you're interested

I'm very interested... not sure how much time I can put into it in the short term though :(... `cqparts` has already gone over schedule.
If I didn't have to worry about money, I'd do this all day long!; I think this is fantastic.

@RustyVermeer 

> locate cursor in the editor, then, left-click on a face, edge, or vertex in the viewer.

I just had another thought... at a high level, it would be great if we could see things returned in a python prompt.
For example:

```python
>>> a = 1 + 2  # nothing shown
>>> 1 + 2  # "3" is displayed
3
>>> # similarly...
>>> cadquery.Workplane('XY').box(1,1,1).faces('>Z')
<cadquery.cq.Workplane at 0x7f18c38d2b90>
```

It would be great if that would highlight the `Workplane`, `Face`, `Solid`, or anything else that is returned.
It could be a hook into the `__repl__` of any class, communicating back to a local service, but is only enabled if it's explicitly connected via a library-wide hook.
However, `__repl__` isn't just called by a python console, so that could be messy... there may be a way to hook the python console itself.

dcowden, 2018-01-08:
@RustyVermeer i love the idea of making sure you can show an object at the command line! maybe instead of relying on __repl__ we could just  have a function that's exported by the GUI that does it, like debug(..)  If we get that working, I can include that as a part of the cqgi api.  Right now, cqgi really only contemplates the evaluation of an entire script in one shot-- not the idea of running a script and then keeping the context alive and allowing interactive user input. But I could add that.

jmwright, 2018-01-08:
@fragmuffin You have been added as a collaborator. I was learning three.js and Electron while I was developing that project, so hopefully the high CPU utilization is due to a simple bug. Even running at idle with no 3D object displayed the app runs all of the cores on my I7 laptop at about 50%. I took another look through the [3D viewer code](https://github.com/jmwright/cadquery-gui/blob/master/assets/js/cqlib/mviewer.js) and didn't see anything obvious, but I never learned three.js in-depth and it's been 2 years. If I remember right that viewer code was borrowed from the ParametricParts.com codebase, but was modified quite a bit to make it work for the app. If we can get the CPU issue figured out, I'd like to develop a set of stress tests to see how far we can push the tech.

> If I didn't have to worry about money, I'd do this all day long!; I think this is fantastic.

Glad you're enjoying it. I think there's a group of us that would love to be able to devote more time to developing (and using) CadQuery. This year will be my 4th as a CadQuery contributor, and I still enjoy it a lot.

closed 2018-01-15 by fragmuffin