cqparts/cqparts #43: Fasteners Catalogue

fragmuffin, 2018-01-26:
Create a catalogue of fasteners

- [ ] A _what-you-can-probably-buy-somewhere_ catalogue
- [ ] Scraped from a website

fragmuffin, 2018-01-26:
**Issue with Scraping**
The issue with this is that most online hardware shopping websites don't have enough information, and the information that is shown varies too much to be valuable.

Bunnings (an AU hardware chain) was used as a case-study, see [bunnings.py](https://github.com/fragmuffin/cqparts/blob/master/src/cqparts_fasteners/catalogue/scripts/bunnings.py) for (partial) implementation.

**Need a better source**
I'd still like to create a catalogue from a website, I just need to find a site that has a better maintained database.

fragmuffin, 2018-01-26:
A potential candidate: https://www.boltdepot.com

jmwright, 2018-01-26:
Very cool idea to scrape the sites like that. I've run into the same problem. Information is spotty and inconsistent on most vendors' sites.

fragmuffin, 2018-01-26:
**Imperial**
I'm also having a lot of trouble with imperial standards, [UFS](https://en.wikipedia.org/wiki/Unified_Thread_Standard) switches from numbers, to measaurements, but people rarely use a `#` to signify they're using number codes, so what happens when you get to 1 inch?, does an undecorated `1` mean `#1` or `1"`... imperial drives me crazy.

[boltdepot](https://www.boltdepot.com) seems to have the decency to list all fastener's measurements independent of its name, which is fantastic.

fragmuffin, 2018-01-26:
@jmwright 
I do understand it though; the site is for their customers, and most of their customers are human (if not all :wink: ).
The data-entry is also probably done by a different person every time, so their methods will vary.
Another possibility is that their data is intentionally difficult to decipher so their site is difficult to scrape :worried: .

Regardless... this initiative is a detour from my main goal of releasing this library, so I just want to knock over an easy site first. Hopefully I can recruit someone in future to build a scraper that's a bit more forgiving with product data.

fragmuffin, 2018-01-31:
I've now got 2349 bolts, and 544 woodscrews from [boltdepot](https://www.boltdepot.com/) in catalogues

With commit 8775735 the _bolts_ and _woodscrews_ catalogues can be scraped and built from scratch with:

```bash
cd ~/prj/cqparts/src/cqparts_fasteners/catalogue/scripts
./boltdepot.py --catalogues bolts,woodscrews scrape build
```

Then to pull a fastener from a catalogue:

```python
import os
import cadquery
import cqparts
from cqparts.display import display
import json

import logging
cadquery.freecad_impl.console_logging.enable()
log = logging.getLogger(__name__)

filename = os.environ['HOME'] + '/prj/cqparts/src/cqparts_fasteners/catalogue/boltdepot-bolts.json'

catalogue = cqparts.catalogue.JSONCatalogue(filename)
i = catalogue.get_query()

obj = catalogue.get(i.id == '246')
log.info(json.dumps(obj.serialize(), indent=2))

display(obj)
```

fragmuffin, 2018-02-02:
I started creating a "native" catalogue of nuts, bolts, screws, and anything else you might need.
But I have to be honest... I can't stand to look at any more nuts, bolts, thread standards, pitch lookup tables, or wrench sizes any more this week!

The boltdepot catalogue is *very* extensive, and real data!, so I'm also questioning whether the _native_ option is worth keeping.
I've committed the `native.py` script in 8d53f24 for reference, but I'll be deleting it and closing this issue now.

fragmuffin, 2018-02-02:
The boltdepot scraper is not finished, I'd like to add many more products, I've just started with the basics.
Issue #45 created to track that.

closed 2018-02-02 by fragmuffin

jmwright, 2018-02-02:
Does it hurt to have a native option, even if it's only a reference? I don't think you need to cover every case at this point, but if somebody else wants to later, they'd have an example to start with.

fragmuffin, 2018-02-03:
The [`boltdepot.py`](https://github.com/fragmuffin/cqparts/blob/master/src/cqparts_fasteners/catalogue/scripts/boltdepot.py) script is good enough to serve as an example, the code for each screw, bolt, and nut is easy enough to find and each one takes ~70 lines of code.

I was also thinking that a list of _fasteners-you-could-probably-buy-somewhere_ is a bit pointless when there's real data available.

A catalogue like that will quite likely end up with some components no sold by anyone, and other common fastener dimensions that are completely missed by a looping algorithm.
It started to become an exercise in statistics, which is a very long stray from my main goal at this point; to release the base `cqparts` library.

I need to spend my time making content in other libraries; more examples that people can use.
Then once it's released and being used, I can recruit other people to build content... that's the only way this library idea is going to grow. If I'm wrong, and it's only me, then I'd like to know sooner rather than later, because I can't be the only author, it just won't work.