cqparts/cqparts #46: Web Display service improvements

fragmuffin, 2018-02-03:
**Problem**
At the moment:

```python
from cqparts.display import web_display
from cqparts_misc.basic.primatives import Cube
web_display(Cube(size=10))
```

Will start a `http` service, and open a browser. It will then wait for the user to manually halt the process with `Ctrl+C`... a rudimentary and gritty implementation.

**Potential Solution**
Create a more friendly experience for the user:

* non-blocking call to `display` (eg: `http` service stays alive).
* open multiple `display` calls from a script in the same window.
* on first `display` call from a new process, clear world.

fragmuffin, 2019-01-25:
A company called [CADenas](https://www.cadenas.de/) has a commercial product called [PARTsolutions](https://partsolutions.com/) that implements a very nice web-based model viewer.

It doesn't seem to implement a model hierarchy to show/hide selected trees, but it implements selectable/togglable features such as:
[image]
 - perspective
 - slice
 - turntable
 - isometric view
 - show dimensions (doesn't always work)
 - render: wireframe
 - render: shaded with edges

Just to mention a few.
I'm not suggesting we copy it verbatim, but it's a good source of inspiration.

zignig, 2019-01-25:
hmmm ponderment , will investigate. The cqparts-server uses a  vuejs application. It's a bit stale at the moment and I need to port it over to bulma css. It's currently using semantic-ui and it's kind of heavy.

I have been thinking about a more comprehensive we application ( with source editing ) for some time now...