cqparts/cqparts #48: Discussion : Administering a level of Simplification

fragmuffin, 2018-02-06:
I believe it's obvious that a level of optional model simplification is necessary to render high complexity models.
However, I don't have a grand design of **how** this "simplification" will be administered to a model at a high level.

# Possible Aproaches

To use the [toy car](https://fragmuffin.github.io/cqparts/doc/tutorials/assembly.html) as an example:

```text
car
    ├○ chassis
    ├─ front_axle
    │   ├○ axle
    │   ├○ left_wheel
    │   └○ right_wheel
    └─ rear_axle
        ├○ axle
        ├○ left_wheel
        └○ right_wheel
```

## Maximum Level

Limit the number of levels an assembly's hierarchy can reach.

* `< limit` : full complexity
* `= limit` : simplified model
* `> limit` : omitted

So limiting the car to a single layer, the `front_axle` and `rear_axle` would be simplified (if available, otherwise omitted)

```python
# something like...
car = Car(simple_level=2)
```

## `fnmatch` mapping

Map an assembly's nested component `find()` key, so if it matches, it's simplification is forced

```python
simplified_keys = [
    'front_axle.*',
    'rear_axle.*',
]
# then something like...
car = Car(simplified_keys=simplified_keys)
```

gntech, 2018-04-24:
Perhaps it is possible to do what is done in some other CAD-softwares; render a "lightweight" geometrical representation of models instead of the actual exact B-REP.
The format used is an approximate faceted (polygon) graphic format. NX uses JT-format and Catia uses something called CGR.

One inherent drawback is that measurements taken from entities in the lightweight 3d-model is not exact but approximate.

https://en.wikipedia.org/wiki/JT_(visualization_format)
https://en.wikipedia.org/wiki/Polygon_mesh

fragmuffin, 2018-04-24:
Thanks @gntech , I agree; it's definitely worth mentioning.

At the moment I've got a very simple version of this in the `cqparts.Part` class; the default simplified model (generated by `cqparts.Part.make_simple()`) is a bounding box of the full fidelity model (your basic 12 poly' rectangle).

However, this issue is addressing _how_ best to indicate what components in a hierarchy are "simplified", and which are to be full fidelity.

I've raised #61 as a more specific topic... it's a good one!