cqparts/cqparts #54: Require cadquery master release

fragmuffin, 2018-03-06:
current `master` of `cqparts` requires `cadquery>1.0.0` which isn't released yet.

@dcowden @jmwright:
When is the next release of `cadquery` planned?

fragmuffin, 2018-03-06:
also: do you see any reason why this shouldn't be released?...

I'm already 2 months past my target [sigh], so if there are any _significant_ changes required before release, then it'll have to wait.

jmwright, 2018-03-06:
@fragmuffin We don't really plan our releases. Right now I've got probably 2 more hours on fixing/adding tests for Shapes, and I kind of wanted to take a look at the following issue before doing the 1.1.0 release.

https://github.com/dcowden/cadquery/issues/186

> also: do you see any reason why this shouldn't be released?...

I've been following your progress and it seems like cqparts should be ready for an initial release.

fragmuffin, 2018-03-06:
@jmwright
Sounds good.
Also: would you consider pre-releasing to the PyPi test site?
https://testpypi.python.org/pypi
It's meant for testing the deployment process, but it's also used for dependent packages, like `cqparts`

fragmuffin, 2018-03-06:
Although, now I look into it further, that still won't help me because 2 other requirements, `Jinja2` and `tinydb`, aren't on the test site either.

I'll adjust my strategy here. I think I need to install pre-requisites from the official site instead.

jmwright, 2018-03-09:
So this is a blocker for #8 now, correct?

fragmuffin, 2018-03-09:
@jmwright no, #8 was to create the deployment automation... I'm fairly confident that it will work, so #8's been closed.
However this is a blocker for release of `cqparts`, and any lib depending on it: `cqparts_*`.
I can't release a library if people can't install it in working condition.

fragmuffin, 2018-03-15:
@jmwright @dcowden
What is involved for a `cadquery` release?
How long does it take?

dcowden, 2018-03-15:
@fragmuffin  it is not hard. we change the metadata , and then run a couple of commands...

jmwright, 2018-03-15:
@fragmuffin Tagging a release on GitHub should push it to PyPi.

dcowden, 2018-03-15:
oh @jmwright  thanks for correcting me, i totally forgot we automated that. yay us!

fragmuffin, 2018-03-15:
@jmwright @dcowden
The docker image too.

Do you mind if I ask why the are so few `cadquery` releases?
It looks like it's been in 1.0 for some time, but it's had many PRs since then.

I ask because I want a release so I can close this off and release `cqparts`.
Could `cadquery` be released before dcowden/cadquery#186 is fixed, then again after.

jmwright, 2018-03-16:
> Do you mind if I ask why the are so few cadquery releases?

There's no official reason. It's probably simply that a lack of rules/process around when to do releases leads to a lack of discipline. It's only been relatively recently that CadQuery development has accelerated some due to contributions from new collaborators. Before that, there could be months at a time with no commits, which led to widely spaced releases. There might be some of the habits formed during that time left over. I don't really know how we'd do our releases unless we increment the correct part of the version number (major, minor, bug) with every PR and do a release.

> Could cadquery be released before dcowden/cadquery#186 is fixed, then again after.

Yes. My reason for wanting #186 in the 1.1.0 release was that it adds a little pressure to get it done. #186 can be part of a 1.1.1 release though.

Let me try to finish up the shape.py tests tomorrow and then we can do a release.

jmwright, 2018-03-16:
I decided that the remaining tests in shapes.py could wait until the 1.1.1 release.

https://github.com/dcowden/cadquery/releases/tag/v1.1.0

https://pypi.python.org/pypi/cadquery

Travis is complaining that the deployment to PyPi failed, but it didn't. All of the files and documentation appear to be intact.

fragmuffin, 2018-03-16:
Oh! Excellent!
I'll release `cqparts` tomorrow evening (I've got some boards that need soldering urgently first)

fragmuffin, 2018-03-18:
**Release Checklist**

- **PyPI Packages**
    - [x] `cqparts`
    - [ ] `cqparts_bearings`
    - [ ] `cqparts_fasteners`
    - [ ] `cqparts_misc`
- **Other Tasks**
    - [ ] github release (one per library?)

fragmuffin, 2018-03-18:
problem to fix before content libs are released... currently I'm using the root `README.rst` as the content for PyPI, however that readme is specific to `cqparts`, and misleading for content libs.

fragmuffin, 2018-03-19:
Done!, all released!

I think the github release side of things can be improved.
The way it's set up at the moment is sustainable, but will be hard for people to follow as more releases come along... the  disadvantage to having a single repository host multiple libraries I guess.

@jmwright @dcowden @adam-urbanczyk
I just want to thank you all, you've been so helpful and positive while I've been learning and building these libraries.
Hopefully `cadquery` and `cqparts` by extension has a long life, and there are many more improvements to make, and things to learn.

I'll be making small updates while I use the libraries, but my involvement will be less over the next few months while I do some "real" work :wink:.

I've got to find a way to pay the bills, or I'll have to stop open-source work almost entirely.
But like I've said before, if money wasn't a problem, I'd probably just do this for the rest of my days :heart:

closed 2018-03-19 by fragmuffin

dcowden, 2018-03-19:
@fragmuffin  It's been a real pleasure having you as part of the team! thanks so much for the contributions!

As you've been working, cadquery has been gaining stars, we're up to > 225 now! that's pretty cool.

adam-urbanczyk, 2018-03-21:
@fragmuffin  thanks for your contributions and good luck with your real work!