cqparts/cqparts #56: Test FreeCAD 0.17 Python 2.x bindings

fragmuffin, 2018-03-07:
Currently testing is split into 2 environments:

- FreeCAD v0.16 :: Python 2.x
- FreeCAD v0.17 :: Python 3.x

But does FreeCAD 0.17 have Python 2.x compatible bindings?
I assume it does, so I should also be testing that.

- [ ] FreeCAD v0.17 :: Python 2.x

jmwright, 2018-03-07:
> But does FreeCAD 0.17 have Python 2.x compatible bindings?

It does for now at least. I'm not sure when FreeCAD is going to completely transition to Python 3, but it hasn't happened yet.

jmwright, 2018-03-07:
I don't know for sure, but I would guess yes. At some point it will become more work than its worth to keep all the Python 2 stuff working. I'm sure the answer is somewhere in this 52 page forum post that's been ongoing since 2015.

https://forum.freecadweb.org/viewtopic.php?t=12534

fragmuffin, 2018-03-07:
Haha!, no, I'm good.
I made this issue a bit too quickly I think...
If FreeCAD **are** going to support python 2.x after [it expires](https://pythonclock.org/) then I wouldn't agree with that decision anyway... I'll close this issue as a `wontfix`...

good talk ;)

closed 2018-03-07 by fragmuffin