cqparts/cqparts #64: STEP Assembly import for multi-part STEP files

fragmuffin, 2018-05-01:
Currently no support for multipart STEP files importing to an `Assembly`.

Resulting code would be:

```python
from cqparts import Assembly
obj = Assembly.importer('STEP')('multipart.step')
```

For a single body STEP file, the resulting `Assembly` instance would have a single component.
If the STEP file is multipart, the `Assembly` will have multiple components.

Thanks to _Drew Marschner_ for pointing this out in the cadquery group.

fragmuffin, 2018-05-04:
cqpargs google group discussion: https://groups.google.com/forum/#!topic/cadquery/nZfqMEUX-Fg

fragmuffin, 2018-05-04:
Multipart STEP imports now working.
Although names & colours are lost
To be continued in issue #72

closed 2018-05-04 by fragmuffin

kevykibbz, 2022-07-12:
i get this error when i try to run from cqparts import Assembly

Traceback (most recent call last):
  File "<console>", line 1, in <module>
  File "C:\Users\Teacher-5F84DF\Desktop\web\webprojects\lib\site-packages\cqparts\__init__.py", line 77, in <module>
    from .component import Component
  File "C:\Users\Teacher-5F84DF\Desktop\web\webprojects\lib\site-packages\cqparts\component.py", line 2, in <module>
    from .constraint import Mate
  File "C:\Users\Teacher-5F84DF\Desktop\web\webprojects\lib\site-packages\cqparts\constraint\__init__.py", line 12, in <module>
    from .mate import Mate
  File "C:\Users\Teacher-5F84DF\Desktop\web\webprojects\lib\site-packages\cqparts\constraint\mate.py", line 3, in <module>
    from ..utils.geometry import CoordSystem
  File "C:\Users\Teacher-5F84DF\Desktop\web\webprojects\lib\site-packages\cqparts\utils\__init__.py", line 18, in <module>
    from .geometry import CoordSystem
  File "C:\Users\Teacher-5F84DF\Desktop\web\webprojects\lib\site-packages\cqparts\utils\geometry.py", line 6, in <module>
    import FreeCAD
ModuleNotFoundError: No module named 'FreeCAD'

what should i do?