cqparts/cqparts #66: Integration As Third Party Library

jmwright, 2018-05-02:
Figured I'd start an issue for this because I'm sure I'll have multiple questions before this integration is done.

### Background
I'm working on including cqparts with the CadQuery module for FreeCAD. One of the core philosophies of that module/workbench has always been that it's batteries included. This means that I include copies of all the required libraries and then manipulate the Python path so that FreeCAD's Python interpreter can find them.

### Question
For cqparts I've added a `ThirdParty` directory, and I've cloned the entire repo there. Right now I set the Python path so that `ThirdParty/cqparts/src` is searched. However, I noticed there was a gitignore file in the `src` directory. Do you have a mechanism already in place that would allow me to just grab the contents of the `src` directory, and skip all the higher level items (like docs)? I can look into partial checkouts, but I'm not sure they're the answer.

fragmuffin, 2018-05-02:
@jmwright yeah, I can see how cloning the whole repo could get messy.

>  Do you have a mechanism already in place [...] to just grab the contents of the `src` directory [?]

I do... well, not exactly... the deployment strategy takes care of that.
But that's all done by the `setuptools.setup()` method.

Could you get the most recent version deployed to PyPI and extract it, instead of cloning?

jmwright, 2018-05-02:
> Could you get the most recent version deployed to PyPI and extract it, instead of cloning?

Yeah, I can. I have a script in the repo for collecting libraries that are installed using pip, but stopped using it because it made development setups messy in certain cases. I guess I need to create a new script that just downloads and extracts the packages without ever installing them locally.

jmwright, 2018-05-02:
It looks like not everything that's in the repo's src directory is represented on Pypi. Is that because the missing ones are just placeholders?

fragmuffin, 2018-05-02:
@jmwright 

> I guess I need to create a new script that just downloads and extracts the packages without ever installing them locally

I think that's probably best. I had to do this in an old job a lot. It was a good way of creating a localised distrobution of "core" Python libs without creating my own package manager.
It seems that's the sort of thing you're doing too.

> It looks like not everything that's in the repo's src directory is represented on Pypi. Is that because the missing ones are just placeholders?

Yep; spot on. Some of those libs are empty or just aren't ready to be released.
One way would be to download and extract `cqparts` + everything from pypi of the form `cqparts_*`.

    pip search cqparts-

jmwright, 2018-05-04:
@fragmuffin `MarkupSafe-1.0.tar.gz` is downloaded when I do `pip download` on the files, whereas everything else downloaded is in .whl archives. Is MarkupSafe needed for the normal operation of cqparts?

jmwright, 2018-05-04:
> Is MarkupSafe needed for the normal operation of cqparts?

Ignore this question, I figured it out.

fragmuffin, 2018-05-04:
Ok, bit now I'm curious... I don't even recognise that file.

jmwright, 2018-05-04:
cqparts scripts won't run without it installed.

jmwright, 2018-05-05:
cqparts is now embedded with the [CadQuery FreeCAD module](https://github.com/jmwright/cadquery-freecad-module).

The first example from the readme works great, and the cqparts.display() function even works, although it does not use CQGI and so some features are not available (like execution timing).

closed 2018-05-05 by jmwright

fragmuffin, 2018-05-05:
Oh, I see, `MarkupSafe-1.0.tar.gz` is required by `Jinja2` which is used for `cqparts.display.web_display`

> the cqparts.display() function even works, although it does not use CQGI

Yes, I've been avoiding the CQGI because it forces a layer over the top of python.
It parses the python code based on [this logic](https://github.com/jmwright/cadquery-freecad-module/blob/master/CQGui/Command.py#L156), then injects code into the script (changing it), then executes that.
Because `cqparts` is a major abstraction from the normally seen single-file approach, I'd much rather avoid this pathway.

Also, `show_object` isn't imported, so as soon as a script is QCGI compatible, it's no longer Python compatible. That's a big red flag for me.

Can I use the CQGI without triggering the code regeneration?
Can we just import `show_object` from somewhere and call that?

fragmuffin, 2018-05-05:
@jmwright after all that... I forgot to...

Thank you!, it means a lot to me that `cqparts` is being developed along side `cadquery`.

jmwright, 2018-05-05:
> Can I use the CQGI without triggering the code regeneration?
> Can we just import show_object from somewhere and call that?

I think those are questions we need @dcowden for. Dave, please see the comments from @fragmuffin above about CQGI.

From my perspective there's no requirement to use CQGI. cqparts.display() works just fine, which I was excited to see. Your examples should work in the FreeCAD module unmodified.

> Thank you!, it means a lot to me that cqparts is being developed along side cadquery.

You're welcome. And thanks for the contribution of cqparts to the community. I'm not sure why I didn't think to include cqparts in the FreeCAD module sooner. Recent discussions in your issue tracker finally made the light come on.

jmwright, 2018-05-05:
@fragmuffin Just thought of this too. It would be helpful if you somehow announce new releases so that I'll know to update the cqparts libs in the FC module. I have started doing announcements on both the GitHub issue tracker and the Google Group to try to cover as many users as possible.

fragmuffin, 2018-05-05:
@jmwright 

> It would be helpful if you somehow announce new releases so that I'll know to update the cqparts libs in the FC module

Yes, I can do that... I just added that to my deployment readme: 5b6acd1e0087ceb3c670efaa76dc8b22e37a8715

dcowden, 2018-05-05:
@jmwright  @fragmuffin  CQGI  is designed to handle two problems.

The first is exposing parameters and values. CQGI adds ability for an executing environment ( like cadquery freecad module or a website ) to change parameter values without needing the developer to do a bunch of work to define the variable metadata.  IE, a script with the single line:

`width=1.0`

is enough to allow CQGI to understand that this is a floating point variable that can be modified. Imagine that a user uploads a python script to a website, and you want the object it produces to be customized, without diving into the source code and changing it. That's what CQGI is designed to handle.

cqparts has much more formal abstractions to handle this, so it doesnt add any value at all. If you're building up an assembly with cqparts, you're definitely changing the source code.

The other value cqgi provides is a standard way for a script to interface with an executing environment that can visually display objects ( IE, the show_object function).  This need is still present, even with CQParts, and is critical IMO.   Without this abstraction, every tool that visualizes models must actually change the source code to match their environment, and/or every library must provide custom methods for every executing environment.  This is not what we want.  When we moved to CQGI, we gained a huge benefit-- all of the examples published 'just work' both locally and in CQFM without modification.

> Also, show_object isn't imported, so as soon as a script is QCGI compatible, it's no longer Python compatible. That's a big red flag for me.

I understand this concern.  The simple fix would be to require importing it, and providing a default implementation in cq itself when scripts run without a UI.  That's how things work now.

Bottom line: I think its a big benefit to all if there's a standard interface that all cq like libraries use to interface with executing environments to show objects.  I think the show_object method is fine as it exists now-- except maybe we can change the semantics of how it is imported and how it works.  We can get the benefits of the standard API without requiring that it works the way it does now.

fragmuffin, 2018-05-05:
@dcowden 

> Without this abstraction, every tool that visualizes models must actually change the source code to match their environment, and/or every library must provide custom methods for every executing environment, and/or every library must provide custom methods for every executing environment. This is not what we want.

What's your opinion on [`cqparts.display.display`](https://github.com/fragmuffin/cqparts/blob/master/src/cqparts/display/__init__.py)?
It switches its display method based on [the detected environment](https://github.com/fragmuffin/cqparts/blob/master/src/cqparts/utils/env.py).

I've been meaning to make that more flexible, details in #77

dcowden, 2018-05-05:
I like the idea, but what I don't like about it is that every project has to implement their own version of essentially your same code. And, you need to anticipate and code for every possible executing environment.

We need A glue layer that exposes consistent interfaces from both perspectives.

Ie, everyone (libraries and executing environtmts) should import cqgi. Libraries that want to display objects call cqgi.display or something,and executing environment implement it.

fragmuffin, 2018-05-05:
I think I see now, that's where #77 will expand capabilities of `cqparts.display`... future-proofing it a bit.

dcowden, 2018-05-05:
No offense, but I'm not sure that any implementation of display logic Inside of cqparts is right, because it then can't be shared with cadquery Core or cadquery freecad module.

I think cqgi should be neutral. Cqparts should import cqgi and ask it to display things. In an ideal world there are lots of display options provided

fragmuffin, 2018-05-05:
> No offense, but I'm not sure that any implementation of display logic Inside of cqparts is right

None taken, I think this is an excellent discussion that's probably overdue.

> Cqparts should import cqgi and ask it to display things.

So I guess that brings me back to my original question:
Can I use the CQGI without triggering the code regeneration?

dcowden, 2018-05-05:
> So I guess that brings me back to my original question:
> Can I use the CQGI without triggering the code regeneration?

Yes, this is the key question. I think the answer today is 'no', but  the answer should be 'yes'

Can you review the [cadquery gateway interface spec](https://dcowden.github.io/cadquery/cqgi.html), specifically, the script side, and lets iterate on the right interface.

As it stands now, CQGI exposes two methods to a script:  show_object()  and debug(). The script calls show_object and gives cq object as well as a dictionary each time. debug also takes an object and a dictionairy, and is designed to allow script authors to highlight various features when debug is enabled. ( for example, to highlight them in red).

I think these interfaces are good as-is, with the exception that show_object and debug magically appear, and maybe they should need to be imported.

The execution environment side is the one that doesnt work well for cqparts.  The script environment expects to be able to parse a script, and then exposes a CQModel object that represents a CQModel that can have its parameters manipulated, and then can be executed by calling its build() method. 

cqparts could implement these interfaces, or i'm open to designing new interfaces if that's what makes sense.

The goal is that a script executing environment ( for example, CQFM ) should not have to know whether a script is a cqparts script or a regular old python script--- as long as the script imports cqgi and calls show_object, it should work.

fragmuffin, 2018-05-07:
@dcowden sorry for the delay; I still want to respond to this, I just have to find the time ;)

dcowden, 2018-05-07:
No worries! take your time, we're all volunteers!

fragmuffin, 2018-05-10:
The typical pattern followed for creating a parametric model is
- define parameters
- build model (with optional paths, like debug)
- display model

However, the problem with this method is that the request to display is done **after** the model has been made.
Therefore the only way to change parameters is to intervene at a higher level (ie: higher than the Python interpreter running it, which is what `cqgi` does now).
So I think there **is no clean solution** if everything is run in `__main__`.

However... _does_ everything have to be run in `__main__`?

The above steps to create a model look a lot like any function:
- define parameters
- process parameters
- output

So if a cube in the current `cadquery` mindset is made with:

```python
import cadquery

# define parameters
size = 10

# build model
cube = cadquery.Workplane('XY').box(size, size, size)

# display model
show_object(cube)
```

Instead, could it be:

```python
import cadquery
from cadquery.cqgi import debug, show_object

def cube(size=10):
    obj = cadquery.Workplane('XY').box(size, size, size)
    debug(cube.faces(">Z").workplane())
    return obj

show_object(cube)
```

now `show_object` takes in a function that:
- defines only named parameters, all with default values
- returns a `cadquery.Workplane` instance

`show_object` can use `inspect` to look at the function's parameters and their default values.

I know this is one step toward `cqparts`, but I don't see any other way that keeps the code strictly python.

dcowden, 2018-05-10:
@jmwright  @fragmuffin thanks. Let break this down and take the conversation in steps, since its pretty complex.  First, Are we  in agreement that there must be four phases of model resolution:

(1) read model -- we read the model ( somehow), and understand what parameters and such it offers
(2) set parameters/options -- we allow the user and/or executing environment to manipulate the exposed params and options
(3) execution -- we run the code ( this takes a while for complex models )
(4) render output -- we show the graphical result. This may be on a different process than the execution, or may be in process.

Are we all in agreement that solutions would follow these 4 phases?

jmwright, 2018-05-11:
@dcowden That sounds right to me.

jmwright, 2018-05-11:
@fragmuffin As a side note, I've mentioned cqparts in the reworked FreeCAD module documentation [here](https://github.com/jmwright/cadquery-freecad-module/blob/master/docs/usage.md#cqparts).

fragmuffin, 2018-05-11:
@jmwright nice!

fragmuffin, 2018-05-11:
@dcowden yep, I also think that's right.

dcowden, 2018-05-12:
ok cool!   so given these 4 phases,  lets also define a few terms:

the **script code** is executable code that is interpreted in step 1 and executed to produce object output ( for now lets assume this output is in the form of OCC shape objects )

the **executing environment** is code interfacing with the user, providing visual representations of the output OCC shape objects, and interfaces to view the script code and, possibly, to change the parameters it exposes.

Right now, I think we have these  **executing environments** in mind:
  * cqfm, which loads a script from the filesystem, and displays objects on screen using freecad
  * a web based viewer, which stores scripts in a database, and then displays objects in a users' browser after running the script on the server,
  * the cadquery documentaion, where the scripts run and then are used to produce images along with their original source
  * a developer's machine, where the user may essentially not have an executing environment at all ( or, more precisely, has whatever executing environment comes with CQGI by default )

What we need is a set of interfaces that allow the 4 phases, but without the **script code** and the **executing environment** directly touching each other. If we can achieve this, we know that both parts are independent 

From a cqparts viewpoint, there are two problems with the current approach:

1. the show_object method is magical. It is not imported, and in the best case it must be implemented by the **executing environment**, which is a problem 
2. the current parse method cannot be used to produce CQModel objects for any types of scripts other than those for which it is designed ( standard python scripts that call show_object )

In this example by @fragmuffin 

```
import cadquery
from cadquery.cqgi import debug, show_object

def cube(size=10):
    obj = cadquery.Workplane('XY').box(size, size, size)
    debug(cube.faces(">Z").workplane())
    return obj

show_object(cube)
```

We've solved issue #1 with an explicit cqgi import.   An executable model is represented as a function-- the names and values are the default parameters, the return value is the object.  The metadata would be computed by inspecting the functions' parameters.  

This solution isn't too bad from the **script** side.  There are some peculiarities ( for example, what does it mean to call show_object several times with the same function or with different functions).  

But more importantly, I think this solution leaves too much work to the executing environment.  All executing environments would need to do the same code to inspect the function, learn its parameters, store a reference to the function, and then call it when requested by the user.  I'd like cqgi to do this work as well, so that it is common and shared.  

Another limitation is that it doesnt provide much support for the executing environments' support for showing data to the user.  CQGI's  [BuildResult](https://github.com/dcowden/cadquery/blob/master/cadquery/cqgi.py#L144) contemplates a bunch of information the end user probably wants-- the execution time of the script, whether it was successful, and, of course, the objects.   I think it is important that this is a full object, because it's likely that it might be serialized-- for example if the environment wishes to save a rendered set of objects without re-calculating them, or to send it over the wire in the case of a web based solution.

One final limitation is that the notion of passing a function for direct execution by the executing environment omits security concerns.  cqgi today adds some [extra machinery](https://github.com/dcowden/cadquery/blob/master/cadquery/cqgi.py#L103) that allows the executing environment to change the locals and globals in scope to narrow how dangerous a script may be-- again, with web execution environments in mind.

We need a way to clean up the script side, but without throwing out all of the work cqgi is currently doing to make it easier for the executing environment.

I need to think a bit more about this, but honestly I think most of what is currently in cqgi is fine. I like the abstractions that BuildResult, CqModel, and ShapeResult provide.   I think we just need more methods to convert scripts into CqModel objects for execution.

I'll try to noodle through this and come up with an idea.

zwn, 2018-05-21:
> now `show_object` takes in a function that:
> 
>    * defines only named parameters, all with default values
>    * returns a `cadquery.Workplane` instance
> 
> `show_object` can use `inspect` to look at the function's parameters and their default values.

This sounds very reasonable (let me add I am a programmer and not a CAD expert). 

That is what hit me when looking at `cqgi` -- it is trying really hard to reimplement function calling! Only the parameters appear to be global variables, the return value is accumulated to a list during the execution and the function has no name.

> `show_object(cube)`

That's another way of saying that the function `cube` should be decorated by decorator `show_object`.

It is funny how cadquery tries to really hard to capture the design intent of the CAD but the structure of the script is actually quite alien to someone used to working with python all the time.

My take would be probably similar to

```python
import cadquery
from cadquery import cqgi

@cqgi.part
def cube(size=10):
    obj = cadquery.Workplane('XY').box(size, size, size)
    cqgi.debug(cube.faces(">Z").workplane())
    return obj
```

Where `cube` is a factory function for cubes. When called, a shape is returned. This shape could be further copied/cloned and each copy would have its own position but shared the shape. The resulting shape would be fully defined so it can have a technical drawing. I am referring to #90 here.

dcowden, 2018-05-21:
i like the clarity of the above, when the part in question is going to be used from somewhere else.  One of the goals of the original cqgi spec was to reduce boilerplate to a minimum. This resulted in removing the imports and, also allowing the author to create just a plain old python file, not a file that declares a function. 

Early cqgi designs had scripts that looked like:

```
import cadquery
from cadquery import cqgi

def build(**kwargs):
    obj = cadquery.Workplane('XY').box(size, size, size)
    cqgi.debug(cube.faces(">Z").workplane())
    return obj
```

This is extremely similar to the proposal above, with the exception of using a pre-defined method name instead of an arbitrary one with a decorator.   Of course it is limited because one python file cannot contain > 1 function. 
What motivated the move away from that was that it seemed bad to have so much boilerplate for a simple thing.  Maybe that was the wrong idea,because the average person interested in cq has been much more 'geeky' than i originally intended/hoped. I think clearly the community doesn't like the magic of the current way of working.

Another motivating factor originally was that I wanted it to be possible to run a cq script from the command line.  the above would have to have an additional invocation of the function at the end to actually run it, which is kind of annoying.  But, at some point along the way I lost this functionality anyway by removing the need to import show_object. I'd like to recover this ability as a part of the solution.

I could get behind the above @zwn's proposal above.  We would accept that a cadquery script MUST be executed by a cqgi compliant container.  To retain the ability to run cq scripts from the command line, I could implement a main function in cqgi that runs a script, such as:

python cqgi.py /usr/src/cube.py

So we'd have the above interface exposed by scripts,  and then we'd have the following cqgi executors:
   * command line
   * sphix doc
   * cadquery freecad module
   * adam's new gui
   * zwn's new gui

And then we'd need to go through all of the examples and change their syntax.  

I'm ok with this direction but I need a +1 from a lot of folks-- that'd include @fragmuffin @gntech @fragmuffin and @adam-urbanczyk