cqparts/cqparts #67: Best practice for hierarchial design

gntech, 2018-05-02:
I have migrated a design for a coffeetable from plain freecad python script into a cadquery/cqparts script.

I wonder what the best practice is to pass values between parts in an assembly.

Some of the dimensions for the parts depend on each other. The coffee table assembly needs only length, width and height but calculates some other variables that are used other parts of the table.

I have studied your ball bearing script.
You collect variables for the ball bearing like so:
`outer_diam = PositiveFloat(30, doc="outer diameter")`

and then you pass them to the components, like so:
```
    def make_components(self):
        return {
            'outer_ring': _Ring(
                outer_radius=self.outer_diam / 2,
                inner_radius=(self.outer_diam / 2) - self.outer_width,
                width=self.width,
                ball_radius=(self.ball_diam / 2) + self.tolerance,
                rolling_radius=self.rolling_radius,
),
```
and then the subcomponents need to collect the variables again. Doesnt this feel like duplicated effort?
Wouldnt it be possible for a subcomponent to get the variables directly from the parent?

My suggestion is to use for example `super.outer_diam` in the subcomponent, is there something inherently bad with this approach?

fragmuffin, 2018-05-02:
@gntech 

> Doesnt this feel like duplicated effort?

It definitely does. sometimes every parameter is copied over without being modified... that gets tedious.

**Nested Dict Idea**
I've been considering creating a means to pass a nested dict of parameters to build an assembly.
The plan has some holes in it, but it's more a plan for serialising an assembly's nested parameters.

**Parent Parameters**

> My suggestion is to use for example super.outer_diam in the subcomponent

That's a good idea, but I'm reluctant to have it as a permanent fixture for the `Component` class. I'd rather encourage each `Component` to be independent of it's parent where possible.

However, I can introduce a parameter type that's a `Component` instance, called `ComponentRef`:

```python
import cadquery
from cqparts import Part, Assembly

class _Leg(Part):
    parent = ComponentRef(doc="table-top assembly")
    def make(self):
        return cadquery.Workplane('XY').box(
            self.parent.length / 10.,
            self.parent.width / 10,.
            self.parent.height - self.parent.thickness,
        )

class Table(Assembly):
    length = PositiveFloat(1000, doc="table length (mm)")
    width = PositiveFloat(500, doc="table width (mm)")
    height = PositiveFloat(800, doc="table's height (mm)")
    thickness = PositiveFloat(20, doc="tabletop thickness")
    def make_components(self):
        return {
            'front_left': _Leg(parent=self),
            # and so on
        }
    def make_constraints(self): 
        # blah blah my point's been made
```

gntech, 2018-05-04:
I think ComponentRef is a quite nice solution

fragmuffin, 2018-05-04:
Thank you for the discussion @gntech, it's exciting to see `cqparts` get more usable.

closed 2018-05-04 by fragmuffin

gntech, 2018-05-04:
cqparts is brilliant! When I get some time I will share the coffetable design file. I think it could serve as an advanced example for cadquery/cqparts usage.

fragmuffin, 2018-05-04:
@gntech yes please!
I'd love to see how more people design their models, it would greatly help evolve `cqparts`.
This addition of `ComponentRef` is a perfect example of that.

gntech, 2018-05-10:
@fragmuffin 
I have now uploaded the design files for the coffee table I mentioned to here
https://github.com/gnhdesign/ct1

I have also written a blog post about it on my website
https://www.gntech.se/en/projects/ct1/

fragmuffin, 2018-05-10:
@gntech that is so cool!