cqparts/cqparts #72: STEP Multipart Import with colours and names

fragmuffin, 2018-05-04:
Multi-part `STEP` import introduced in #71
However, names & colours are lost in the process.

**Target Outcome:**
When importing a step file, retain:

- [ ] object name
- [ ] object material properties (just colour?)

Rayfoundry, 2018-06-28:
This would be a great addition.

fragmuffin, 2018-06-29:
Agreed @Rayfoundry , I have grand plans for this.
I've noticed that most (if not all) `STEP` importers will pull in **all** geometry before you have a chance to see the hierarchy, or to choose exactly which sub-assembly you **actually** want to use.

When `cqparts` can do this, it will instantiate an import `Assembly` instance with a cursory scan of the file, and allow you to pick and choose what geometry you want to import.

I'm looking forward to implementing this one!

Rayfoundry, 2018-06-29:
@fragmuffin Great! cqparts with it's glTF support and planned transition to python-occ (cadquery) will be a really great and versatile tool. We've recently tested some STEP files with the current freecad backend, but it didn't manage to process more complex larger files. python-occ on the other side works flawless with large complex files. Is there any build we can check out that's running on the python-occ backend already?

fragmuffin, 2018-06-29:
`cqparts` is a wrapper around `cadquery`, and _before_ this issue is closed, `cqparts` doesn't offer much in addition to cadquery's STEP import.

@adam-urbanczyk has done excellent work in porting `cadquery` to pythonocc in their fork:
https://github.com/adam-urbanczyk/cadquery , instructions are there on how to get started.

However, `cadquery` doesn't support export to glTF... so if you're after a hedless import of STEP, and export to glTF, then you'll want to watch:
- [`cqparts` PythonOCC milestone](https://github.com/fragmuffin/cqparts/milestone/2), &/or
- #49 : Port glTF exporter to cadquery (will be limited to exports of single meshes)

I had a quick look at https://www.rayfoundry.de/ , what you're doing looks fascinating!
Are you considering using `cqparts` for your application? I'd like to know how I can help :+1:

jmwright, 2018-06-29:
@Rayfoundry 

> Is there any build we can check out that's running on the python-occ backend already?

There are instructions for setting up a CadQuery Anaconda environment that I think is Python-OCC only in the CQ-editor [installation instructions](https://github.com/CadQuery/CQ-editor#installation). I've been using that environment some, but I'm not sure exactly how @adam-urbanczyk built it. cqparts is not supported yet though. I think that will have to wait on #98 to be resolved.

adam-urbanczyk, 2018-07-03:
@Rayfoundry  @jmwright  this env is based on the cq-occ conda package:

https://anaconda.org/CadQuery/cadquery-occ

Simplest way for now is indeed to use the env spec from here: https://github.com/CadQuery/CQ-editor/blob/master/cqgui_env.yml

Or just run this (for pure cq-occ w/o GUI):
```
conda create -n test_cq -c pythonocc -c oce -c conda-forge -c dlr-sc pythonocc-core=0.18 pyparsing mock;
```
I think BTW that  @CapableRobot did some experiments with cq/cqparts based on pythonocc
https://github.com/CapableRobot/cadquery
https://github.com/CapableRobot/cqparts

fragmuffin, 2018-09-26:
As I write this, I'm waiting minutes for a 60MB `.step` to import to FreeCAD.
The import locks up the entire UI while running (I'm running an i7 with 8G RAM)
I think we can do better.

**Example use-case:**
import a large multi-part `.step` file as a `cqparts.Assembly`, then export it as `.gltf`.

```python
from cqparts import Assembly

my_thingy = Assembly.importer('step')('big_file.step')
my_thingy.exporter('gltf')('thingy.gltf')

# or in 1 line
Assembly \
    .importer('step')('big_file.step') \
    .exporter('gltf')('thingy.gltf')
```

## Simplistic Solution
1. Import everything in the `.step`, create the geometry, and load it into RAM.
1. Recursively iterate through each `cqparts.Component` and export it to `.gltf` buffers (and, of course, the `.gltf` index file itself)

## Smart Solution
1. Skim-read the `.step` and remember: 
    - data: "sections" of importable geometry (save as start/end `seek` points in the `.step` file)
    - metadata: names, hierarchies, material data, etc.
1. Recursively iterate through each `cqparts.Component` to create the `.gltf` index file (if buffers aren't embedded into the `.gltf` file)
1. Recursively iterate through each `cqparts.Component`, populate a thread stack where each process:
    - creates a stream object that can be:
        - saved to file, or
        - tunnelled through `http` (or whatever else 
    - stream is populated via a pipeline:
        - `.step` section import to geometry, which is passed as
        - geometry to generate `.gltf` buffer data.

**Benefits**
- Much faster
    - response-time to _begin_ exporting, or displaying on a web-app
    - overall
- Less RAM
- Error resistant : if there's an error in the `.step` file, that part geometry could be ignored, and an exception caught.
- Could be used to service parallel network requests (1 thread per request)

This should happen after porting to OCC

fragmuffin, 2018-09-26:
Hmm... there's one problem with that...
The `.gltf` buffer references have min/max ranges for each axis (x/y/z). So in order to get those ranges, the entire geometry would need to be known.
I haven't looked in detail at the `.step` format, perhaps we'll get lucky and those ranges will be metadata.

fragmuffin, 2019-01-02:
There may be another problem with the "smart solution"

> 1. Skim-read the .step and remember:
>    * data: "sections" of importable geometry (save as start/end seek points in the .step file)

The `.step` format is a sort of link-list of indexed vertices, edges, splines, faces (and so on).
Usually the indexed geometry for a single part is confined to a seekable range within the `.step` file, but this is a common byproduct of creating the file.
From what I understand, a `.step` file's content separated into lines, could (almost) be randomised and re-written, and still import the same shapes.

**Alternative that may still be possible**
A Skim-read could cache:
* an index list: `[(<step index>, <file seek>, {<referenced indexes>, ...}), ... ]`
* map parts to a dict: `{<part uid>: <step-index>, ...}`

Then to import a single part, the `.step` file could be fragmented; separated by tracing all referenced lines relating to that part, then "import" the fragmented `.step` file as a single `Part`.

**Tasks / Questions to answer**
* Feasibility study:
   * [ ] is this plausible? can the `.step` be fragmented like this?
   * [ ] different `.step` version support - this will require a partial "parsing" of the `.step` file, how susceptible to change is the `.step` format?, has it's indexing & core metadata structure always been the same?
* Performance test(s)
   * [ ] is this "smart" way going to be just as slow as a standard "import everything" approach?