cqparts/cqparts #74: Performance profiling

gntech, 2018-05-04:
I have made an cqparts assembly containing 7 parts. The models take 4-5 seconds to calculate and render on my fairly beefy machine 8-core machine. I think this is quite a lot taking in to consideration that the model is not super huge.

Is there some  way to get a log showing the calculation time for the different parts and build stages of cqparts to see where all this time is going. Maybe we can find segments of the build that can be optimized

jmwright, 2018-05-05:
@gntech In the latest commit to the FreeCAD module there is now an option in Settings.py called `report_execute_time` that will show you how long a script took to execute. It's false by default, so you'll need to set it to true to access the feature. cqparts is also integrated into the module now, and while you can use cqparts' `display()` function to show your parts, that won't give you the execution time. The following code shows an example from the readme with the necessary changes at the end to use CQGI.
```python
# create unit cube as cqparts.Part
import cadquery
import cqparts
from cqparts.params import PositiveFloat

class MyCube(cqparts.Part):
    size = PositiveFloat(1, doc="cube size")
    def make(self):
        return cadquery.Workplane('XY').box(self.size, self.size, self.size)

# create cube instance
cube = MyCube(size=10)

# display cube (optional)
# from cqparts.display import display
# display(cube)
show_object(cube.local_obj)
# If you want to display an assembly it looks like you'd want to call world_obj instead of local_obj
```
Hopefully that will help you do some performance profiling on your scripts.

fragmuffin, 2018-05-05:
Or, for a much more generic aproach:

```python
import cProfile
import pstats
from io import BytesIO
import logging

log = logging.getLogger(__name__)

def profile_this(func):
    def wrapper(*args, **kwargs):
        profiler = cProfile.Profile()

        # run func
        profiler.enable()
        func(*args, **kwargs)
        profiler.disable()

        # get statistics on function code
        stats_stream = BytesIO()
        prof_stats = pstats.Stats(profiler, stream=stats_stream).sort_stats('cumulative')
        prof_stats.print_stats()

        # Write stats to a logger
        log.warning(stats_stream.getvalue().encode())

    return wrapper

import cadquery
cadquery.freecad_impl.console_logging.enable()

import cqparts
from cqparts.params import *
from cqparts.display import display

class MyCube(cqparts.Part):
    size = PositiveFloat(1, doc="cube size")
    def make(self):
        return cadquery.Workplane('XY').box(self.size, self.size, self.size)

@profile_this
def test():
    cube = MyCube(size=10)
    cube.build()

test()
```

that should work for anything python

fragmuffin, 2018-05-11:
## Tasks

- add time deltas for:
    - [ ] `Part.make()`
    - [ ] `Assembly.make_components()`
    - [ ] `Assembly.make_constraints()`
    - [ ] `cqparts.display.display` (build in to new `DisplayEnvironment` class #77)
    - [ ] `Component.build()`
- exporter time deltas
    - [ ] tessellate geometry
    - [ ] build export format
- [ ] log time deltas at `DEBUG` level as they're recorded.

fragmuffin, 2018-05-14:
@jmwright yep, no problem, I was just going to store them as `_` prefixed variables.