cqparts/cqparts #75: Examples Not Finished

jmwright, 2018-05-04:
I started to try to use the fastener_easyinstall example for testing, but it has multiple syntax errors and isn't complete. I've fixed the syntax errors, but am not sure of the intent of the last section that isn't finished. Also, I'd like to include a cqparts example submenu in the FreeCAD module to give people a sample of what cqparts is about. However, the other two examples are just stubbed in. I see an issue about finishing one of the other examples, but nothing about the first one.

fragmuffin, 2018-05-05:
@jmwright 

> I started to try to use the fastener_easyinstall example for testing, but it has multiple syntax errors and isn't complete.

That's very interesting... I don't have a strict testing feedback loop to check for errors, but it's safe-guarded against syntax errors because almost everything documented is in a sphinx `doctest` block, and those tests pass.

All the code in the [easy install fastener](https://fragmuffin.github.io/cqparts/doc/cqparts_fasteners/easy-install/index.html) comes from the code that exports the _three.js_ rendered displays on that page... [`generate.py`](https://github.com/fragmuffin/cqparts/blob/master/docs-sphinx/_static/iframes/easy-install/generate.py)

I did find one bug, I've since moved the `Fasteners` class, just fixed in 68b2b5af75f598c6e137b77c78ba1b95963bc08a. However that's not a syntax error.
I also compared the published documentation to the _generate_ script, and they're identical.

> [I] am not sure of the intent of the last section that isn't finished

I'm very confused... the above linked tutorial is very much finished.
Could you send me a link of what you're using... is it the same as the link above?

> am not sure of the intent of the last section that isn't finished

Yes, I really should prioritise these issues: #57 #7
I'll try to work on the logo example today.

I'd also like to put some simpler examples.
Do you have any ideas of a simple example of an `Assembly` ?
Preferably something that only requires ~50 lines of code; something that newbies could see on one screen.

jmwright, 2018-05-05:
> I'm very confused... the above linked tutorial is very much finished.

I think we're talking about two different examples. I mean the one [here](https://github.com/fragmuffin/cqparts/blob/master/examples/fastener_easyinstall.py).

* Have a look at this line [here](https://github.com/fragmuffin/cqparts/blob/master/examples/fastener_easyinstall.py#L28). There's a TODO comment without a hash in front of it. That function also needs a pass in it if it's only going to contain a comment.
* [This line](https://github.com/fragmuffin/cqparts/blob/master/examples/fastener_easyinstall.py#L37) is just a comment with no pass in the function.
* [Here](https://github.com/fragmuffin/cqparts/blob/master/examples/fastener_easyinstall.py#L119) the `parts` and `start` parameters have already been defined with keyword args, and positional args come after them. There's also no closing parenthesis on the function call.

Now that the FreeCAD module embeds cqparts, you can just update the module (FreeCAD addons manager in version 0.17+), start a new script, and copy that example code in. When you hit F2 to execute it should complain about the errors I'm highlighting.

> Do you have any ideas of a simple example of an Assembly ?

I like the toy car example [here](https://fragmuffin.github.io/cqparts/doc/tutorials/assembly.html) for a simple one. If you want to make it slightly more complex, you could add bearings in the wheels or something like that. I think that a body, axles and wheels are good to show the basic concepts of creating an assembly in cqparts.

fragmuffin, 2018-05-05:
> I think we're talking about two different examples. I mean the one [here](https://github.com/fragmuffin/cqparts/blob/master/examples/fastener_easyinstall.py).

Oh!, I completely forgot about that whole folder!
yes, all of that needs to be amended urgently!

I think some of those scripts are still in the same form from when this whole project was in it's inception phase, so I didn't even know how it was all going to work!

> I like the toy car example [here](https://fragmuffin.github.io/cqparts/doc/tutorials/assembly.html) for a simple one

Yeah, I love that little car... when I made that I realised that `cqparts` was no longer just a concept, it was actually working. But that's just sentiment; I'm glad to hear that you like it as an example too.

I'll fix up that examples folder as soon as I can

# Task:

- [ ] fix or remove broken scripts in [`examples` folder](https://github.com/fragmuffin/cqparts/tree/master/examples)
- [ ] add [toy car example](https://fragmuffin.github.io/cqparts/doc/tutorials/assembly.html)
- [ ] add [biscuit example](https://fragmuffin.github.io/cqparts/doc/cqparts_fasteners/biscuit/index.html)
- [ ] add example files to the test-suite; regression testing

jmwright, 2018-05-05:
@fragmuffin As soon as you're happy with the examples I'll work on finding them a home in the CadQuery > Examples menu in the FC module (a.k.a. CQFM). I'm also about to do a major rework on the CQFM documentation, so I'll highlight cqparts support in there too.

fragmuffin, 2018-05-21:
**Task**
- [ ] close #93 when happy with these examples, and any from there are added.