cqparts/cqparts #80: Best practice for embedding gltf-viewer on website

gntech, 2018-05-08:
On your website you have several embedded gltf-viewers of your car assembly and parts for example.

This is a really neat feature and something I would want to have on my website as well to showcase my 3d-models and designs.

What do you think about documenting the steps to do this in the documentation. And on the same time look into what the best practice would be? Maybe there are more convinient ways than what you do right now?

I have been looking at clay-viewer which looks really nice and opens and views the gltf-models from cqparts just fine. I am not sure how to embed it yet though.

https://pissang.github.io/clay-viewer/editor/
https://github.com/pissang/clay-viewer

jmwright, 2018-05-08:
This might be a good conversion for @RustyVermeer to be involved in. He's done some similar work with web viewers,.

https://rustyvermeer.github.io/design/web_viewer/

gntech, 2018-05-09:
I did some experimentation last night with threejs and the GLTFLoader.
It is a very convenient solution and quite easy to get started with. I tested it with the Duck.gltf example. and followed these instructions.

https://threejs.org/docs/#manual/introduction/Creating-a-scene
https://threejs.org/docs/#api/lights/HemisphereLight
https://threejs.org/docs/#examples/loaders/GLTFLoader

However there seems to be some incompatibilties with the gltf-files generated by cqparts. I couldnt render them using threejs.

However, the files generated from cqparts opens in Clay-viewer without problems. 
https://pissang.github.io/clay-viewer/editor/

@fragmuffin do you know the reason why the gltf from cqparts doesnt render in threejs?

gntech, 2018-05-09:
Here is an example for how to load a gltf model using ClayGL

http://examples.claygl.xyz/examples/basicGLTFCameraAnimation.html

RustyVermeer, 2018-05-09:
@jmwright sorry for silence for some time... been busy.

@gntech in general, I think it's difficult to define 'best practices' for something like a viewer. This is in part due to the many options users have for file output types. That alone makes it somewhat difficult to plan for what people might have in mind for their own use cases.

As such, I think over time people conclude that something like WebGL is fairly accessible for people to 'roll their own' viewers. This is certainly what happened for me.

The link Jeremy provided to an example viewer of mine uses .json filetype and a Three.js scene I made myself (it's not necessarily the best way it could be made).

I have a slightly modified version of that web viewer which uses .gltf filetype
[https://github.com/RustyVermeer/desk/tree/master/web_view](https://github.com/RustyVermeer/desk/tree/master/web_view)

In order to get this to work, I DID have to modify the qparts code.

The culprit, if I recall correctly is in the materials.

You can test this yourself by opening a gltf that you want to load with Three.js and finding all of the materials nodes eg:

```  
"materials": [
    {
      "alphaCutoff": 1.0, 
      "alphaMode": "MASK", 
      "pbrMetallicRoughness": {
        "baseColorFactor": [
          0.7529, 
          0.7529, 
          0.7529, 
          0.8
        ], 
        "metallicFactor": 0.1, 
        "roughnessFactor": 0.7
      }
    }
  ], 
```

And change the alphaMode from MASK to BLEND. An example from my gltf file:

```
  "materials": [
    {
      "alphaCutoff": 1.0, 
      "alphaMode": "BLEND", 
      "pbrMetallicRoughness": {
        "baseColorFactor": [
          0.7529, 
          0.7529, 
          0.7529, 
          1.0
        ], 
        "metallicFactor": 0.1, 
        "roughnessFactor": 0.7
      }
    }
  ], 
```

The code change to cqparts, I believe was just one line:

[https://github.com/fragmuffin/cqparts/blob/ed71b559b4f0c1be04cecbcd81da7c27e53e4ebd/src/cqparts/display/material.py#L173](https://github.com/fragmuffin/cqparts/blob/ed71b559b4f0c1be04cecbcd81da7c27e53e4ebd/src/cqparts/display/material.py#L173)

@fragmuffin I changed this on a local copy of the cqparts library as a 'quick fix' for my own purposes. I do not know if such a change is appropriate, or if this should be handled in a different way?

gntech, 2018-05-09:
@RustyVermeer Thanks for tip about the material! When I opened this issue I felt like I was hitting my head against the wall trying to get the 3d-model to render using ThreeJS. Now it works flawlessly.

Regarding a fix for the alphamode issue. Perhaps "Blend"-alphaMode is just not supported by ThreeJS. As I mentioned, the model opens fine in Clay-viewer without changing the alphamode.

Maybe we could add an optional "alphaMode"-parameter to the GLTFExporter()-function that defaults to "MASK"? Thoughts?

RustyVermeer, 2018-05-10:
Hm.. that could work. I can foresee a future though where every user wants something just *slightly* different. Over time it might get cumbersome to maintain code to handle all kinds of parameters / cases.

Maybe it makes sense to have a sensible template with defaults that can be useful in the major WebGL setups (Babylon.js, Three.js, cesium.js, maaaybe one more?). Then, if someone wants anything more particular, they provide a flag that points the gltf converter to a custom template.

That custom template would be assumed correct, from cqparts perspective, and it would just dump the appropriate geometry into the correct spots in the template.

Just a rough idea. Cqparts is essentially all the hard work of @fragmuffin , who would probably have a clearer picture of how to appropriately handle something like this.

fragmuffin, 2018-05-10:
Woah, I've missed a lot in 3 days!

@gntech 

> do you know any reason for why the gltf from cqparts doesnt render in threejs?

Yes, as @RustyVermeer mentioned, the `ALPHA` is an issue, but I didn't know `BLEND` would work.
@donmccurdy found an alternative solution [here](https://github.com/mrdoob/three.js/issues/7125#issuecomment-386035472).

The real problem is that the rendering properties were "designed" by trial and error :roll_eyes:.

My plan has been to improve this in issue #36

> Maybe we could add an optional "alphaMode"-parameter to the GLTFExporter()-function that defaults to "MASK"? Thoughts?

- [ ] I can do that.

@RustyVermeer, you're not wrong about these parameters being difficult to maintain, but in this case I can add a `**kwargs` parameter to the exporter, and log a warning for any unused parameters.

@gntech : to your original question

> What do you think about documenting the steps to do this in the documentation. And on the same time look into what the best practice would be? Maybe there are more convinient ways than what you do right now?

There are definitely better ways to do it, at present each render is done in its own frame. This doesn't seem too bad, until you do a full refresh and watch the aframe framework get downloaded 10 times for a single page.

I was thinking of eventually creating a gltf based viewer that also displayed metadata from cqparts (like a hierarchy of parameters and their values), and had some basic options for rendering, like hide/show particular parts, or wire-frame, semi-transparency, animations (like a turntable, also animations for more complex constraints when they're introduced)

But also, I like the idea of a guide for display in a range of frameworks... I can aim for:

- [ ] Three.js
- [ ] A-Frame
- [ ] Babylon.js
- [ ] cesium.js

Am I missing any?

donmccurdy, 2018-05-11:
For what it's worth you can embed my viewer (https://gltf-viewer.donmccurdy.com/) if you like. It uses threejs, but the issue with threejs will be fixed soon. Embed it like this:

```html
<iframe src="https://gltf-viewer.donmccurdy.com#kiosk=true&model=<MODEL-URL>" frameborder="0" width="600px" height="400px"></iframe>
```

But there are much more efficient ways to render than iframes, if you have many parts on the same page. :)

mbb670, 2020-01-22:
@donmccurdy 

I’m attempting to use your iframe but having a bit of trouble. the viewer shows up fine but the file will not show up. I keep getting an error that says “An embedded page at gltf-viewer.donmccurdy.com says:
Unable to retrieve this file. Check JS console and browser network tab”

I have the file linked in from my own server and have tried several other ways. 
Any thoughts?
Thanks!

donmccurdy, 2020-01-24:
What do you see on the JS Console? The most likely cause is a CORS issue, which requires that you configure your server to allow my domain (or any domain, if you want) to access the file. See: https://enable-cors.org/server.html

louisryko, 2020-12-08:
Hi Dom,

I've experiencing the same issue: “An embedded page at gltf-viewer.donmccurdy.com says:
Unable to retrieve this file. Check JS console and browser network tab”

I've made a simple HTML page with your iframe in it. I'm using WordPress, added code in PHP to allow CORS (following instructions from 2 separate blogs), but still, nothing... any tips?

donmccurdy, 2020-12-08:
Usually this error means that your .glb/.gltf file is hosted in a way that prevents my site from accessing it. I'm afraid I don't know much about WordPress configuration, but feel free to open an issue on https://github.com/donmccurdy/three-gltf-viewer and I'll try to help there. 

More recently, the https://modelviewer.dev/ project provides a really nice model embedding solution. It's faster and more customizable than embedding my viewer, and can avoid CORS complexity, so I'd recommend giving it a look. 🙂

louisryko, 2020-12-09:
Very helpful. Thank you Dom! 👍

BapsDegryse, 2020-12-16:
Hello Don,

I have tried to use the iframe with a glb in my firebase storage, but it looks like you discard the url arguments after the "&".

Since the storage url uses a token 
for example the <MODEL-URL>: "https://firebasestorage.googleapis.com/v0/b/myproject.appspot.com/o/Test%2Ftest-web.glb?alt=media&token=my-secret-token" will discard "&token=my-secret-token" and use it as an argument for your site. This means that firebase storage will refuse my read since i'm not identified anymore...

I don't have any idea to make it work, do you have any tips?

And great job on this viewer by the way, it's awesome :D

BapsDegryse, 2020-12-16:
All good, I just had to use encodeURIComponent(myUrl).
Sorry for the useless question :)