cqparts/cqparts #81: Stepper Motor

zignig, 2018-05-09:
Made a stepper motor model. Need some work.

Doesn't do cutouts and the axle alignment is a bit broken.

Nice work with cqparts the abstraction is just right and the tutorials made it pretty easy start.

fragmuffin, 2018-05-10:
@zignig very nice!

[image]

> Nice work with cqparts the abstraction is just right and the tutorials made it pretty easy start.

I'm glad you like it!

This would be a great start to the `cqparts_motors` library, would you consider making a pull request?

zignig, 2018-05-10:
Pull request requested. ;)

Now onto servos and DC motors...

jmwright, 2018-05-10:
@zignig Awesome!

fragmuffin, 2018-05-11:
@zignig awesome!, you might beat me to #7 !

jmwright, 2018-05-17:
Thanks @zignig !