cqparts/cqparts #85: Question: Howto get stdout output from cqparts/cadquery scripts

gntech, 2018-05-11:
I like to have print() statements or log.info() statments in my code that shows information about script progress, calculated design values, and so on. Is it possible somehow? Right now it looks like it is swallowed by FreeCAD somehow. Sometimes I get the output from the script after the entire script is finished. I dont get error or tracebacks from the script which is annoying as well.

``` python
import os
from ct1 import CoffeTable

table = CoffeTable()

working_folder = os.path.dirname(os.path.realpath(__file__))
dir2d = os.path.join(working_folder, "build/2d")
dir3d = os.path.join(working_folder, "build/3d")
dirgltf = os.path.join(working_folder, "build/gltf")

print(table.tree_str(name="ct1"))

print("Exporting part SVG 2D-views")
if not os.path.isdir(dir2d):
    os.makedirs(dir2d)
table.find('leg_0').local_obj.exportSvg(os.path.join(dir2d, "ct1_leg.svg"), view_vector=(0,1,0))
table.find('shelf_0').local_obj.exportSvg(os.path.join(dir2d, "ct1_shelf.svg"), view_vector=(0,0,-1))
table.find('glass_top').local_obj.exportSvg(os.path.join(dir2d, "ct1_glass_top.svg"), view_vector=(0,0,1))

print("Exporting part STEP 3D-models")
if not os.path.isdir(dir3d):
    os.makedirs(dir3d)
table.find('leg_0').exporter("step")(os.path.join(dir3d, "ct1_leg.step"))
table.find('shelf_0').exporter("step")(os.path.join(dir3d, "ct1_shelf.step"))
table.find('glass_top').exporter("step")(os.path.join(dir3d, "ct1_glass_top.step"))

print("Exporting assembly GLTF-model")
if not os.path.isdir(dirgltf):
    os.makedirs(dirgltf)
table.exporter('gltf')(os.path.join(dir3d, "ct1.gltf"))

```

jmwright, 2018-05-11:
@gntech From within FreeCAD, you can use `FreeCAD.Console`.
```python
import FreeCAD

FreeCAD.Console.PrintMessage("Debug\r\n")
FreeCAD.Console.PrintError("Error!\r\n")
```
At one time I had a `print` function defined within the CadQuery module that abstracted the FreeCAD boiler plate code out, but it caused problems so I removed it. I probably just didn't implement it correctly and so it caused conflicts with the stock print function.

jmwright, 2018-05-11:
@gntech Looking at the code, it looks like I implemented a print hook in a different way later on. I had forgotten about that. The following prints correctly to the "Report view" for me.
```python
print("HERE")
```

gntech, 2018-05-11:
@jmwright I run the script above from command line like so: `python build_ct1.py`  and I get this output to the terminal.

``` bash
$ python build_ct1.py
No handlers could be found for logger "cqparts.assembly"
Exporting assembly GLTF-model		
```

It is very strange, only the last print() statement gets printed

gntech, 2018-05-12:
When I run the script from freecad all print statements are shown. However, it seems they are put on hold so everything prints after the script is done.

fragmuffin, 2018-05-12:
This was also addressed in dcowden/cadquery#200

```python
import cadquery
cadquery.freecad_impl.console_logging.enable()
import logging
log = logging.getLogger(__name__)

log.debug("not shown")
log.info("displayed as normal")
log.warning("shown in amber")
log.warning("shown in red")
```

also, `cqparts` uses `logging` to print debug information.

fragmuffin, 2018-05-13:
@gntech has this answered your question?
also, what worked for you?... I'd like to put this into [the documentation](https://fragmuffin.github.io/cqparts/doc/) somewhere, perhaps as the first tutorial.

gntech, 2018-05-13:
Tanks for your suggestion. However, It will be some time before I will be able to test it. I will answer here then

fragmuffin, 2018-05-15:
@gntech no problem. Take your time.

gntech, 2018-05-16:
Okay, I have done some troubleshooting and I think it comes down to the suppress_stdout_stderr()-function in cadqeury. I have opened a new issue there, see https://github.com/dcowden/cadquery/issues/267

closed 2018-05-16 by gntech