cqparts/cqparts #86: Export SVG and STEP from assemblies

gntech, 2018-05-11:
Right now it is not possible to export assemblies to SVG or STEP files. Is there some technical difficutlties with this that needs to be solved or is it just a matter of implementing the function that hooks in to the freecad backend? Similar to how a Part is exported to STEP.

fragmuffin, 2018-05-14:
@gntech it's because assemblies are multipart, and the SVG and STEP export functionality is passed straight to cadquery

They could both be achieved now by fusing everything first, which would probably be ok for SVGs.

But the STEP export should be a proper multipart STEP with metadata to match (eg: colour).
I'm not familiar with the STEP format (yet)

tolot27, 2019-01-01:
How to "fuse everything first"? In https://github.com/easyw/kicad-3d-models-in-freecad/blob/f2fe1731ecc88cb44da38f14bf6a1172e1a3d959/cadquery/FCAD_script_generator/_tools/cq_cad_tools.py#L171 two parts are fused and their colors are retained. I'd love to see this for assemblies, too.

jmwright, 2019-01-02:
Tagging @easyw . Please see the link and post directly above.

fragmuffin, 2019-01-02:
@tolot27 

> How to "fuse everything first"? [...] two parts are fused and their colors are retained

That's really a `cadquery` thing, and I think `cadquery` worplanes didn't have any rendering properties, just geometry.
I mean, I agree that `cqparts` could offer a way to cast an `Assembly` into a `Part`, I can see that would be useful.
But, at this time, I don't know how to mix rendering properties within a single `cadquery.Workplane`.

**SVG**
For SVGs, whether the part is merged or not doesn't really matter.

Hey @zignig , you've done some amazing work automating "screenshots" of assemblies as a sort of pre-render for the `cqparts` webpage.
I haven't looked at how that works yet, but do you think that method could be used to create an SVG?

I've also been playing around a bit with [anime.js](http://animejs.com/) and [vivus.js](https://maxwellito.github.io/vivus/), I'd love it if paths could be id'd for each part & assembly.

zignig, 2019-01-02:
> Hey @zignig , you've done some amazing work automating "screenshots" of assemblies as a sort of pre-render for the `cqparts` webpage.
> I haven't looked at how that works yet, but do you think that method could be used to create an SVG?

As it happens , I have already written something like this 

https://github.com/zignig/cqparts_bucket/blob/master/svg_test.py

It takes all the items gets bounding boxes, packs using rectpack and then exports it to an SVG using a hacked version of the cadquery svg export.

There is also a start of a better version that extracts all the -Z faces , scans through the wires. It does not export an svg or do packing, I was just fiddling with getting line segments out of objects 

https://github.com/zignig/cqparts_bucket/blob/master/get_faces.py

This work was heading towards generating laser cutting plans rather than routing. The first one depends on FreeCad and will need to be rewitten for the occ upgrade.

The second needs a lot more work but it's am interesting start.

zignig, 2019-01-02:
What has been rumbling around is my brain is pipeline that takes an assembly and exports it in the correct forms. Laser cut to avg,  printables to st, etc. I have been working on the  webapi to this end over the last few days.