cqparts/cqparts #87: Feature: Predefined 2d-export views for component

gntech, 2018-05-11:
When creating a design that is meant to be manufactured you most of the times need some way to get 2d-drawings for the different components of the assembly or the part.

Depending on the design of the component a different number of views are needed to fully define the part.

I think it would be nice to add a function to the component that lists which views are needed to define the part.

Then you could just call something similar to component.export_drawings() and you would get the number of dxf- or svg files you need to be able to manufacture.

To improve further the best would be if the views are placed on a single dxf with correct view placement relations between them. 

https://en.wikipedia.org/wiki/Engineering_drawing#Multiple_views_and_projections

fragmuffin, 2018-05-14:
@gntech yes! This is absolutely the intent I had for `cqparts`.
I'd also like to see gcode and slicer integration to directly produce prototypes.

That's an excellent example for a start point; a multi view DXF export.

I've been considering defining various decorators to set particular component features. Such as `@mate` and `@view`:

```python
import cadquery
import cqparts
from cqparts.component import mate, view
class Thing(cqparts.Part):
    def make(self):
        return cadquery.Workplane('XY').box(1, 1, 1)

    @mate('top')
    def mate_top(self):
        return Mate(self, CoordSystem((0, 0, 0.5)))

    @view('left')
    def view_left(self):
        return View(self, CoordSystem((0, 0.5, 0), normal=(0, 1, 0)))
```

The decorators can apply the `@property` behaviour (by default), and be listed by the class.
Then we can get all mates, or all views to a component.

What do you think should be part of a `View`?

- camera location
- bounding box (default taken from component)
- view type (orthogonal / perspective)
- render type (wireframe ... full render)
- included components (for assemblies)
- (probably more)

gntech, 2018-05-16:
My thoughts are:

## Standard views
https://en.wikipedia.org/wiki/Multiview_projection#First-angle_projection
A set of the six standard views is defined only by the NORMAL and ROTATION of the FRONT view. The rest of the six standard views (top, bottom, back, left, right) are derived from the front view.

After the front view is defined, the following 2d projection should be possible to be created automatically, see image link below.
https://upload.wikimedia.org/wikipedia/commons/b/b4/First_angle_unfolded.svg

But most of the times you only need to show perhaps 2, 3 or 4 of those six standard views to define the shape. My thought was that there should be a list where you set which views out of the six are needed
`chosen_views = [front, top, left]`

Note however that a certain combinations of views is not possible. You cant have front and back without the right or left view.

## Section views
Quite often the six standard views are not enough to define a design. Say you have a counterbore hole in a plate. You cant see how deep the counter bore is without a section view.

A section view should be defined only by a PARENT_VIEW and a section line relative to the view or a section plane from the model. If a plane is used a sanity check is that the plane should be perpendicular to the parent view.

## Other thoughts
The default should be first-angle projection and optionally we should be able to use third-angle projection for the US people. ;)

An isometric view is also nice to include in a corner of the drawing since it can serve as a good sanity check to see if you have interpreted the views correctly.

Each view could have the view name printed below "FRONT", "BACK", etc.

Boundary box dimension or max overall horisontal and vertical dimensions for each view is really nice to have. It gives a sanity check for that the scale of the DXF is correct when CNC-manufacturing and also gives a nice hint for how large piece of stock material is needed.

Render type is good to be able to choose as you suggest.

## Possible code example
``` python
import cadquery
import cqparts
from cqparts.component import Mate, View, SectionView

class Thing(cqparts.Part):
    def make(self):
        return cadquery.Workplane('XY').box(2, 2, 2)

    @mate('top')
    def mate_top(self):
        return Mate(self, CoordSystem((0, 0, 0.5)))

    @frontview()
    def front_view(self):
        return View(self, CoordSystem((0, 0.5, 0), normal=(0, 1, 0)))

    @sectionview('counterbore1')
    def section_1(self):
        return SectionView(self, Parent=derived_left_view, sectionline=((0, 1), (10, 1)))
```

@RustyVermeer From what I have seen from your github projects I think this discussion can be of interest to you. Feel free to join :)