cqparts/cqparts #91: Catalogue single entry point

fragmuffin, 2018-05-18:
**PLACEHOLDER**
TODO: fill this in after work.

zignig, 2018-05-18:
Possible Considerations

- namespace: a string that separates the catalogue totality into slices. ( default to "::cqparts:" for us )
- the python __class_name___ as id
- a shiny new UUID type 4 for each item 
- a SHA512 of the origin code and the parameters 
- a signature of some kind (RSA,GPG,FNORD,etc...)

and then JSON of the parameters including Xrefs to other catalogue via class name and/or uuid

The presentation of this data should be in a form that can be handed to a "find best quickly" or a hard SAT solver that can try __all__ the combinations and give you an actual valid solution.

fragmuffin, 2018-05-20:
@zignig 
> - namespace: a string that separates the catalogue totality into slices. ( default to "::cqparts:" for us )
> - the python class_name_ as id
> - a shiny new UUID type 4 for each item

I think these can be combined, they're sort of already taken care of in the current implementation.

```python
from cqparts_misc.basic.primatives import Cube
from cqparts.catalogue import JSONCatalogue
catalogue = JSONCatalogue('test.json', clean=True)
catalogue.add(obj=Cube(), id='unit_cube', criteria={'a': 1})
catalogue.close()
```

Now the contents of `test.json` is below.
- _namespace_ : is the python module
- _class name_ : is there
- _UUID_ : is conceptually the catalogue's filename + component's `id` (in this case: `unit_cube`)

```json
{
  "items": {
    "1": {
      "obj": {
        "params": {
          "_render": {
            "color": [
              192,
              192,
              192
            ],
            "alpha": 1
          },
          "_simple": false,
          "size": 1
        },
        "class": {
          "name": "Cube",
          "module": "cqparts_misc.basic.primatives"
        },
        "lib": {
          "version": "0.2.1",
          "name": "cqparts"
        }
      },
      "id": "unit_cube",
      "criteria": {
        "a": 1
      }
    }
  },
  "_dbinfo": {
    "1": {
      "lib": "cqparts",
      "ver": "0.1",
      "name": "JSONCatalogue",
      "module": "cqparts.catalogue.json",
      "lib_version": "0.2.1"
    }
  }
}
```

> - a SHA512 of the origin code and the parameters

That may be difficult given the nested nature of components; they contain instances of many classes, each of which may call functions form anywhere.
A code CRC would imply consistency of the model's creation, but I think this is borderline impossible to achieve.
I think the `cqparts_motors` library version number would be enough.

- [ ] _task_: add module version of root module for the added class (ie: for `Stepper`, would be `cqparts_motors`)

> - a signature of some kind (RSA,GPG,FNORD,etc...)

This could optionally be added to the `criteria` for now.
But I think this would be a great addition to #58 (whatever catalogues end up being)

> and then JSON of the parameters including Xrefs to other catalogue via class name and/or uuid

I could add an optional `meta` dictionary to contain things like this and the signature.
Or perhaps `criteria` should just be changed to `meta`.

- [ ] _task_: add optional `meta` dict alongside `criteria`.
- [ ] _question_ : rename `criteria` to `meta`?

zignig, 2018-07-01:
From building stuff in cqparts-bucket , I feel that the catalogue should return a class with the parameters set rather than an instance. I have been passing objects down and instantiating inside the Assembly.

When I try to mount each of the Steppers in the catalogue (https://github.com/zignig/cqparts-bucket/blob/master/stepper_cat.py) if fails because the stepper mount is expecting a stepper to call().

zignig, 2018-08-29:
Found a python sub plumbing solution to this 

for i in steppers:
    print i
    s = catalogue.deserialize_item(i)
    cl = type(str(i['id']),(Stepper,),{})
    p = cl.class_params(hidden=True)
    for j,k in p.items():
        print j,type(k)
        pr = type(k)
        setattr(cl,j,pr(i['obj']['params'][j]))
    stepper_list.append(cl)

takes a serialised instance and turns it into a class with the new parameters set.
mmmm crunchy.

Will make a generic version of this for inclusion in ParametricObject.