cqparts/cqparts #93: Parts Bucket

zignig, 2018-05-21:
I have started a repository for my cqparts experiments.

https://github.com/zignig/cqparts-bucket

@fragmuffin feel free use as examples.

fragmuffin, 2018-05-21:
Thanks @zignig , I'll have a closer look at these when I work on #75

zignig, 2018-05-29:
Check this out  :)

https://github.com/zignig/cqparts-bucket/blob/master/train.py

cqparts makes building things so nice.

fragmuffin, 2018-05-29:
> cqparts makes building things so easy.

yay!, I'm so glad you think so.

You may also be interested in this: [`cqparts_toys.train.track`](https://fragmuffin.github.io/cqparts/doc/api/cqparts_toys.train.html#module-cqparts_toys.train.track)
I haven't released the `cqparts_toys` lib yet, but that's the sort of thing I had planned for it.

My dad wants to make a coffee table with wooden tracks for the grand-kids, and I thought making something super custom would be awesome!

zignig, 2018-05-29:
The toys lib with track is why i wrote the train in the first place. :)  the current train size is wrong, but hey it's parametric.

I will keep making, a solid examples folder can compilment your fastidious code and documentarion.

zignig, 2018-06-17:
I think that cqparts is cool.

https://zignig.github.io/

it  takes a while to load 

https://github.com/zignig/cqparts-bucket/blob/master/robot_base.py

jmwright, 2018-06-19:
Very cool @zignig !

zignig, 2018-06-26:
whee!

https://zignig.github.io/rover.png

justbuchanan, 2018-06-26:
Wow, that looks great!

Btw, I was inspired by your `zignig.github.io` site and started working on a variant that's auto-updated through a CI build. The idea would be to have my collection of models (a significantly more sparse version of your cqparts-bucket) automatically build and deploy a gallery every time I push to the repository.

So far it's a little hacky and I just have it rendering the car example from cqparts, but eventually I'm planning to add more models and have the main page list them and allow for viewing each one.

I think it would be neat to eventually have a general purpose tool / configs that you can add to any "bucket" of parts github repo and automatically get an up-to-date github pages gallery of them.

My test repo for this is at https://github.com/justbuchanan/cadquery-models. Let me know if you have any pointers or if this sounds like something people would be interested in collaborating on.

zignig, 2018-06-26:
@justbuchanan , I am working on a CI builder just as you describe. We should collaborate. 

Presently I am using my cqparts-server (https://github.com/zignig/cqparts-server) to visualize my development. I am looking to put the CI into it in the longer term. 

As for the image it is rendered in Blender (https://blender.org/) and I'm using a gltf importer I found for blender (https://github.com/ksons/gltf-blender-importer)

cq-parts has got me back into programming for myself, instead of just work.

fragmuffin, 2018-06-26:
@zignig @justbuchanan 
The CI path is an excellent use for a headless CAD system!
I also want to automate exporting & displaying `gltf` models for `sphinx`.

I've put a few sphinx-specific tools into [`utils.sphinx`](https://github.com/fragmuffin/cqparts/blob/master/src/cqparts/utils/sphinx.py) ([docs](https://fragmuffin.github.io/cqparts/doc/api/cqparts.utils.html#module-cqparts.utils.sphinx)).
The idea would be to automate `gltf` exports for pages such as [Make your own `Part`](https://fragmuffin.github.io/cqparts/doc/tutorials/part.html), then to automate a 3d display of a sample of each documented `Component` (such as [`MaleFastenerPart`](https://fragmuffin.github.io/cqparts/doc/api/cqparts_fasteners.html#cqparts_fasteners.male.MaleFastenerPart))

When I get sick of doing the process manually, I'll figure out the enigma that is sphinx :wink:

zignig, 2018-06-26:
You're right sphinx doc is complicated. ;)

I would like to get to the point where I write a design and the 'compiler' will check that the design is correct and return warnings and errors. Then it can generate all the manufacturing files and complete documentation including assembly instructions with a single command.

jmwright, 2018-06-27:
@justbuchanan @zignig +1 on the idea of using CI

fragmuffin, 2018-07-18:
Wow @zignig that is so cool!
I feel bad about taking so long with #110, but that's an excellent use-case to make sure it works ;)

zignig, 2018-07-19:
I like it ;) , with your permission I would like to peel the rover into it's own repository and build _the_ showcase cqparts assembly. 

Have it over documented and script the complete build sequence from base render to build instructions.

fragmuffin, 2018-07-19:
@zignig yeah!
You don't need my permission, but if you did, you'd have it 👍

closed 2018-12-19 by fragmuffin