cqparts/cqparts #95: CadQuery Varibles Editor doesn't set variables

robodan, 2018-05-24:
In b2, a simple cadquery script can set the global variables through the CadQuery Variable Editor.  This works fine for me in FreeCad.
[b2.txt](https://github.com/fragmuffin/cqparts/files/2036598/b2.txt)

In bm2, I explect the same variable setting in the cqparts example.  However, the variables never change.  The Variable Editor sees them, and lets me edit them.  But the logging shows the original values, not the changed ones.
[bm2.txt](https://github.com/fragmuffin/cqparts/files/2036600/bm2.txt)

Ubuntu Linux 18.04 amd64
FreeCad 0.17
Cadquery 1.2.0  pip install
cqparts 0.2.0  pip install

robodan, 2018-05-24:
Oops, I included the wrong file.  It should be this (which I can't seem to upload as a file).
I only expect config and scale to be configurable.  They are simple variables (same as b2.txt).

from __future__ import print_function
from __future__ import division
from __future__ import absolute_import

import cadquery
import cqparts
from cqparts.params import *

from math import *

import logging

# Setting log level to DEBUG here also shows more info from cadquery
logging.basicConfig(level=logging.DEBUG)

# Global parameters can be configured by the CadQuery Variables Editor
config = "FFF"                            # 3D printing
scale = 1.0                               # another config test

class MyPart(cqparts.Part):
    """FIXME:  sample part docstring."""
    width = PositiveFloat(5, doc="just that big")
    length = PositiveFloat(5, doc="how far we'll go")

    def make(self):
        # FIXME:  replace all this with you part
        logging.info("config: %r scale: %f" % (config, scale))
        obj = cadquery.Workplane('XY').circle(self.width / 2)
        obj = obj.extrude(self.length)
        return obj
        

if __name__ == '__main__':
    mypart = MyPart()
    mypart.exporter("stl")("example_part.stl")
else:
    from cqparts.display import display
    mypart = MyPart()
    display(mypart)

robodan, 2018-05-24:
> @robodan <https://github.com/robodan> Can you wrap that in triple back
> ticks so that I can copy it? GitHub is trying to format it like markdown.

Should be fixed now.

jmwright, 2018-05-24:
Your first script uses the `show_object` function which taps into CQGI, and your second script uses cqpart's `display` function which doesn't hook into CQGI. That's why the second script's variables aren't dynamically editable. There have been recent discussions on portability issues like this. I think that everyone would like things to work the same whether you're using stock CadQuery or cqparts. For now, I was able to make your second script work as expected by commenting out the `display` line and putting a `show_object` line right below it.

robodan, 2018-05-24:
show_object of what?
This errors for me:  show_object(mypart)

jmwright, 2018-05-24:
Sorry, you have to pull `local_obj` out of mypart to show it through `show_object`. If it's an assembly I think you'd have to pull out `global_obj`, but @fragmuffin would know better than I on that.
```
show_object(mypart.local_obj)
```

fragmuffin, 2018-05-25:
@robodan yep, what @jmwright should work.

I think to leverage the current [`jmwright/cadquery-freecad-module`](https://github.com/jmwright/cadquery-freecad-module) with `cqparts`, you'd need to pass globally defined variables to the part's constructor.

```python
g_width = 5
g_length = 5

mypart = MyPart(width=g_width, length=g_length)
show_object(mypart.local_obj)

from cqparts.utils import CoordSystem
mypart.world_coords = CoordSystem(origin=(0,20,0))  # (see note below)
show_object(mypart.world_obj)
```

@jmwright is that ~~wright~~ correct?

note: requesting `world_obj` before setting `world_coords` will raise an exception.
[read this](https://fragmuffin.github.io/cqparts/doc/cqparts/part-build-cycle.html#world-coords-and-world-obj) for details.

jmwright, 2018-05-25:
@fragmuffin Correct. The global variables will be accessible to CQGI, and passing them into MyPart serves the original intent.

fragmuffin, 2018-05-26:
@robodan I'll close this issue with the above workaround.
This is also not strictly a problem with `cqparts`, but the `cadquery.cqgi`.

I've raised dcowden/cadquery#273 because this is not fixed, but the discussion needs to move to the right place.

Thank you for the feedback @robodan !
`cqparts` is moving fast, and I'm so glad people like yourself are critiquing it

closed 2018-05-26 by fragmuffin