cqparts/cqparts #96: Add Component copy behaviour

fragmuffin, 2018-05-24:
Implement explicit copy behaviour

```python
>>> c1 = Cube(size=10)
>>> c2 = c1.copy()
>>> from copy import copy
>>> c3 = copy(c1)
```

Both methods create objects independent of the original; changes to `c1` do not effect its copies.

`Assembly` component copies would be nested.

**Question**
Is there a way to intelligently copy a `Component` before its been built, so that the original, and the copy are only built once...
Initial implementation, probably avoid the tricks; keep it simple stupid (KISS principal)

zwn, 2018-05-25:
The `from copy import copy` would create a shallow copy of the original (according to https://docs.python.org/3.6/library/copy.html it would call a `__copy__` method on the instance, not `__deepcopy__`) but you seem to be describing a deep copy operation.

fragmuffin, 2018-05-25:
@zwn good point, I was thinking about that today.
How do you think a `copy` vs `deepcopy` would behave?
With respect to `Part` and `Assembly` instances?

The only difference I can think of is the `world_coords` and `world_obj`.
* `copy`: parameters, and `local_obj`
* `deepcopy`: parameters, `local_obj`, `world_coords`, and `world_obj`

I don't think a copy of an `Assembly` that references the original _assembly's_ components would be useful.

gntech, 2018-05-25:
> but you seem to be describing a deep copy operation.

I agree, the copy() function performs what is called a `shallow copy` and I think it corresponds very well to what we called `explicit cqparts.Part instance` creation in the other thread.

Deepcopy corresponds very well when we want to make a totally independent instance of a part.

Below is an excerpt from the Python documentation about copy vs deepcopy.

The difference between shallow and deep copying is only relevant for compound objects (objects that contain other objects, like lists or class instances):

* A shallow copy constructs a new compound object and then (to the extent possible) inserts references into it to the objects found in the original.
* A deep copy constructs a new compound object and then, recursively, inserts copies into it of the objects found in the original.

# That means for us
## copy() == **explicit cqparts.Part instance**
A (shallow) copy should keep references to the local_obj of the original but create a new world_coords  and world_obj. 

## deepcopy() == **explicit cqparts.Part independent copy**
A deepcopy should copy everything without references to the original. Thus creating a totally independent copy from the original.

This will be very Pythonic if we implement it this way.

gntech, 2018-05-25:
And my opinion is that there should be no different behaviour for Parts vs Assembly.

* A shallow copy for an assembly will recursively perform shallow copy on the included parts.
* A deepcopy of an assembly will recursively perform deepcopy of the included parts.

fragmuffin, 2018-05-25:
@gntech yep!, that's a good argument!
I love it ;)

fragmuffin, 2018-05-27:
closed as duplicate of #97

closed 2018-05-27 by fragmuffin