cqparts/cqparts #97: CAD Component Instances

fragmuffin, 2018-05-26:
Change directly flows on from #90

The target is for the following code-fragments to evaluate as they're shown

## Implicit `Part` behaviour

```python
>>> c1 = Cube(size=10)
>>> c2 = Cube(size=10)
>>> c1.local_obj is c2.local_obj
False
>>> c1 == c2
True
>>> hash(c1) == hash(c2)
True
>>> c1.local_obj = c1.local_obj.faces('>Z').hole(1)
>>> c1 == c2
False
```

## Explicit `cqpart.Part` `instance` modifier

```python
>>> c1 = Cube(size=10)
>>> c2 = c1.instance()
>>> c1.local_obj is c2.local_obj
True
>>> c1.local_obj = c1.local_obj.faces('>Z').hole(1)
>>> # OR
>>> #c2.local_obj = c2.local_obj.faces('>Z').hole(1)
>>> c1.local_obj is c2.local_obj
True
```

```python
>>> c1 = Cube(size=10)
>>> c2 = c1.instance()
>>> c1.world_coords = CoordSystem(origin=(10,0,0))
>>> c2.world_coords = CoordSystem(origin=(-10,0,0))
>>> c1.world_obj = c1.world_obj.faces('>X').hole(1)
>>> c2.world_obj = c2.world_obj.faces('>Y').hole(1)
>>> c1.local_obj is c2.local_obj
True
>>> c1.world_obj is c2.world_obj  # would never be True
False
>>> display(c1.local_obj)  # cube
>>> display(c2.local_obj)  # cube
>>> display(c1.world_obj)  # cube with hole along X axis
>>> display(c2.world_obj)  # cube with hole along Y axis
```

zwn, 2018-05-26:
```python
>>> c1 = Cube(size=10)
>>> c2 = c1.instance()
```
Isn't this (shallow) `copy()` from #96?

gntech, 2018-05-26:
@zwn yes I agree
This is strictly not necessary but couldn't `.instance()` be a nice `alias` or `convenience` function for the explicit shallow copy, eg:

``` python
>>> from copy import copy
>>> c1 = Cube(size=10)
>>> c2 = copy(c1)
```

The corresponding alias for deepcopy could then be `.instance(breaklinks=True)`

zwn, 2018-05-26:
Probably as I don't have any CAD background, I find a method on instance of an object called `instance` confusing - I wouldn't be able to guess what such a method does. However with shallow/deep copy I at least have a direction where to look. It is confusing for me mixing CAD with OOP. If anything, there is also a precedence in [dict.copy()](https://docs.python.org/3.6/library/stdtypes.html#dict.copy). Maybe calling it `clone` would also work for me. But preferably I would just stick with python nomenclature. Just my 2c.

gntech, 2018-05-26:
On second thought, I agree as well, keep it as pythonic as possible

fragmuffin, 2018-05-27:
This issue and #96 really are the same thing... I don't know what I was thinking ;)
So I've closed #96 as a duplicate.

fragmuffin, 2018-06-02:
I've been thinking about this a lot lately, and I've come to a simple, yet unhelpful conclusion:
> :thought_balloon: "This is not going to be easy"

What's been keeping me up at night:

**No forced geometry creation**
A copy must not force geometry to be created... because that geometry may not even be relevant to the particular script being run.
This is a strong design decision at the core of `cqparts` I'm determined to keep, but it is making thing's difficult.

**serializing / deserializing**
Copy traceability must be retained when serializing & deserializing (not necessary for `deepcopy`)
Does every part need a unique ID now?

**Copied part's class?**
What is the copied part instances class?
- new class type (eg: `PartCopy`)
- the original class (`type(orig_obj) is type(copied_obj)`) with modifications.
- a new class inheriting from both `type(orig_obj)` and `PartCopyMixin`
  (in which case, what happens with a copy of a copy?)

Anyway... work is being done, I'll keep you posted.

zwn, 2018-06-04:
If you go with the PartCopy idea, what happens when the primary Part is deleted?

I think that using an attribute pointing to something shared is probably the way to go. But I think that all instances should be equal - pointing to the same "something".

fragmuffin, 2018-06-04:
@zwn

> what happens when the primary Part is deleted?

Could you elaborate on that?
I can't think of a case where you would create a `Part`, then delete it.
In a conventional editor, yes, absolutely, but not in the context of `cqparts`

zwn, 2018-06-05:
> I can't think of a case where you would create a Part, then delete it.

Me neither but the language allows it so we should know what should happen in such a case. Maybe in a bigger project, with several maintainers... you know what happens to big old code where all the original authors have left 😉.

Anyway, I am thinking along the lines of what is "native" or "obvious" in python. There shallow copy of something is of the same type as the original (ie. copy of a `list` is a `list`).

fragmuffin, 2018-06-05:
@zwn

> Me neither but the language allows it so we should know what should happen in such a case.

Fair point 😉 .
With the `PartCopy` implementation in d524b8251adac8a9e862e0fd9dac05771706aafb the `copy(my_part)` returns a `PartCopy` instance, which holds a reference to the original object.
So that object can only be forgotten, and garbage-collected if it, and all its copies are deleted.
If reference to the original object is "deleted", the copies still have a reference, so it's not garbage-collected.

That's still true to python form:

```python
>>> l = [1, 2, 3]
>>> a = {'x': l, 'y': 100}
>>> a
{'x': [1, 2, 3], 'y': 100}
>>> del l
>>> l
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
NameError: name 'l' is not defined
>>> a  # still has a reference to 'l'
{'x': [1, 2, 3], 'y': 100}
```

> I am thinking along the lines of what is "native" or "obvious" in python. There shallow copy of something is of the same type as the original (ie. copy of a `list` is a `list`).

I agree, but I had **a lot** of trouble getting that to work smoothly... very very hacky code with lots of "magic"... so I reverted it before committing.
I came to the conclusion that: a copy of a `Part` behaves quite differently to a `Part`, but it's still inherited.
I understand that's a compromise, but I think it's necessary.

A `deepcopy` on the other hand, **will** definitely return an instance of the same class.
I'll still have to do a bit of "magic" to make sure geometry isn't built unless it's necessary.

fragmuffin, 2018-06-05:
Thinking about this some more... there could be a compromise between `copy(Cube())` returning:
- a `Cube` instance
- a `PartCopy` instance

What if `copy(Cube())` returned a class that inherits from `Cube` that's dynamically built.
Say `CubeCopy`.

```
>>> c1 = Cube()
>>> Cube._copy_class is None
True
>>> c2 = copy(c1)
>>> type(c2).__name__
'CubeCopy'
>>> Cube._copy_class is type(c2)
True
```

That way no "magic" needs to happen when requesting a copy's attributes

gntech, 2018-06-05:
I dont understand the workings of cqparts fully but what would be wrong with this:

What if `c1 = Cube()` only creates the local obj and `c1` is then wrapped in `a = WorldPart(c1)` class that has the woorld coordinates and makes a world obj from the local obj? `b = copy(a)` would have the same reference to `c1` but its own world coordinates and own world_obj.

`Cube()` only creates the actual geometry. `WorldPart()` wrapps a geometry and define world coordinates.

zwn, 2018-06-05:
> `Cube()` only creates the actual geometry. `WorldPart()` wraps a geometry and define world coordinates.

@gntech I was just about to suggest something very similar 👍 .

fragmuffin, 2018-06-05:
@zwn @gntech very interesting idea.
It's definitely much cleaner than what exists right now.

I've created #107 to discuss

fragmuffin, 2018-06-09:
I've put this on hold, prioritising #107 

I think #107 will remove the need for the implementation of `copy(my_part)` above.
In which case, this issue will just be the implementation of `deepcopy(my_part)`.