cqparts/cqparts #99: OCC test environment

fragmuffin, 2018-05-26:
Create OpenCascade (`occ`) docker environments.
The `occ` environments won't have freecad installed.k

## Base Environments

- [ ] rename `env/ubuntu-py2` -> `env/ubuntu-fc-py2`
- [ ] rename `env/ubuntu-py3` -> `env/ubuntu-fc-py3`
- [ ] create `env/ubuntu-occ-py2` (?)
- [ ] create `env/ubuntu-occ-py3`

## Test Environments
- [ ] rename `test/env/ubuntu-py2` -> `test/env/ubuntu-fc-py2`
- [ ] rename `test/env/ubuntu-py3` -> `test/env/ubuntu-fc-py3`
- [ ] create `test/env/ubuntu-occ-py2` (?)
- [ ] create `test/env/ubuntu-occ-py3`

justbuchanan, 2018-10-21:
Hi @fragmuffin, I'm interested in helping out with OCC support in cqparts. I'm not quite sure where to start and I don't know what's already being worked on, so if you can point me to the areas I could be the most useful, that would be great!

Testing seems like a good first step and I have some familiarity with the tools used in the project, so if you haven't started on the build environment for occ yet, I'd be happy to try and put something together.

fragmuffin, 2018-10-21:
Hi @justbuchanan , help with OCC support would be great!
I agree; the first step is to get a valid test environment.

**PRs to `feature/freecad-occ`**
The migration to OCC will be a big one, so it will be managed in chunks, all being merged into `feature/freecad-occ`. (any PR's to `master` for OCC upgrades will be rejected).
Then, when that branch is ready, and a supporting occ working version of `cadquery` is on pypi, `feature/freecad-occ` will be merged to master, and released.

**Test Environment**
`cqparts` uses docker to control environments for testing and deployment, base containers are defined in: [`/env`](https://github.com/fragmuffin/cqparts/tree/master/env), then the images built there are used by [`/tests/env`](https://github.com/fragmuffin/cqparts/tree/master/tests/env) and [`/deployment/env`](https://github.com/fragmuffin/cqparts/tree/master/deployment/env).

If you'd like to create a docker environment for `occ`, that would be a huge help!

**Add to CI**
We'd want to see these tests get run on _AppVeyor_ and _TravisCI_

**Test Results?**
I understand that almost every test will fail at this point... that's fine, and expected... we'll have to start somewhere.
The failed test-cases will give us a task-list for progress.

Let me know if that interests you.
Or if you'd prefer to contribute in another way, I'd be happy to set up the docker environment to get you started.

justbuchanan, 2018-10-23:
Sounds good, thanks for all the info and code pointers.

I've started playing around with getting the docker setup and have made some progress. I've got the occ fork of cadquery installed through conda and am able to run the test suite. Most of the tests fail, but that's to be expected :). As a test, I tried merging the capablerobots [fork](https://github.com/capablerobot/cqparts) of cqparts on top of my test env work and a lot more tests are passing.

I'll clean up the code a bit and send you a PR for review.

@zignig , thanks for the link to that script. That builds `oce` from source - is that preferable or should we use conda?

fragmuffin, 2018-10-23:
Sounds good @justbuchanan !

regarding testing on daily builds vs packaged content: I think the default environment for testing should be whatever we tell users to install (including the specific version, or just _latest_), which would be from `conda`.
If the _latest_ required content changes to be incompatible, the tests _should_ fail, because that behaviour will fail on a users box too.
Similarly, if we have passing test-cases from recent content only released as source, that doesn't help our users much.

Tagging: @CapableRobot & @osterwood

closed 2018-10-27 by fragmuffin