cqparts/cqparts #100: cqparts server display env' temporary files

fragmuffin, 2018-05-27:
Merged in with #89, then
adapted to a `DisplayEnvironment` as `CQPartsServerDisplayEnv` in #94 .
Implementation in [`src/cqparts/display/cqparts_server.py`](https://github.com/fragmuffin/cqparts/blob/master/src/cqparts/display/cqparts_server.py)

Temporary model export is not secure, and is never deleted.

**Security**
When displayed via cqparts server, the model is saved locally in the OS temp folder, accessible by anyone using that box.

See [`tempfile.mkdtemp()`](https://docs.python.org/2/library/tempfile.html#tempfile.mkdtemp) for security implementation.

**Storage Leak**
The lifespan of the folder is also not managed, so each displayed object's exported model may stay on the box indefinitely.

zignig, 2018-05-28:
Considering just uploading all the files to the cqparts-server and then delete the originals. Need top make sure that the temp files are cleaned out properly.

Perhaps, going forward alter the codec(s) so the generated models can reside in memory or even streamed directly.

Peering deep into the future it would be nice to be able to do a selective display / generation of .gltf files. Only generate the dirty components and then upload.

zignig, 2018-06-07:
can close as of #105