cqparts/cqparts #102: web-template Not Included in PyPi Release

jmwright, 2018-05-30:
This is related to [cqparts-server issue 1](https://github.com/zignig/cqparts-server/issues/1#issuecomment-392838442).

The [web-template directory](https://github.com/fragmuffin/cqparts/tree/master/src/cqparts/display/web-template) doesn't seem to be included in the PyPi package, and so I get an error when trying to use cqparts-server after installing cqparts using pip. If you download the .tar.gz file directly from PyPi, you'll see it's not in there. This may be intentional, but I'm not sure how else to get the web renderer to work.

fragmuffin, 2018-05-30:
@jmwright you're absolutely right. this has been fixed in the `master` branch in #78
However, I forgot that this hasn't been released yet!
I'll release 0.2.1 tonight ([changelog](https://github.com/fragmuffin/cqparts/blob/master/deployment/CHANGELOG.md)) (in about 6hrs)

closed 2018-05-30 by fragmuffin

jmwright, 2018-05-30:
Cool, thanks!

fragmuffin, 2018-05-30:
@jmwright I'm having trouble with the release.
I tested #78 before merging, but I can't get the build working and installing in a docker container.

```
 error: can't copy 'cqparts/display/web-template/model': doesn't exist or not a regular file
```

release will be delayed, I'll try to sort this out as soon as I can.

zignig, 2018-05-30:
I possibly have an update for the web interface , perhaps it can wait until the deploy is stable. 

@fragmuffin @jmwright should we also look at pushing cqparts display() back into CaqQuery ?

fragmuffin, 2018-05-30:
Fixed it!.. `0.2.1` is released... I'll announce in a separate "issue"
I was re-discovering ancient bugs because `pip` and `setuptools` weren't updated on my test environment (fixed in 5c5d0ab445e52932d494ee65f9616a30dedc5c43).

@zignig I'd be happy to release more often than I have been, because `cqparts` is still in alpha, I'm not worried about pushing releases out too quickly.
I've just fixed up my deployment environment a little, so that will make it a bit easier... also, when #79 is closed, deployment should only take me about 15min instead of a few hours.

fragmuffin, 2018-05-30:
@zignig the `cqparts.display` will always be there because with `cqparts` as a wrapper to `cadquery`, there will always be additional information relevant to display that `cadquery` shouldn't have to know about; to remain compartmentalised.

When the `cadquery` standalone GUI makes its debut, we'll figure out what's passed straight through to `cadquery`, and what needs to be send directly to the GUI as metadata.

jmwright, 2018-05-30:
> 0.2.1 is released.

Thanks @fragmuffin . I'll get it updated in the FreeCAD module too.

zignig, 2018-05-30:
OK, #79 is on it's way, good job. 

I would like to get the web interface bolted down (hehe) for the next release then. Use the same base for the pop up window and the cqparts-server interface.  Simple and clean, no iframes, just an html rectangle.

fragmuffin, 2018-05-30:
@jmwright hold on!, don't congratulate me too quickly (I probably shouldn't do this while I'm so tired).
I've done a terrible job with the licensing... everything still says it's a GPLv3 license.
I'm updating that now, but I know from experience that I can't "undo" the v0.2.1 release I've just done.
Which means a v0.2.2..... but that's no big deal.

fragmuffin, 2018-05-30:
@jmwright @zignig on second thought... it's silly to do a second release that's functionally identical **just** to fix the metadata.
`v0.2.1` **is** apache licensed, the metadata will be fixed in the next release, which will probably be soon.