cqparts/cqparts #106: Make components hashable

fragmuffin, 2018-06-05:
`Component` instances are not immutable, so they cannot be "hashed".

Parameters can be changed before their geometry is created.
For example:

```python
>>> from cqparts_misc.basic.primatives import Cube
>>> cube = Cube(size=10)
>>> cube.size = 20
>>> cube.local_obj  # returns a 20x20x20 cube
```

This provides a freedom, but can also be confusing...

```python
>>> from cqparts_misc.basic.primatives import Cube
>>> cube = Cube(size=10)
>>> cube.local_obj  # returns a 10x10x10 cube
>>> cube.size = 20  # does NOT reset cube.local_obj
```

zwn, 2018-06-05:
@fragmuffin What is the use case for unlocked parameters? Is it important to be able to update parameters later on in the same script?

fragmuffin, 2018-06-05:
I've found it useful when multiple parts are unioned to create a single part.
Specifically [`MaleFastenerPart` : line 177](https://github.com/fragmuffin/cqparts/blob/master/src/cqparts_fasteners/male.py#L177), but that could be changed, and it's probably the only time I've used it.

> Is it important to be able to update parameters later on in the same script?

So... I'm thinking "no"; perhaps it's not.

However, that doesn't change a _component's_ behaviour with regard to changing `local_obj`.
In the case of changing its `local_obj` geometry, it would need to start unlocked, then be locked in order to make it hashable.

In which case... we may as well have the parameters changeable, but they do need to be more strict.