cqparts/cqparts #107: Separate local from world by class

fragmuffin, 2018-06-05:
Should _parts_ have a separate class instance for it's _world_ presence?

**Current Implementation**

```python
>>> from cqparts_misc.basic.primatives import Cube
>>> from cqparts.utils import CoordSystem

>>> cube = Cube()
>>> cube.local_obj  # geometry in the part's native coordinate system

>>> cube.world_coords = CoordSystem((1, 2, 3))
>>> cube.world_obj  # geometry translated by (1, 2, 3)
```

**Potentially changed to**

```python
>>> from cqparts_misc.basic.primatives import Cube
>>> from cqparts.utils import CoordSystem
>>> from cqparts import WorldPart

>>> cube = Cube()
>>> cube.obj  # geometry in the part's native coordinate system

>>> world_cube = WorldPart(cube)
>>> world_cube.coords = CoordSystem((1, 2, 3))
>>> world_cube.obj  # geometry translated by (1, 2, 3)
```

As suggested by @gntech @zwn in https://github.com/fragmuffin/cqparts/issues/97#issuecomment-394616079

fragmuffin, 2018-06-05:
`Assembly` components would need to be split into 2 groups.
- pre-solved local Parts
- solved world Parts

There would also need to be a `WorldAssembly` equivalent of `WorldPart`.

zwn, 2018-06-05:
Well, I am thinking about a system where the actual position and orientation of Parts is completely external to them. In such a system the pose of a Part would only be part of its containing Assembly and it would be based on CoordinateSystem of the containing Assembly. Such an Assembly could itself be easily placed within another Assembly just as Part can be. The top level Assembly CoordinateSystem would be the world coordinates. Actually just like a scene graph.

fragmuffin, 2018-06-06:
@zwn @gntech @zignig 
I'm liking that a lot... just thinking through how it would work for an `Assembly`.
Using the old [toy car](https://fragmuffin.github.io/cqparts/doc/tutorials/assembly.html) example:

```python
>>> car = Car()
>>> car.build()
>>> wheel = car.find('front_axle.left_wheel')
>>> isinstance(wheel, cqparts.Part)
True

# local
>>> type(wheel.local_obj)
cadquery.Workplane

# world
>>> type(wheel.world_coords)
cqparts.utils.CoordSystem
>>> type(wheel.world_obj)
cadquery.Workplane
```

**Question**
How would you like that interface to go?
The only real interaction change would be getting a specific component's world coordinates... because:

```
>>> wheel.world_obj  # equivalent to...
>>> wheel.world_coords + wheel.local_obj
```

Also a scene graph structure would give us 2 coordinate systems:
1. relative to its parent `Assembly`
1. accumulated of all parents which would be the part's _world_ coordinate system

zignig, 2018-06-06:
I like it. 

@fragmuffin  if the coordinates inside a assembly are localised, would it mean that the Fasteners and some of the make_alterations would not need the world to local and back again flipping for the calculations.

Also, if  only the top level was in world coordinates you could extrapolate an exploded view down through the hierarchy. I've been pondering how to auto explode assemblies, as they get more levels this becomes increasingly important, and it looks so cool. ;)

fragmuffin, 2018-06-06:
@zignig 
> if the coordinates inside a assembly are localised, would it mean that the Fasteners and some of the make_alterations would not need the world to local and back again flipping for the calculations

yeah!, I think these calculations would be less fiddly.
But only if we have a clean and intuitive way of getting `CoordSystem` instances from solved assemblies.

> I've been pondering how to auto explode assemblies, as they get more levels this becomes increasingly important, and it looks so cool. ;)

It would also make exploding a simpler process... one way to do it would be to recursively offset assemblies and parts by a factor along their local `Z` axis.
It would look cool! and I think that's important... `cadquery` and `cqparts` as tools are still being adopted by the larger community, and visual appeal is a big part of that.

fragmuffin, 2018-06-07:
Regarding the **new** behaviour of an `Assembly` (ignore what's written in this issue's description).
What do you think of the following?.... again using the [toy car](https://fragmuffin.github.io/cqparts/doc/tutorials/assembly.html) as an example.

```python
>>> car = Car()
>>> wheel = car['front_axle']['left_wheel']  # equivalent of...
>>> #wheel = car.find('front_axle.left_wheel')

# wheel is the original part wrapped in a WorldPart class
>>> type(wheel)
<WorldPart: ...>
>>> type(wheel.wrapped)  # original Part instance added to the assembly
<Wheel: diameter=30.0 ...>
>>> wheel.diameter  # parameters passthrough to wrapped instance
30.0
>>> wheel.diameter = 20  # cannot set parameters
ValueError: cannot set parameters of a WorldPart
>>> wheel.local_obj  # local object is a passthrough to the wrapped object's local geometry
<cadquery.cq.Workplane ...>

# WorldPart class holds
#    - coordinate systems
#    - buffered translated geometry (for assembly, and world, only transformed from local on request)
#    - link to parent assembly
>>> wheel.asm_coords  # the object's coordinate system relative to its parent assembly
<CoordSystem: ...>
>>> wheel.world_coords  # the object's world coordinate system
<CoordSystem: ...>
>>> wheel.asm_obj  # equivalent of wheel.asm_coords + wheel.local_obj
<cadquery.cq.Workplane ...>
>>> wheel.world_obj  # equivalent of wheel.world_coords + wheel.local_obj
<cadquery.cq.Workplane ...>
>>> wheel.parent  # the component's parent for calculating "world" coordinates
<WheeledAxle: ...>
```

And, of course, the _world_ vs _local_ in a `Part` would be removed

```python
>>> cube = Cube(size=10)
>>> cube.obj  # changed from 'local_obj' to just 'obj'
<cadquery.cq.Workplane ...>
```

zwn, 2018-06-07:
I am wondering what exactly `WorldPart` is in this scenario or what role it has:

 * Is it created on request when you take something out of a built assembly?
 * Is it a generic node (somewhat internal) in the built scene graph?
 * Is it something that is part of the API the user uses to build the final `Assembly`?

So far I am 👍 on the last paragraph where `Part` has only `.obj` which is local.

fragmuffin, 2018-06-07:
@zwn 

> Is it a generic node (somewhat internal) in the built scene graph?

Is probably the best description.

In the current implementation, each component of an `Assembly` gets its `world_coords` set when the constraints are solved.

So `WorldPart` (which I'm now thinking should inherit `WorldComponent`) would be applied internally by the _constraint solver_, but could also be used independently to more manually represent a _component_ somewhere other than its origin coordinate system.

For example `CoordSystem((1, 2, 3)) + Cube()` would return a `WorldPart` instance that wraps the `Cube` instance

fragmuffin, 2018-06-09:
I think this is the right direction, I'm going to start dev on this this week-end.

fragmuffin, 2018-06-09:
I've promoted this issue from a question to a change-request... working on it now.

fragmuffin, 2018-06-09:
oops

fragmuffin, 2018-06-09:
**Interface Changes**
I have to change how mates, and the constraints solver works.
This is because the solver relies on mates linking to `Component` instances that have `world_coords`.
Now that that's been moved to a wrapper, the `Mate` instance will need a reference to the wrapper, which can't be done by the wrapped class.

I've wanted to change how mates work anyway, so this seems like a good opportunity to do so.

```python
class Box(cqparts.Part):
    size = PositiveFloat(10)
    def make(self):
        return cadquery.Workplane('XY').box(self.size, self.size, self.size)
    
    @mate
    def top(self):
        return CoordSystem((0, 0, self.size / 2))
```

the `@mate` decorator can help group all mates together, and empowers the `solver` to create `Mate` instances with appropriate links to other `Placed` components.

This also changes `make_constraints()` definitions from
```python
Coincident(
    self.components['small_box'].mate_bottom,
    self.components['big_box'].mate_top,
),
```

To something like:
```python
Coincident(
    self['small_box'].mate('bottom'),
    self['big_box'].mate('top'),
)
```

Where `my_assembly[name]` returns a `Component.Placed` instance.
Before `my_assembly.solve()` is run, these instances will have a coordinate system of `None`.
When the solver is done, they'll all have valid `CoordSystem` instances for their `coords` (or an exception is raised... just like it does currently)

zignig, 2018-06-09:
This is cleaner.

Changing to a @mate decorations is more specific than the @property, it also give the opportunity to enumerate available mate points. It would be possible to have a component raise an exception if nothing is bound to an existing required mate point.

I am attempting to look at cqparts not in the sense of python code but a way query and distil tangible objects. The defaulting of parameters and applying an overlay onto existing design is still eluding me.

MyStepper = stepper.Stepper().search("NEMA17")
MyBolt = bolt.Bolt().search("3Mx6_caphead")
cnc = ThreeAxisCNC(bolts=Mybolt,motor=MyStepper)

hmmm, Needs some more GURU MEDITATION.

fragmuffin, 2018-06-11:
@zignig 

`@mate` decorator implemented in f4165d6166ca19d47cbf768a609d443fa328d516
what do you think?

Unfortunately PR #110 is going to brake (almost) everything that's been made using `cqparts` up to this point, but I think it's worth it.
I'm just glad `cqparts` is still in alpha, breaking updates like this can be justified.

zignig, 2018-06-11:
WHHAAAT ! i have to rewrite all my examples !  :)

I think getting the structure at the bottom is important , using the  assembly['sub'] syntax is much cleaner than assembly.components['name'] . As for the mate change go for it, there will probably need to be a selection of @decorators going forward. They should be used sparingly.

fragmuffin, 2018-06-12:
> WHHAAAT ! i have to rewrite all my examples ! :)

I have to rewrite EVERYTHING!... well, that's a bit dramatic, but I'll have a lot to change.

> using the `assembly['sub']` syntax is much cleaner than `assembly.components['name']`

There may be a bit of confusion here, because `assembly['sub']` will return a "placed component":

```python
>>> isinstance(assembly.components['name'], cqparts.Component)
True
>>> isinstance(assembly['name'], cqparts.Component.Placed)
True
>>> assembly['name'].wrapped is assembly.components['name']
True
```

I'm still designing this interface.

I don't like that confusion, but I don't know how else to do it.
The power of that interface is a component can be used multiple times; a single `cqparts.Component` can be "placed" multiple times.

```python
class WheeledAxle(cqparts.Assembly):
    def make_components(self):
        wheel = Wheel()  # a cqparts.Part
        return {
            'axle': Axle(),  # another cqparts.Part
            'left_wheel': wheel,
            'right_wheel': wheel,  # the same object for left and right
        }
    def make_constraints(self):
        return [
            Fixed(self['axle'].mate('origin'), CoordSystem()),
            Coincident(self['left_wheel'].mate('origin'), self['axle'].mate('left')),
            Coincident(self['right_wheel'].mate('origin'), self['axle'].mate('right')),
        ]
```

A "placed" component wraps the given component, and gives it placement (both assembly coordinates, and a link to its parent, giving it world coordinates).
So in the above `self['left_wheel']` and `self['right_wheel']` are 2 `Part.Placed` instances, but both wrap the same object

```python
# scoped in make_constraints above
>>> self['left_wheel'].wrapped is self['right_wheel'].wrapped
True
```

This also means I'll have to re-consider how to apply Constraints to models... my thinking hat is on.

zignig, 2018-06-12:
Ah ok, that changes it but it still works. In some of my bucket models i am passing down a bare function and the instantiating multiple copies inside the assembly.

Wrapping the object will have the advantage that (in the fullness of time) , the gltf export can reference the same binary object rather than having to generate the same part multiple times.

Will need to watch for alterations on objects after wrapping so a modified object is seen as a distinct instance.

zignig, 2018-06-12:
Or is a explicit .clone() function the go? The syntax might get confusing.

fragmuffin, 2018-06-12:
@zignig 

> Will need to watch for alterations on objects after wrapping so a modified object is seen as a distinct instance.
> Or is a explicit .clone() function the go? The syntax might get confusing.

No; I think the `.clone()` function is scrapped in place of this `Component.Placed` wrapper.
copy functionality still needs to be implemented, but it will be the `deepcopy` implementation discussed.

as for local object changes after it's passed to a scene graph node (wrapped in `Component.Placed`).
I think #106 is a discussion worth having in this respect...
in other words: perhaps the local object should be read-only when it's wrapped in a `Component.Placed`.

However, I'm not sure I like that either, because that means you can't make local geometry changes based on the object's placement.

I think I should work on some more examples using the new API, to illustrate how it'll work, but it'll also  get me to think more about it.

zignig, 2018-06-13:
Have a look at https://github.com/zignig/cqparts-bucket/blob/master/cnc.py for how I am doing multiple instances of objects. 

Not sure it's the best way. but it does to an override able default construct.

zignig, 2018-06-14:
I completely missed the point of your change..... A better example would be the case that has 8 of the same screw. Create one screw and have the other 7 referencing (archetypal) the original instance.

carry on. ;)

fragmuffin, 2018-06-14:
@zignig 

Oh, yes, I had a look at your `cnc.py` work, I believe I've done something similar for [fasteners `male.py`](https://github.com/fragmuffin/cqparts/blob/master/src/cqparts_fasteners/male.py) (namely the `head`, `drive`, and `thread` parameters).

> A better example would be the case that has 8 of the same screw. Create one screw and have the other 7 referencing (archetypal) the original instance.

Yes, exactly.
It's intended to serve the designer so they may make an array of parts or assemblies, and it should be as efficient as possible.
But it also serves to reduce replicated geometry in the scene graph on display & export.

I'm having a bit of trouble with designing this concept... it's a big change, but I can only work at it for ~1hr at a time (time constraints).
And it's really not the sort of thing that can be done in small chunks.

I'll get it there eventually :+1: 

I think I'll continue to use the `Car()` :car: as the example, with only 1 `Wheel` instance, and only 1 `WheeledAxle` instance... That should be a good proof of concept, with only 4 `Part` and 2 `Assembly` instances.

zignig, 2018-06-14:
> ~1hr at a time (time constraints).

grr , RealLife(tm) getting in the way again?  I have two midgets so I understand.

keep up the good work. I'm still enjoying the building of stuff.  Found a working blender importer (https://github.com/ksons/gltf-blender-importer) so cqparts-server will have nice renders at some point.

zignig, 2018-06-15:
once you are happy with your changes or before, put the canonical car into the examples folder. I will read , grok and convert my existing builds for the creation of a plump and yummy examples folder.

fragmuffin, 2018-06-15:
Will do @zignig ! 
Thanks for your support!