cqparts/cqparts #111: Assembly Issue

jmwright, 2018-06-14:
I have [this cqparts assembly](https://stemn.com/files/5b15410c252c9400192ab940/5b161126b1527c0019ec0eef?revision=5b217b7178761f00199167a6) where I'm just trying to get two parts placed together. I'm just throwing them together for now. If I do a fixed constraint on both parts they import fine, but if I do a coincident one, the servo is returned as None, which causes the `display()` call to fall.

Oddly, this works fine in FreeCAD 0.17 and Python 2.7.14. The problem only manifests itself with FreeCAD 0.17 and Python 3.6.5. In that version I get the following error every time I execute the assembly script.
```python
solving for Assembly without word coordinates set: <RobotAssembly: >
```
I am running on Ubuntu 17.10 64-bit.

jmwright, 2018-06-14:
I should amend this to say that 0.17/2.7.14 doesn't throw any errors, but the mates don't seem to have any effect. I'm not sure if I'm just missing a step, or if there's a bug.

fragmuffin, 2018-06-14:
@jmwright 

The "solving for Assembly without word coordinates set" message should just be a warning.
It's just letting you know that the `Assembly` you're displaying doesn't have world coordinates, so the build process _assumes_ the `Assembly` is at the world origin.
You should be able to get rid of this warning by setting `world_coords` before building:

```python
robot_assembly.world_coords = CoordSystem()
```

What you're doing looks fine to me... I'll look into it after work.

fragmuffin, 2018-06-14:
https://stemn.com looks really interesting, but I've got to say, I'm having a really bad experience with it :cry: . Hopefully its Beta stage is kind to it.
But to the problem at hand...

@jmwright 

> If I do a fixed constraint on both parts they import fine, but if I do a coincident one, the servo is returned as `None`, which causes the `display()` call to fall.

I can't seem to replicate this issue.
The closest I have to the environment you describe is a deployment docker container: [`cqparts-deploytest:ubuntu-py3`](https://github.com/fragmuffin/cqparts/tree/master/deployment/env/ubuntu-py3)

If I inject a `ipdb.set_trace()` into the script just before it's displayed, my environment is:

```python
__import__('ipdb').set_trace()

>>> cq.__version__
'1.2.0'
>>> cqparts.__version__
'0.2.1'

>>> import sys
>>> sys.version_info
sys.version_info(major=3, minor=5, micro=2, releaselevel='final', serial=0)

>>> import FreeCAD
>>> FreeCAD.Version()
['0', '17', '13062 (Git)', '/opt/conda/conda-bld/git_cache/github.com/FreeCAD/FreeCAD master', '2018/01/20 13:09:58', 'master', 'f6825a2686d80b0976c3d1900a9f56ce3c9f7d91']

>>> robot_assembly.components['servo1']
<Servo: mount_hole_rad=2.25, mount_wing_offset=9.5, mount_wing_overhang=15.2, mount_wing_thickness=2.5, overall_height=36.1, overall_length=40.3, overall_width=20.0>
```

If I've understood the problem you're experiencing correctly, the last line should have returned `None`.

jmwright, 2018-06-14:
> https://stemn.com looks really interesting, but I've got to say, I'm having a really bad experience with it 😢 . Hopefully its Beta stage is kind to it.

I'm using it to help provide feedback and because I think it has potential. They recently added pipelines, which allow you to run scripts and commands against your repo. I'm using pipelines to automatically generate STEP files from the cqparts scripts for previewing, and do BoM generation. STEMN is definitely not ready for production use, but they'd love to have more open hardware feedback to help guide the development.

> If I've understood the problem you're experiencing correctly, the last line should have returned None.

In my Python 3 environment it returns None for the servo. In my Python 2 environment it returns the objects correctly, but the constraints don't seem to have any effect. I'm doing all this from inside the FreeCAD module.

jmwright, 2018-06-14:
Here's a screenshot of the assembly in FreeCAD 0.17 running in a Python 2.7 environment. Notice how the servo is positioned as if I had used a fixed constraint at the origin. I would expect it to be embedded in the left wall of the chassis (orange part).

[image]

jmwright, 2018-06-14:
Sorry, one more thing. If you're having trouble cloning it on STEMN, [here's a link](https://www.dropbox.com/s/5xqbizkggsgcqln/Asymmetrical%20Quadruped%20Robot%20Chassis.zip?dl=0) to the zip file on Dropbox.

fragmuffin, 2018-06-15:
@jmwright 

> In my Python 3 environment it returns None for the servo. [...] I'm doing all this from inside the FreeCAD module.

You mentioned that `robot_assembly.components['servo1'] is None`, but there's still servo geometry in that screenshot... so I'm a bit confused :thinking:

Your code is iterating through each component of `robot_assembly`, in this case each one is a `Part`.
If a `Part` instance is passed to `display` then its `local_obj` will be displayed.
If the `Part` is a component of an `Assembly`, then its `world_obj` is displayed.
Implementation: https://github.com/fragmuffin/cqparts/blob/20180530/src/cqparts/display/freecad.py#L41

Whether this is a _bug_ or a _feature_ is not so relevant at this point because `cqparts` is undergoing a bit of an overhaul: #107 

> if I do a coincident one, the servo is returned as `None`, which causes the `display()` call to fall.

I'm assuming an exception is thrown, could you copy that text in?

jmwright, 2018-06-15:
The last few lines of the assembly script are where I went wrong. I was stepping through and displaying the components one-by-one because I was trying to use `show_object()` from CQGI originally. I didn't understand you have to call `display()` on a whole assembly to get the constraints to work (seems obvious in hindsight). Because I was displaying them one at a time, the parts were always placed at the origin, instead of the constraints being solved and the parts being placed where they should. Everything is working fine now. As I've updated the CadQuery FreeCAD module to get rid of the last few intermittent Python 3 bugs, I think I've gotten rid of the NoneType display error too.

Thanks for the help.

closed 2018-06-15 by jmwright