cqparts/cqparts #113: Mates failing with object acess

zignig, 2018-06-21:
I am getting some weirdness with accessing objects causing assemblies to fail on mates

I have a working example fail here 

https://github.com/zignig/cqparts-bucket/blob/master/bork.py

The relevant part is 

class broken(cqparts.Assembly):

    def make_components(self):
        a = stack()
        b = stack()
        comps = {
            'a' : a,
            'b' : b
        }
        #t = b.components['box'] <- uncomment this line to break
        return comps

    def make_constraints(self):
        constr = [
            Fixed(self.components['a'].mate_origin),
            Coincident(
                self.components['b'].mate_origin,
                Mate(self,CoordSystem(origin=(20,0,0)))
                )
            ]
        return constr

if I un-comment  the assignment , the second instance of the 'stack()'  has a misplaced object.

When I was building the mounted steppers and tried to access sub objects , I experienced this weirdness.

zwn, 2018-06-21:
I believe that `components` property is populated by the result of the call to `make_components`.

zignig, 2018-06-21:
How would I access a property of a sub assembly in place then ?  could I use find('name') instead ?

fragmuffin, 2018-06-23:
@zwn 

> I believe that `components` property is populated by the result of the call to `make_components`

Yes, that's correct, but that shouldn't break the build.

@zignig 

> How would I access a property of a sub assembly in place then ? could I use `find('name')` instead ?

You _should_ be able to do this, but you're absolutely right, this is definitely a bug :bug: .
`find()` is just a utility to access nested assemblies, sort of like an XPath in XML... so using `find()` won't be any different.

**Explanation of Error**
This error is occurring because when accessing an assemblies `component` dict, it (as @zwn stated) builds the assembly.
Part of the build process is assigning each object's `world_coords`.
The build process is designed very simplistically to build from the trunk upward, it assumes that nothing is given `world_coords` until it has been _solved_.

Key lines are:
- [if `components` hasn't been set yet, call `build`](https://github.com/fragmuffin/cqparts/blob/20180530/src/cqparts/assembly.py#L90)
- [`Assembly.build` calls `solve`](https://github.com/fragmuffin/cqparts/blob/20180530/src/cqparts/assembly.py#L264)
- [`Assembly.solve` calls `solver`, passes `self.world_coords`](https://github.com/fragmuffin/cqparts/blob/20180530/src/cqparts/assembly.py#L125)
- if the given `world_coords is None`, they're [assumed to be 0](https://github.com/fragmuffin/cqparts/blob/20180530/src/cqparts/constraint/solver.py#L23)

obviously, this logic isn't going to work in your case; when you ask for the `components` of a sub-assembly, the build process applies the solver, which sets the position of the cylinder.
I believe the square base is unaffected because it's positioned with a `Fixed` constraint, which is not bound by any conditions; it just forces a component's `world_coords` to the given value (+ any offset).

**Solution**
I'm thinking the solution to this problem is also bound in the work I'm doing in #107 (stalled, but I'm still working on it).

Ultimately there will be 2 "build" processes:
- to create the hierarchy of components
- to place each component in its sub-assembly (which can be recursively summed to get world coordinates)

**Issue Closure Criteria**
In the interest of regression testing, I'll keep this issue open until I:
- [ ] create unit test-case(s) to expose "backwards" build behaviour (after #107 is closed)

Nice find @zignig !, thank you for reporting it :+1:

zignig, 2020-07-10:
cleaning out old  issues, 

probably resolved with 

https://github.com/cqparts/cqparts/pull/158

zignig, 2020-07-10:
oops

closed 2020-07-10 by zignig