cqparts/cqparts #115: STEPPartImporter Not Working

jmwright, 2018-07-08:
I'm not sure what I'm doing wrong, but I'm trying to import a STEP file using something like the following code...
```python
...
from cqparts.codec.step import STEPPartImporter
...
class RobotAssembly(Assembly):
    def make_components(self):
        chassis = Chassis()
        stepper1 = Stepper()
        stepper2 = Stepper()
        stepper3 = Stepper()
        stepper4 = Stepper()
        heatset1 = STEPPartImporter('./parts/cots/94180A312_HEAT-SET INSERTS FOR PLASTICS.STEP')

        return {'chassis': chassis, 'stepper1': stepper1, 'stepper2': stepper2,
                'stepper3': stepper3, 'stepper4': stepper4, 'heatset1': heatset1}
```
... but I get the following error on the return statement. No error is thrown on the import.
```python
Running the Python command 'CadQueryExecuteScript' failed:
Traceback (most recent call last):
  File "/home/jwright/.FreeCAD/Mod/cadquery-freecad-module/CQGui/Command.py", line 232, in Activated
    imp.load_source('__cq_freecad_module__', tempFile.name)
  File "/tmp/tmplYhe9O", line 51, in <module>
    robot_assembly.world_coords = CoordSystem()
  File "/home/jwright/.FreeCAD/Mod/cadquery-freecad-module/ThirdParty/cqparts/component.py", line 49, in world_coords
    self._placement_changed()
  File "/home/jwright/.FreeCAD/Mod/cadquery-freecad-module/ThirdParty/cqparts/assembly.py", line 115, in _placement_changed
    self.solve()
  File "/home/jwright/.FreeCAD/Mod/cadquery-freecad-module/ThirdParty/cqparts/assembly.py", line 125, in solve
    for (component, world_coords) in solver(self.constraints, self.world_coords):
  File "/home/jwright/.FreeCAD/Mod/cadquery-freecad-module/ThirdParty/cqparts/assembly.py", line 105, in constraints
    self.build(recursive=False)
  File "/home/jwright/.FreeCAD/Mod/cadquery-freecad-module/ThirdParty/cqparts/assembly.py", line 226, in build
    self.verify_components(new_components)
  File "/home/jwright/.FreeCAD/Mod/cadquery-freecad-module/ThirdParty/cqparts/assembly.py", line 159, in verify_components
    ) % (name, component))

invalid component type from make_components(): ('heatset1', <cqparts.codec.step.STEPPartImporter object at 0x7f2f086fea10>) (must be a (str, Component))
```
Have I made an obvious mistake? Isn't the importer call supposed to return a Part object?

fragmuffin, 2018-07-08:
@jmwright 
`STEPPartImporter` is a class that does *not* inherit from `Component`... hence the error.
The `STEPPartImporter` is a `Component` factory class that will return an instance of a class it creates t import the STEP file you give it.

Have a read of https://fragmuffin.github.io/cqparts/doc/cqparts/import-export.html?highlight=import#getting-an-importer

```python
filename = './parts/cots/94180A312_HEAT-SET INSERTS FOR PLASTICS.STEP'

# change this...
heatset1 = STEPPartImporter(filename)

# to this
heatset1 = cqparts.Part.importer('step')(filename)
# if the step file has multiple parts, you'll want
heatset1 = cqparts.Assembly.importer('step')(filename)
# If you don't know what's in the file, I'd recommend going with the Assembly importer.
```

Let me know if that solves your problem.

jmwright, 2018-07-08:
Ok, sorry, I should have spent more time in the docs. I got fixated on STEPPartImporter and didn't look hard enough for anything else.

I still get an error, but it's different now.
```python
Running the Python command 'CadQueryExecuteScript' failed:
Traceback (most recent call last):
  File "/home/jwright/.FreeCAD/Mod/cadquery-freecad-module/CQGui/Command.py", line 232, in Activated
    imp.load_source('__cq_freecad_module__', tempFile.name)
  File "/tmp/tmpKETji6", line 52, in <module>
    robot_assembly.world_coords = CoordSystem()
  File "/home/jwright/.FreeCAD/Mod/cadquery-freecad-module/ThirdParty/cqparts/component.py", line 49, in world_coords
    self._placement_changed()
  File "/home/jwright/.FreeCAD/Mod/cadquery-freecad-module/ThirdParty/cqparts/assembly.py", line 115, in _placement_changed
    self.solve()
  File "/home/jwright/.FreeCAD/Mod/cadquery-freecad-module/ThirdParty/cqparts/assembly.py", line 125, in solve
    for (component, world_coords) in solver(self.constraints, self.world_coords):
  File "/home/jwright/.FreeCAD/Mod/cadquery-freecad-module/ThirdParty/cqparts/assembly.py", line 105, in constraints
    self.build(recursive=False)
  File "/home/jwright/.FreeCAD/Mod/cadquery-freecad-module/ThirdParty/cqparts/assembly.py", line 224, in build
    components_iter = genwrap(self.make_components(), "make_components", dict)
  File "/tmp/tmpKETji6", line 21, in make_components
    heatset1 = cqparts.Part.importer('step')(os.environ.get("MYSCRIPT_DIR") + '/parts/cots/94180A312_HEAT-SET INSERTS FOR PLASTICS.STEP')
  File "/home/jwright/.FreeCAD/Mod/cadquery-freecad-module/ThirdParty/cqparts/codec/step.py", line 87, in __call__
    'make': make,

type() argument 1 must be string, not unicode
```

I'm running this in FreeCAD stable in Ubuntu 17.10 using the latest version of the CadQuery workbench.
```
OS: Ubuntu 17.10
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.17.13522 (Git)
Build type: None
Branch: releases/FreeCAD-0-17
Hash: 3bb5ff4e70c0c526f2d9dd69b1004155b2f527f2
Python version: 2.7.14
Qt version: 4.8.7
Coin version: 4.0.0a
OCC version: 7.2.0
Locale: English/UnitedStates (en_US)
```
A side note for anyone using cqparts from the CadQuery workbench: The workbench executes scripts in a temporary file, so if you want to access step files embedded with your project, you can use the [environment variables](https://github.com/jmwright/cadquery-freecad-module/blob/master/docs/usage.md#environment-variables).

fragmuffin, 2018-07-08:
@jmwright 

> Ok, sorry, I should have spent more time in the docs

No problem mate! that's what I'm here for ;)

also... fixing bugs...

> I still get an error, but it's different now.

That's a good find! I believe that's being raised because I'm trying to set the name of a dynamically created class with `unicode` (I would have guessed that was a python3 issue, but you're using `2.7`)

I'll fix this later today.

Thanks for the feedback @jmwright !

jmwright, 2018-07-08:
Great, thanks.

I'm trying to figure out how to handle cqparts updates in the FreeCAD workbench. I update the CadQuery library between releases so that users always have access to the latest version. I'm guessing I should do the same with cqparts.

fragmuffin, 2018-07-08:
@jmwright 

> I'm trying to figure out how to handle cqparts updates in the FreeCAD workbench. I update the CadQuery library between releases so that users always have access to the latest version

Hmm, but as you get more libraries, you'll end up doing a lot of work in between releases... I guess you could automate the process so each user's installation can check for updates and install them as they're released, but then you'd also need a way to intentionally hold back to a particular version.

This design could very quickly lead to making your own mini package management system, but instead...
Could you use `pip install --upgrade --target cadquery/libs/folder -r requirements.txt` where the `requirements.txt` file can be edited by an expert user to isolate particular versions (if they want).

jmwright, 2018-07-09:
I've got a script to update all the requirements from PyPi, but I run weird setups sometimes so that users can get the latest features quickly, and I like to have (and sometimes need) manual control. At this time, I'm only maintaining CadQuery and PyQode updates manually.  CadQuery to roll features out more quickly, PyQode because it's abandonware and I have to make the updates myself. I don't think that adding cqparts to the manual maintenance mix will add that much more work.

fragmuffin, 2018-07-09:
I believe #116 has fixed that issue. Issue closed.

@jmwright if that hasn't fixed the issue you're experiencing, let me know and I'll re-open this issue.

closed 2018-07-09 by fragmuffin

jmwright, 2018-07-09:
Thanks, that fixed it.

I took the opportunity to update everything cqparts related in the FreeCAD workbench. I did all the cqparts_* directories, even if they are just placeholders. Users will be able to get the latest by updating through the add-ons manager.