cqparts/cqparts #129: OCC compatibility TODO list

justbuchanan, 2018-11-29:
I'm opening this issue as a TODO list of what else needs to be done on the [feature/freecad-occ branch](https://github.com/cqparts/cqparts/tree/feature/freecad-occ) to finish up compatibility with the [pythonOCC fork of cadquery](https://github.com/cadquery/cadquery). The initial list is just the result of running the test suite and seeing what breaks. See the test.log attached [here](https://github.com/cqparts/cqparts/pull/127#issue-233435136).

- [ ]  cadquery-occ needs to provide accessors for individual `Matrix` elements. Maybe through `__get__()`? cqparts currently inspects matrix elements via `m.A11, m.A12, etc` which don't currently exist in the OCC fork:
https://github.com/cqparts/cqparts/blob/f057c2d413dfa3552289d57a132ea077c0152e4f/tests/t_cqparts/test_coordsystem.py#L20-L24
- [ ] Fix `import Part as FreeCADPart` in threads/base.py.
- [ ] Fix issues with calculation tolerance. Several tests currently fail because cadquery-occ seems to use a more lenient tolerance value for calculations than the current freecad implementation. The tests fail because results aren't accurate to the expected values within enough decimal places. See this test failure for example:
```
======================================================================
FAIL: test_parent (t_cqparts.test_params.ComponentRefTests)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/justinbuchanan/src/github/cq/cqparts/tests/t_cqparts/test_params.py", line 396, in test_parent
    self.assertAlmostEqual(bb.xmin, -1)
AssertionError: -1.01 != -1 within 7 places
```

- [ ] multi-part assembly issues. I haven't looked into this yet, so I'm not sure what the cause is, but one test failure is:
```
======================================================================
FAIL: test_multipart_assembly (t_cqparts.test_codecs.TestStep)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/justinbuchanan/src/github/cq/cqparts/tests/t_cqparts/test_codecs.py", line 238, in test_multipart_assembly
    self.assertEqual(len(thing.components), 2)
AssertionError: 1 != 2
```

There are likely other issues, but here's what I've seen so far. Please feel free to add to this list. Also, if you're working on something, please put your name next to the item or comment so we can avoid duplicate work :).

jmwright, 2020-02-16:
I'm most of the way through an update of cqparts to CQ 2.0. It's on [this fork](https://github.com/jmwright/cqparts). It's built on top of the great work @adam-urbanczyk has been doing to get CadQuery updated to OCCT 7.4.x using [auto-generated bindings](https://github.com/CadQuery/OCP). There are some nice performance improvements and added features. Just FYI - I do intend to clean up the commented FreeCAD references once everything is working.

@fragmuffin Would you be supportive of merging this into master once it's been vetted properly and OCCT 7.4 support is mainline in CQ 2.0, or will it need to live on as a branch or fork?

As of this writing (and I'm quitting for the night), I have 0 assert failures and 4 errors. It's only a matter of time before I stamp those last 4 out, but if anybody is interested in taking a look, I'll post them below. Any other feedback on the changes that have been made are welcome too. If there's support for merging into master, I'll put a PR together once the tests run successfully and I have some instructions on how to run my branch against CQ 2.0 and OCCT 7.4.
```
======================================================================
ERROR: test_nested (t_cqparts.test_assembly.BoundingBoxTests)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/jwright/Downloads/repos/cqparts/tests/t_cqparts/test_assembly.py", line 348, in test_nested
    bb = obj.bounding_box
  File "/home/jwright/Downloads/repos/cqparts/src/cqparts/assembly.py", line 306, in bounding_box
    component.bounding_box for component in self.components.values()
  File "/home/jwright/Downloads/repos/cqparts/src/cqparts/assembly.py", line 306, in <genexpr>
    component.bounding_box for component in self.components.values()
  File "/home/jwright/Downloads/repos/cqparts/src/cqparts/assembly.py", line 306, in bounding_box
    component.bounding_box for component in self.components.values()
  File "/home/jwright/Downloads/repos/cqparts/src/cqparts/utils/geometry.py", line 38, in merge_boundboxes
    wrapped_bb = WrappedType(*(min_params + max_params))
TypeError: __init__(): incompatible constructor arguments. The following argument types are supported:
    1. OCP.Bnd.Bnd_Box()
    2. OCP.Bnd.Bnd_Box(theMin: OCP.gp.gp_Pnt, theMax: OCP.gp.gp_Pnt)

Invoked with: 15.0, -35.0, -45.0, 55.0, 35.0, -5.0

======================================================================
ERROR: test_single_layer (t_cqparts.test_assembly.BoundingBoxTests)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/jwright/Downloads/repos/cqparts/tests/t_cqparts/test_assembly.py", line 340, in test_single_layer
    bb = obj.bounding_box
  File "/home/jwright/Downloads/repos/cqparts/src/cqparts/assembly.py", line 306, in bounding_box
    component.bounding_box for component in self.components.values()
  File "/home/jwright/Downloads/repos/cqparts/src/cqparts/utils/geometry.py", line 38, in merge_boundboxes
    wrapped_bb = WrappedType(*(min_params + max_params))
TypeError: __init__(): incompatible constructor arguments. The following argument types are supported:
    1. OCP.Bnd.Bnd_Box()
    2. OCP.Bnd.Bnd_Box(theMin: OCP.gp.gp_Pnt, theMax: OCP.gp.gp_Pnt)

Invoked with: -1.0, -1.0, 0.0, 1.0, 1.0, 3.0

======================================================================
ERROR: test_from_matrix (t_cqparts.test_coordsystem.CoordSystemTests)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/jwright/Downloads/repos/cqparts/tests/t_cqparts/test_coordsystem.py", line 79, in test_from_matrix
    0,0,0,1
TypeError: __init__() takes from 1 to 2 positional arguments but 17 were given

======================================================================
ERROR: test_random_failsafe (t_cqparts.test_coordsystem.CoordSystemTests)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/jwright/anaconda3/envs/pywrap/lib/python3.6/site-packages/mock/mock.py", line 1369, in patched
    return func(*newargs, **newkeywargs)
  File "/home/jwright/Downloads/repos/cqparts/tests/t_cqparts/test_coordsystem.py", line 127, in test_random_failsafe
    cs = CoordSystem.random()
  File "/home/jwright/Downloads/repos/cqparts/src/cqparts/utils/geometry.py", line 211, in random
    normal=rand_vect(-1, 1),
  File "/home/jwright/Downloads/repos/cqparts/src/cqparts/utils/geometry.py", line 50, in __init__
    super(CoordSystem, self).__init__(origin, xDir, normal)
  File "/home/jwright/Downloads/repos/cadquery/cadquery/occ_impl/geom.py", line 472, in __init__
    self._setPlaneDir(xDir)
  File "/home/jwright/Downloads/repos/cadquery/cadquery/occ_impl/geom.py", line 718, in _setPlaneDir
    self.yDir = self.zDir.cross(self.xDir).normalized()
  File "/home/jwright/Downloads/repos/cadquery/cadquery/occ_impl/geom.py", line 122, in normalized
    return Vector(self.wrapped.Normalized())
OCP.Standard.Standard_ConstructionError: gp_Vec::Normalized() - vector has zero norm

----------------------------------------------------------------------
Ran 197 tests in 1.422s

FAILED (errors=4, skipped=4)
```
Related issues: #98 #144

zignig, 2020-02-16:
If my week is slightly less manic than my calendar claims, i will try to replicate results.

adam-urbanczyk, 2020-02-16:
@jmwright I see that `t_cqparts.test_assembly.BoundingBoxTests` fail due to calling OCP directly with wrong kind/number of arguments. It would be good to not depend directly on OCP/OCC in this package. If you now which functionality is needed in CQ I can quickly add it to the OCP branch.

jmwright, 2020-02-16:
@zignig Sounds good, thanks. If you get a chance, and if you're running Linux, it will probably be easiest to build a pywrap Anaconda environment. pywrap is the project that generates the OCP bindings.

https://github.com/CadQuery/pywrap

Once you have that environment, you can either `make` OCP directly, or I can provide a .so file download for you so that you don't have to go through all that. I copy the OCP .so file into the root of the cloned cadquery repo, then run the following line from within `cqparts/tests`:
```
PYTHONPATH="/home/jwright/Downloads/repos/cqparts/src:/home/jwright/Downloads/repos/cadquery" ./run.sh
```
There may be a better way to do it, but that setup lets me tinker with cadquery and cqparts at the same time pretty easily. Let me know if you have any questions.

@adam-urbanczyk 

> It would be good to not depend directly on OCP/OCC in this package. If you now which functionality is needed in CQ I can quickly add it to the OCP branch.

Thanks, I'll work on changing those calls and figuring out what's missing from cadquery that might be needed.

jmwright, 2020-02-16:
Here's a temporary link to the OCP .so file to help shorten the time to get up and running with cqparts on CQ 2.0 and OCCT 7.4.0.

https://www.dropbox.com/s/3w9cm5cwl4dly2u/OCP.cpython-36m-x86_64-linux-gnu.so?dl=0

jmwright, 2020-02-16:
@adam-urbanczyk CQ 2.0 already had an `add` method on BoundBox, so I just used that.

https://github.com/jmwright/cqparts/commit/556daf01f474c068fcd4ff1cd582051cd9ecb9a9

Getting close. Down to just 2 errors.
```
======================================================================
ERROR: test_from_matrix (t_cqparts.test_coordsystem.CoordSystemTests)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/jwright/Downloads/repos/cqparts/tests/t_cqparts/test_coordsystem.py", line 79, in test_from_matrix
    0,0,0,1
TypeError: __init__() takes from 1 to 2 positional arguments but 17 were given

======================================================================
ERROR: test_random_failsafe (t_cqparts.test_coordsystem.CoordSystemTests)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/jwright/anaconda3/envs/pywrap/lib/python3.6/site-packages/mock/mock.py", line 1369, in patched
    return func(*newargs, **newkeywargs)
  File "/home/jwright/Downloads/repos/cqparts/tests/t_cqparts/test_coordsystem.py", line 127, in test_random_failsafe
    cs = CoordSystem.random()
  File "/home/jwright/Downloads/repos/cqparts/src/cqparts/utils/geometry.py", line 199, in random
    normal=rand_vect(-1, 1),
  File "/home/jwright/Downloads/repos/cqparts/src/cqparts/utils/geometry.py", line 38, in __init__
    super(CoordSystem, self).__init__(origin, xDir, normal)
  File "/home/jwright/Downloads/repos/cadquery/cadquery/occ_impl/geom.py", line 472, in __init__
    self._setPlaneDir(xDir)
  File "/home/jwright/Downloads/repos/cadquery/cadquery/occ_impl/geom.py", line 718, in _setPlaneDir
    self.yDir = self.zDir.cross(self.xDir).normalized()
  File "/home/jwright/Downloads/repos/cadquery/cadquery/occ_impl/geom.py", line 122, in normalized
    return Vector(self.wrapped.Normalized())
OCP.Standard.Standard_ConstructionError: gp_Vec::Normalized() - vector has zero norm

----------------------------------------------------------------------
Ran 197 tests in 1.385s

FAILED (errors=2, skipped=4)
```

jmwright, 2020-02-19:
The tests pass now, but the test related to [CadQuery issue #283](https://github.com/CadQuery/cadquery/issues/283) is being skipped for now. I have left the display tests which rely on FreeCAD in for now (they're being skipped), until that functionality can be replaced by CQ-editor or something equivalent. The docs also need to be updated. In any case, cqparts should be usable with CQ 2.0 now, even with all the work that's left to do for a full transition.