cqparts/cqparts #138: gltf tolerance access

KammutierSpule, 2018-12-27:
I noticed a variable tolerance in the gltf.py
Could it be added as a parameter to the call ?
def __call__(self, filename='out.gltf', embed=False, tolerance... ? ):
I have no knowledge of python, so not sure what is the best way of implementing it.

fragmuffin, 2018-12-30:
@KammutierSpule That's a good point.
There was a reason for this originally. When I first wrote the gltf exporter it made more sense to me to make the functions doing the conversion `classmethod`s.

```python
>>> from cqparts.codec.gltf import GLTFExporter
>>> from cqparts_misc.basic.primatives import Cube

>>> cube = Cube()  # cqparts.Part instance
>>> GLTFExporter.part_buffer(cube)
<cqparts.codec.gltf.ShapeBuffer at 0x7fc056a93b50>
```

note: there is no `GLTFExporter` instance needed to get a `ShapeBuffer`.
this allows people to use the internal `GLTFExporter` functionality in a much more flexible way if they wish.

And `classmethod`s only have access to the class, not an instance of that class, even if it's called from `self`.
However, there's a strong argument for the above code to have an optional `tolerance=n` in the `part_buffer` call.

**TL;DR**
You're right, there should be a `tolerance` parameter... adding it now.
I tend to get wrapped up in search of a perfect design, which is mostly impossible anyway ;)

fragmuffin, 2018-12-30:
discussion continued in #141

closed 2019-01-05 by fragmuffin